mikakaraila/node-red-contrib-opcua #167: No Data in node red

Janakiramank, 2020-06-22:
Hi Mika,

I've connected the OPC UA server successfully but no data is coming. I'm able to read the same data from some other OPC UA client but not in node rod. I've very little knowledge on this OPC UA. 

There is 2 namespace in OPCUA server ns=2 & ns=3. I'm able to read the data from ns=3 where it is working without any username and password. ns=2 is allowed credential but even after entering the credential i'm not able to read the data. null values are coming. Need your support to resolve this issues
[image]

[image]
[image]

alebuert, 2020-06-22:
Hi,
i got the same Problem. Downloaded the newest version from today. Reading values always gives "null". When using datatype=Null in the inject node, the error message is gone but the result is still null.

Janakiramank, 2020-06-22:
> Hi,
> i got the same Problem. Downloaded the newest version from today. Reading values always gives "null". When using datatype=Null in the inject node, the error message is gone but the result is still null.

You've got any solution???

mikakaraila, 2020-06-22:
If you inject nodeId add into the json message also datatype=Int32

Janakiramank, 2020-06-23:
> If you inject nodeId add into the message topic with datatype=Int32
> Or use opcua item node.
> 
> Here is example with three different ways to read nodeId:
> [image]
> 
> [NODEID_json.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/4815014/NODEID_json.txt)
> 
> NOTE: last line is empty line, needed to send action!
> [nodeids.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/4815017/nodeids.txt)

I've tried ur suggestions but still same issues. Is that anything related to firewall settings? I'm running behind corporate proxy

[image]

mikakaraila, 2020-06-23:
Start with option -v verbose and attach console output here to see what is happening.
node-red -v YOURFLOW.json

And if you can attach also screenshot of nodeId with details from UaExpert.
It would help to understand what is wrong.

alebuert, 2020-06-23:
In my case the connection works pretty good with the iiot-opcua node but I have some other problems with that node... I made a screenshot of the log. It says, acces denied, what I can't understand because it works with iiot-opcua.
[image]

Janakiramank, 2020-06-23:
> Start with option -v verbose and attach console output here to see what is happening.
> node-red -v YOURFLOW.json
> 
> And if you can attach also screenshot of nodeId with details from UaExpert.
> It would help to understand what is wrong.

Thanks for your quick reply. I've attached the details
[image]
[image]
[image]

Log files attached in .txt
[node red log.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/4817631/node.red.log.txt)

mikakaraila, 2020-06-23:
Have you added username & passwd to endpoint at client node?

Because log shows statusocde:
Status-Code:BadUserAccessDenied (0x801f0000)

mikakaraila, 2020-06-23:
Check credentials from the client node:

[image]

Janakiramank, 2020-06-24:
> Check credentials from the client node:
> 
> [image]

I've used the correct username and password. attached screenshot for reference. Still issue is same. I'm able to even browse the object through the browser. But only data is not coming from OPC item
[image]

[image]

mikakaraila, 2020-06-24:
If you try with some other securityMode / securityPolicy than None / None ?

Janakiramank, 2020-06-25:
> If you try with some other securityMode / securityPolicy than None / None ?

I've tried but getting certificate trust issues.

[image]

mikakaraila, 2020-06-25:
Check your server and can you find trusted/rejected folders?
Most probably node-red client certificate is not trusted, it remains as rejected.

Janakiramank, 2020-06-25:
> Check your server and can you find trusted/rejected folders?
> Most probably node-red client certificate is not trusted, it remains as rejected.

Yes. It was in the Rejected folder. I've cut and pasted to trusted folder. But the issue still the same. 

[image]

mikakaraila, 2020-07-05:
Can you attach flow or show what you try to read (inject content)?
Does it contain datatype? 
Because in the debug log String<> Null is error types are not matching.

Janakiramank, 2020-07-17:
> Can you attach flow or show what you try to read (inject content)?
> Does it contain datatype?
> Because in the debug log String<> Null is error types are not matching.

Sorry for my late reply because of Covid19 we were under complete lockdown. I've attached my flow here to refer. I'm inject two different nodes. I'm not able to read where the data is protected with user credential. But I've ensured the user name and password used are correct one.
[flows.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/4936468/flows.txt)

mikakaraila, 2020-07-20:
Please re-test, there were bug/error passing username & password. See #173

Janakiramank, 2020-07-21:
> Please re-test, there were bug/error passing username & password. See #173

This is awesome. My issue is resolved after upgrading to the latest version. Thank you so much, Mika. I really appreciate your quick response and great efforts

mikakaraila, 2020-08-03:
Any comments why did you reopened this one?

Janakiramank, 2020-08-03:
> Any comments why did you reopened this one?

I didn't notice it. May be by mistake. Thank you

closed 2020-08-03 by Janakiramank