mikakaraila/node-red-contrib-opcua #168: Error connecting with no certificate

pfetrial, 2020-06-23:
Hi Mika,

After updating the opcua nodes ?? a simple flow to read the value of a node in a OPC server is no more possible. Above are the error Messages: ( I don't use certificates..)

The client says : test OPC UA connection error: The connection has been rejected by server,Please check that client certificate is trusted by server.Err = (Transaction has timed out ( timeout = 60000 ms))

and in the cmd window:

Welcome to Node-RED
===================

23 Jun 16:53:09 - [info] Node-RED version: v1.0.4
23 Jun 16:53:09 - [info] Node.js  version: v12.16.3
23 Jun 16:53:09 - [info] Windows_NT 10.0.17763 x64 LE
23 Jun 16:53:10 - [info] Loading palette nodes
23 Jun 16:53:12 - [info] Dashboard version 2.22.1 started at /ui
23 Jun 16:53:12 - [info] Settings file  : \Users\i00204714\.node-red\settings.js
23 Jun 16:53:12 - [info] Context store  : 'default' [module=memory]
23 Jun 16:53:12 - [info] User directory : \Users\i00204714\.node-red
23 Jun 16:53:12 - [warn] Projects disabled : editorTheme.projects.enabled=false
23 Jun 16:53:12 - [info] Flows file     : \Users\i00204714\.node-red\flows_CI-13574.json
23 Jun 16:53:12 - [info] Server now running at http://127.0.0.1:1880/
23 Jun 16:53:12 - [info] Starting flows
23 Jun 16:53:12 - [info] [OpcUa-Client:test]    No certificate used.
23 Jun 16:53:12 - [info] Started flows
14:53:13.016Z :message_builder_base          :153   Error   client0 cannot construct object with nodeID ns=0;i=449
23 Jun 16:54:13 - [error] [OpcUa-Client:test] test OPC UA connection error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (Transaction has timed out ( timeout = 60000 ms))

What is the problem?
Is there an another Version?
Thanks
Patrick

mikakaraila, 2020-06-23:
I tested with latest v0.2.64 with Prosys Ua Simulation server. Client node can read nodeId like ns=3;s=Counter no problem.
I test also OPC UA Server inside node-red with UaExpert and works.

[image]

pfetrial, 2020-06-24:
Hello Mika
I made  some others tests but I still stuck. Always the same message appearing in the cmd window ( independent of the node that I whant to read!). 

14:53:13.016Z :message_builder_base :153 Error client0 cannot construct object with nodeID ns=0;i=449

I can not say If the problem is related to OPC nodes because the problem comes from the message builder.. I try to re-install teh opcua nodes.. but nothing helps.
What is your feeling?
Thnaks
Patrick

mikakaraila, 2020-06-24:
What is server you are trying to use?
There are some issues with the same case:
Open62541 & Wago:
https://github.com/open62541/open62541/issues/1276

Node-opcua & Wago:
https://github.com/node-opcua/node-opcua/issues/640

Siemens & FreeOPCUA:
https://groups.google.com/forum/#!topic/freeopcua/xSrXiTP2eSM

pfetrial, 2020-06-24:
It is the uacppserver but same results with Prosys OPC UA.

 

open62541/open62541#1276 <https://github.com/open62541/open62541/issues/1276> 

node-opcua/node-opcua#640 <https://github.com/node-opcua/node-opcua/issues/640> 

https://groups.google.com/forum/#!topic/freeopcua/xSrXiTP2eSM

mikakaraila, 2020-06-24:
OK as I have tested also Prosys UA Simulation Server that works on my computer OK then please check your node-opcua version.
I am using v2.6.4 even latest one is now v2.7.0 it must be depending your environment / node-opcua version.

mikakaraila, 2020-06-24:
This case was with KepServer. You must trust to node-red client.

https://www.gitmemory.com/issue/mikakaraila/node-red-contrib-opcua/151/633257451

pfetrial, 2020-06-25:
Hi Mika,

I want to send an update of the  problem with the OPC nodes  :

I exchanged the opc ua  nodes (node-red-contrib-opcua) used in a simple flow with the IIot nodes (node-red-contrib-iiot-opcua).

With those nodes all my flows are correctly working. I don’t know why. But I think this statement can bring some clarity in the problem (?)

 

 

How can I suppress the non-working nodes and re-install properly the nodes?

Patrick

 

 <https://www.gitmemory.com/issue/mikakaraila/node-red-contrib-opcua/151/633257451> https://www.gitmemory.com/issue/mikakaraila/node-red-contrib-opcua/151/633257451

mikakaraila, 2020-06-26:
It uses very old node-opcua javascript based, I suppose there is some differences now as this one uses typescript based node-opcua. I have to study my code later...

pfetrial, 2020-06-26:
how can I delete every thing and reinstall everything? What are the files to be deleted?

mikakaraila, 2020-06-26:
Look under 
\users\USER\.node-red\node-modules\node-red-contrib-XXXX
folders, they contain node source files.

pfetrial, 2020-07-02:
Thanks

I cleared this folder and install every things from fresh…

A another question : I saw a video where you are talking testing OPC UA server. If I am right , you are using robot framework . 

Are you happy with this framework ? Is this product adapted to test OPC UA server?

Thanks for your reply.

Patrick

mikakaraila, 2020-07-03:
We have been using Robot Framework for a long time. 
I wrote own adapter Python library to run test cases.
They will test all read/write, methods, alarms & history access.

pfetrial, 2020-07-03:
Is this framework/library available on internet ?Can we use it?

We want to use a automatic test sequencer to test specific functionalities of an OPC server ( special behavior of a OPC server from a machine)

Thanks

Patrick

pfetrial, 2020-07-03:
Sorry to bother you..

Do you think that robot framework is adapted to do opc ua server test? We are also looking on CTT ..

mikakaraila, 2020-07-03:
CTT uses own javascript based test library.
Our uses this one: https://github.com/FreeOpcUa/python-opcua
Test library contains some specific parts that I cannot share.

pfetrial, 2020-07-03:
Ok but the test sequencer is your own?

mikakaraila, 2020-07-03:
Test cases are bind to our DNA-OPCUA-Server and DNA-OPCUA-Client.
Performance is great, but it uses node addon library to communicate our DNA protocol.

mikakaraila, 2020-07-03:
CI continues integration runs on Jenkins2 as new code is committed then it builds and runs tests. Robot tests are run with pabot.

pfetrial, 2020-07-04:
Ok thanks

I will give a try with CTT and hope it will suit with my expectations-

closed 2020-07-05 by mikakaraila

pfetrial, 2020-07-18:
Hello Mika,

 

Unfortunately I discovered an another problem with the OPC UA nodes.

https://discourse.nodered.org/t/opc-ua-read-value-ok-but-write-not-possible/30260

 

greetings

Patrick

 

Closed #168 <https://github.com/mikakaraila/node-red-contrib-opcua/issues/168> .

mikakaraila, 2020-07-18:
Some servers Accept only value, try without timestamp and statuscode.

pfetrial, 2020-07-20:
With node-red , can do this?