mikakaraila/node-red-contrib-opcua #170: Use OPCUABrowse and store Tag item to CSV or Database

henjoe, 2020-07-02:
Hi Mika, 

I am just wondering if this is possible using the OPCUABrowse? I tried to use your template in browsing using the OPCUABrowse ->  Template node in node-red and it works fine. What I want to do if it is possible that the item browsed can be put into a text file like CSV or database like sqlite? 

This is actually your template in browsing: this work fine.
[image]

I hope you can help me here, or at least just send the item into a text file instead on <li> tag in UI template node.

Thanks Mika! 

With kind regards,
Henjoe

mikakaraila, 2020-07-03:
I am bit busy at work, vacation starting, but if there are rainy days I will take a look.
This should be quite small flow...

henjoe, 2020-07-03:
Hello Mika, 

I really appreciate your work in here.
No worries on your side. Just take your time, I can integrate this to my node-red project at a later part.

Again,
Thank you.

With kind regards,
Henjoe

mikakaraila, 2020-07-05:
Here is simple example, note I have to limit msg rate not to use too many sessions.
[browse-nodeids.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/4874980/browse-nodeids.txt)

henjoe, 2020-07-05:
Hello Mika! Thanks for this. Highly appreciated! 
Will try this soon at my testing environment. 

I just want to add this little question, I have this 1 OPClientNode that subscribe to 1000 tags at an interval of every (1s). What I notice ia my nodered page (on which I create the logic programming not the dashboard) seems a little bit "laggy", is thia normal? I mean you experience this too? 
But in terms of functionality, it seems like it is working well, it is just that when I edit something to nodered, it gives me a harder time to edit becuase of the lag effect. 

Again, thanks Mika!
Cheers from PH.

Regards,

mikakaraila, 2020-07-05:
I expect 1000 variables uses quite a lot node messages. 
It totally depends your flow logic how much there is processing needed.

closed 2020-07-05 by mikakaraila

henjoe, 2020-07-05:
Thanks Mika,

yeah! definitely a lot of nodes. 

Again, thanks for clarifying! 

With kind regards,
Henjoe