mikakaraila/node-red-contrib-opcua #171: get-installed-path error on Windows 10

JoeZeng133, 2020-07-02:
Error details:
 [error] [OpcUa-Client:c5905f8b.20e04] Error: get-installed-path: module not found "node-opcua-client" in path C:\Users\username\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node-opcua-client

Seems like this error is specific to Windows since multiple similar issues have already been raised and solved on Linux. I don't know how this can be solved so I hard coded the full path into 102-opcuaclient.js

mikakaraila, 2020-07-03:
Strange, I have been using Windows and it works. 
Could it be something to do with installation as you have different than normal one...
Have you installed with -g option?

JoeZeng133, 2020-07-03:
Yes, I used -g option for all of my packages

mikakaraila, 2020-07-04:
Ok then I know what to fix...

mikakaraila, 2020-07-05:
Please test, fixed to v0.2.65

JoeZeng133, 2020-07-05:
thanks for the fix. Here is another scenario: user decided to change its global installing path. Would this be handled?

mikakaraila, 2020-07-05:
As long as you use npm to change "config" it should work.

Here is used code snippet
    var clientPkg = null;
    try {
      clientPkg = installedPath.getInstalledPathSync('node-opcua-client');
      if (clientPkg) {
        verbose_log("Found globally installed path: " + clientPkg);
      }
    }
    catch (err) {
      verbose_log("Not globally installed, checking local folders next");
      clientPkg = null;
    }

    if (clientPkg == null) {
      clientPkg = installedPath.getInstalledPathSync('node-opcua-client', {
        paths: [
        path.join(__dirname, '..'),
        path.join(__dirname, '../..'),
        path.join(process.cwd(), './node_modules'),
        path.join(process.cwd(), '../node_modules'), // Linux installation needs this
        path.join(process.cwd(), '.node-red/node_modules'),
        ],
      });
      verbose_log("Found locally installed path: " + clientPkg);
    }

JoeZeng133, 2020-07-05:
Hi Mika. I have run npm config set prefix "C:\Users\Public\Documents\npm" and install everything there. Still getting the following error:

[error] [OpcUa-Client:3787107a.95a7b] Error: get-installed-path: module not found "node-opcua-client" in path C:\Users\Public\Documents\npm\node_modules\node-red-contrib-opcua\node-opcua-client

I ran npm list -g and found that client is in this folder

C:\Users\Public\Documents\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-client

mikakaraila, 2020-07-05:
Run node-red -v your flow.json

Console output will tell us more...

JoeZeng133, 2020-07-05:
there you go.

5 Jul 12:15:40 - [info] Starting flows
5 Jul 12:15:40 - [info] [inject:b3d6a1cb.798e] repeat = 1000
5 Jul 12:15:40 - [info] [OpcUa-Client:1d428270.86bfce]  No certificate used.
5 Jul 12:15:40 - [info] [OpcUa-Client:1d428270.86bfce] Not globally installed, checking local folders next
5 Jul 12:15:40 - [error] [OpcUa-Client:1d428270.86bfce] Error: get-installed-path: module not found "node-opcua-client" in path C:\Users\zzeng2\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node-opcua-client
    at Object.getInstalledPathSync (C:\Users\zzeng2\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\get-installed-path\dist\index.js:164:11)
    at new OpcUaClientNode (C:\Users\zzeng2\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\opcua\102-opcuaclient.js:90:33)
    at Object.createNode (C:\Users\zzeng2\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\runtime\lib\nodes\flows\util.js:493:31)
    at Flow.start (C:\Users\zzeng2\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\runtime\lib\nodes\flows\Flow.js:193:48)
    at start (C:\Users\zzeng2\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\runtime\lib\nodes\flows\index.js:345:33)
    at tryCatchReject (C:\Users\zzeng2\AppData\Roaming\npm\node_modules\node-red\node_modules\when\lib\makePromise.js:845:30)
    at runContinuation1 (C:\Users\zzeng2\AppData\Roaming\npm\node_modules\node-red\node_modules\when\lib\makePromise.js:804:4)
    at Fulfilled.when (C:\Users\zzeng2\AppData\Roaming\npm\node_modules\node-red\node_modules\when\lib\makePromise.js:592:4)
    at Pending.run (C:\Users\zzeng2\AppData\Roaming\npm\node_modules\node-red\node_modules\when\lib\makePromise.js:483:13)
    at Scheduler._drain (C:\Users\zzeng2\AppData\Roaming\npm\node_modules\node-red\node_modules\when\lib\Scheduler.js:62:19)
5 Jul 12:15:40 - [info] Started flows

mikakaraila, 2020-07-05:
Ok installed path cannot find globally installed package. I will re-test then I should create issue there...

mikakaraila, 2020-07-05:
Workaround to test v0.2.66

JoeZeng133, 2020-07-06:
thanks, now it's working

closed 2020-07-07 by mikakaraila