mikakaraila/node-red-contrib-opcua #172: Can't read values from S7-1500

Ogoku, 2020-07-07:
Hi there,

I can not read values from a S7-1500 PLC with the OpcUA Client Node. 
The OpcBrowser Node works till the third layer, then it will not find any Items.

Error log:
7 Jul 13:17:57 - [info] Flows starten
7 Jul 13:17:57 - [info] [OpcUa-Client:634fe9a7.c05648]  No certificate used.
7 Jul 13:17:57 - [info] Flows gestartet
7 Jul 13:17:57 - [error] [OpcUa-Client:634fe9a7.c05648]  OPC UA connection error: Cannot find ANONYMOUS user token policy in end point description

JSON:
[
    {
        "id": "625817a9.9b743",
        "type": "tab",
        "label": "Flow 1",
        "disabled": false,
        "info": ""
    },
    {
        "id": "634fe9a7.c05648",
        "type": "OpcUa-Client",
        "z": "625817a9.9b743",
        "endpoint": "891ac138.50ffc8",
        "action": "subscribe",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": "1",
        "timeUnit": "s",
        "certificate": "n",
        "localfile": "",
        "securitymode": "None",
        "securitypolicy": "None",
        "name": "",
        "x": 510,
        "y": 560,
        "wires": [
            []
        ]
    },
    {
        "id": "4bdc0b25.78ad54",
        "type": "OpcUa-Item",
        "z": "625817a9.9b743",
        "item": "ns=3;s=MK.Job-ID",
        "datatype": "Int16",
        "value": "",
        "name": "Job-ID",
        "x": 270,
        "y": 560,
        "wires": [
            [
                "634fe9a7.c05648"
            ]
        ]
    },
    {
        "id": "891ac138.50ffc8",
        "type": "OpcUa-Endpoint",
        "z": "",
        "endpoint": "opc.tcp://192.168.124.1:4840",
        "secpol": "None",
        "secmode": "None",
        "login": true
    }
]

mikakaraila, 2020-07-07:
Error message tells cannot find ANONYMOUS user token?

Can you look some variable or folder attributes with UaExpert just to check that they are not protected or something else is blocking access?

Ogoku, 2020-07-08:
See image attached.

Maybe Siemens structure is messed up. 
Worked with a python opc ua client before, there I had no error to acces this variable.

[image]

mikakaraila, 2020-07-08:
What are attributes on that object?

If you use my example flow with delays then you have to add more levels...

Ogoku, 2020-07-08:
I have used [this ](https://github.com/mikakaraila/node-red-contrib-opcua/blob/master/examples/Browser_contrib_ui_flow.json) example to browse.

Attached Image show Attributes of DataBlocksGlobal node
[image]

It seems that I can not browse any of Siemens specific node folders like Input, Output, Memory, Timers, DataBlocks, etc.

mikakaraila, 2020-07-08:
Are you using user name and password? Siemens needs perhaps those. Fill-in credentials.

Ogoku, 2020-07-08:
It is already set. 

As already said I can connect and brose the opc server till DataBlocksGlobal, Inputs, Outputs, etc. 
All nodes after those point are not accessible.

With python module opcua it works very well.

mikakaraila, 2020-07-12:
Can you run node-red -v flow.json to see what is happening?

Ogoku, 2020-07-13:
Sure, here is the result:

13 Jul 13:48:12 - [info] Server now running at http://127.0.0.1:1880/
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] endpoint opc.tcp://192.168.124.1:4840
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Application URI urn:SIMATIC.S7-1500.OPC-UA.Application:%C3%9CGS+SSK1-50K0
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Product URI https://www.siemens.com/s7-1500
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Application Name SIMATIC.S7-1500.OPC-UA.Application:%C3%9CGS+SSK1-50K0
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Security Mode 1
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] securityPolicyUri http://opcfoundation.org/UA/SecurityPolicy#None13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Type 0
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] certificate ... /*endpoint.serverCertificate*/
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] discoveryUrls opc.tcp://192.168.124.1:4840
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] endpoint opc.tcp://192.168.124.1:4840
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Application URI urn:SIMATIC.S7-1500.OPC-UA.Application:%C3%9CGS+SSK1-50K0
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Product URI https://www.siemens.com/s7-1500
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Application Name SIMATIC.S7-1500.OPC-UA.Application:%C3%9CGS+SSK1-50K0
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Security Mode 2
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] securityPolicyUri http://opcfoundation.org/UA/SecurityPolicy#Basic256
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Type 0
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] certificate ... /*endpoint.serverCertificate*/
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] discoveryUrls opc.tcp://192.168.124.1:4840
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] endpoint opc.tcp://192.168.124.1:4840
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Application URI urn:SIMATIC.S7-1500.OPC-UA.Application:%C3%9CGS+SSK1-50K0
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Product URI https://www.siemens.com/s7-1500
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Application Name SIMATIC.S7-1500.OPC-UA.Application:%C3%9CGS+SSK1-50K0
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Security Mode 3
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] securityPolicyUri http://opcfoundation.org/UA/SecurityPolicy#Basic256
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Type 0
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] certificate ... /*endpoint.serverCertificate*/
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] discoveryUrls opc.tcp://192.168.124.1:4840
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] endpoint opc.tcp://192.168.124.1:4840
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Application URI urn:SIMATIC.S7-1500.OPC-UA.Application:%C3%9CGS+SSK1-50K0
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Product URI https://www.siemens.com/s7-1500
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Application Name SIMATIC.S7-1500.OPC-UA.Application:%C3%9CGS+SSK1-50K0
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Security Mode 2
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] securityPolicyUri http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Type 0
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] certificate ... /*endpoint.serverCertificate*/
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] discoveryUrls opc.tcp://192.168.124.1:4840
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] endpoint opc.tcp://192.168.124.1:4840
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Application URI urn:SIMATIC.S7-1500.OPC-UA.Application:%C3%9CGS+SSK1-50K0
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Product URI https://www.siemens.com/s7-1500
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Application Name SIMATIC.S7-1500.OPC-UA.Application:%C3%9CGS+SSK1-50K0
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Security Mode 3
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] securityPolicyUri http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Type 0
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] certificate ... /*endpoint.serverCertificate*/
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] discoveryUrls opc.tcp://192.168.124.1:4840
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Identify Token for : Security Mode= 1
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] policyId Username
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] tokenType 1
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] issuedTokenType null
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] issuerEndpointUrl null
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] securityPolicyUri http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Identify Token for : Security Mode= 2
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] policyId Username
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] tokenType 1
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] issuedTokenType null
[... cut ...]
    at ClientSecureChannelLayer._on_message_received (C:\Users\ogoreks\.node-red\node_modules\node-opcua-secure-channel\dist\source\client\client_secure_channel_layer.js:495:13)
    at MessageBuilder.<anonymous> (C:\Users\ogoreks\.node-red\node_modules\node-opcua-secure-channel\dist\source\client\client_secure_channel_layer.js:143:22)
    at MessageBuilder.emit (events.js:315:20)
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Client status: connection error
13 Jul 13:48:12 - [info] [OpcUa-Client:c2649942.25a84] Client disconnected!

mikakaraila, 2020-07-19:
I changed session creation without credentials, please test.

mikakaraila, 2020-07-20:
See #173 I just fixed it. There was bug in the code passing username & password to session.

Ogoku, 2020-07-21:
@mikakaraila: Thank you for your effort. I will check it when I'm back at office.

mikakaraila, 2020-08-03:
Any news? See also #181

mikakaraila, 2020-08-04:
As #181 is closed

closed 2020-08-04 by mikakaraila