mikakaraila/node-red-contrib-opcua #173: Cannot connect to OPC Server with Security None but User Authentication

sniicker, 2020-07-09:
I createt a Node Red Compact Server without allowing Anonymous access. I set username and passwort in the Config Node.
When i try to connect with the UAExpert Software i have no problem setting security to none but add username and password.
When i try to connect with the node red OPCclient and set the enpoint secuity policy and mode both to none and add the username and password, i get the error: cannot find anonymous user token.

Help says: Certificate is not needed if security is None.

Whats the Problem?

regards sniicker

youngkzy, 2020-07-10:
This was EXACTLY the problem I ran into last time I posted.

mikakaraila, 2020-07-12:
Have to take look. There was some changes in the API and it can cause this problem or my code does accept username & password without certificate. It will need a fix then.

youngkzy, 2020-07-13:
Here is the info from Node-RED posted to the terminal.

Thanks in advance!

13 Jul 13:15:30 - [info] [OpcUa-Client:eff756de.c8f698]         No certificate used.
13 Jul 13:15:30 - [info] Started flows
13 Jul 13:15:30 - [info] [tcp in:78cdb0e7.d8d6a8] listening on port 1026
13 Jul 13:15:30 - Connection successful for database EDB_ with user Sa
17:15:33.682Z :message_builder_base          :153   Error   client0 cannot construct object with nodeID ns=0;i=449
13 Jul 13:15:42 - [error] [OPCUA-IIoT-Listener:e73e25cf.c37de8] Error: String cannot be coerced to a nodeId : ns=2;i=Sort1.Sort_CLX.Global.Gaylord_Data1.Gaylord_Read_Index113 Jul 13:16:33 - [error] [OpcUa-Client:eff756de.c8f698]  OPC UA connection error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (Transaction has timed out ( timeout = 60000 ms))

benbamm, 2020-07-14:
Same problem here. I updated to the latest version and it stopped working :(

mikakaraila, 2020-07-15:
Ok I will take a look next weekend

sniicker, 2020-07-15:
This is my error when i deactivate the anonymous token in the NodeRed OPC Server.

15 Jul 09:11:08 - [info] Starting flows
15 Jul 09:11:08 - [info] [OpcUa-Client:f03ce77a.19b728]         No certificate used.
15 Jul 09:11:09 - [info] Started flows
15 Jul 09:11:11 - [error] [OpcUa-Client:f03ce77a.19b728]  OPC UA connection error: Cannot find ANONYMOUS user token policy in end point description

thanks @mikakaraila

mikakaraila, 2020-07-16:
Can be related to node-opcua #823

mikakaraila, 2020-07-19:
I made change to code, but this worries me:
"Cannot find ANONYMOUS user token policy in end point description"
Some servers have configuration for allowing anonymous access:
[image]

mikakaraila, 2020-07-19:
New version available, please test.

sniicker, 2020-07-20:
Sorry but still same problem.

if i connect without the use of use credentials the Client is connectig as Anonymous. **GOOD**
[image]
[image]

if i try to connect with a user and passwort, the Client is STILL connecting as Anonymous **BAD**
[image]
[image]

if i try the same with ua expert the client is connecting as the user i wanted **desired reaction**
[image]
[image]

best regards

saurabhgupta0502, 2020-07-20:
Hello, It is same issue with latest version also (0.2.68).
I have created OPC-UA Server on WAGO Make PLC and using another WAGO PLC with Node-Red installed on it. I am running the OPC-UA Client on Node-red to read the first PLC.
The OPC-UA Server works fine with UA Expert, but not working on Node-Red.

INCIDENTLY: The connection is working fine on OPCUA- Browser node, but giving error on OPA-UA Client Node.

[image]

[image]

mikakaraila, 2020-07-20:
Ok I have to change options

mikakaraila, 2020-07-20:
New version available. I dont add certificates if security mode is None.´Let´s see if it is now similar with browse node.

sniicker, 2020-07-20:
How can i update?
No new version in palette manager. 
also tried to run npm install node-red-contrib-opcua
this shows Version 0.2.69
but in palette manager is still 0.2.68?

sniicker, 2020-07-20:
Still same problem with Version 0.2.69

mikakaraila, 2020-07-20:
Ok I have to setup test environment. Was not so as I thought...

sniicker, 2020-07-20:
i can send you my flows if you want?

mikakaraila, 2020-07-20:
Yes it would help

youngkzy, 2020-07-20:
I updated and tested  0.2.70 still cannot connect to local server anonymously.

saurabhgupta0502, 2020-07-21:
Which was the last version which successfully connected the Local servers as Anonymous or Username/Password?
I want to use that till the time issues in current version is resolved.

sniicker, 2020-07-21:
@mikakaraila  **Great Job** Man. Woks now. 👍 
I tested with Integration Objects UA Server Simulation and the node red compact server.
both work fine. 😃 
Thanks a lot for you fast soloution. Would be nice from others too.

closed 2020-07-21 by sniicker

saurabhgupta0502, 2020-07-21:
@mikakaraila Thanks!!! 👍
Works now with WAGO PLC OPC-UA Server. Client is hosted on another PLC on Docker.
Not tested with Security None and No Credentials.