mikakaraila/node-red-contrib-opcua #174: Question: How to unsubscibe to specific OPC Tag

henjoe, 2020-07-12:
Dear Mika, 

I hope you're doing fine at this moment. I have some questions regarding unsubscription of OPC-UA-Client node. 
I have a flow where it gets all the list of tags via a text file, all tags there will be automatically subscribe to OPC-UA-Client node. 
This works fine and without a problem. 
But aside from adding tags in a file, I also want that the user will automatically unsbuscribe to a tag when they updated the file. For example, in my file i have this list of tags:
Tag1
Tag2
Tag3

And these 3 OPC tags will be subscribe to opcua-client node, however tomorrow I decided to edit the file and delete the Tag3, so the file now will be like this:
Tag1
Tag2

What I notice is after I edited the file, and inject it to opcua-client, the Tag3 is still in list of subscription, this was actually expected since we just edited the file but we never unsubscribe. Now my question is is there anyway to do what I like to do? I am thinking that it may be easier that when everytime I'll send the list of tags to opcua-client node, I need to unsubscribe to all the current subscription of that node, and then pass the edited list of tags to be subscribe again. 

This is actually my node looks like: 
[image]

the input of opcUA is from a list of tags on a file.

Again, thanks Mika! 

Always stay safe.

Regards,
Henjoe

mikakaraila, 2020-07-12:
I expect I made example couple of weeks ago. Case is closed. But question is that unsubscribe will terminate all variables. There's terminate for individual variable. I have to check this because there were changes in the API.

henjoe, 2020-07-12:
Thanks Mika ! 
Highly appreciated it.

I will try to look on the cased close regarding the unsubscribe, I tell you if this work on my case.

Again, thank you.

With kind regards,
Henjoe

henjoe, 2020-07-13:
Hello Mika, 

Upon looking at the list of issues, I actually can't see the cased close regarding the unsubscribing.

Regards,
Henjoe

mikakaraila, 2020-07-13:
It is #154

henjoe, 2020-08-04:
Hello Mika, 

With the latest update of node-red-contrib-opcua, I am able to succesfully unsubscribe. 
However I noticed this characteristic of the node-opcuaClient.

I have a subscription of over 30 OPC Tags (I used config file to subscribe multiple OPC Tags), if I want to unsubscribe to any of the 30 OPC Tags, I used this function code:

[image]

I can succesfully unsubscribe, BUT only to the first OPC Tag subscribe, as I said above that I have 30 OPC Tags, when I tried to unsubscribe for example the 2nd OPC tag, it gives me this error:

[image]

Again, the unsubscription is only works on the 1st OPC tag to subscribe to the client. 
Is there a way to fix this? 

Thanks Mika.

With kind regards,
Henjoe

mikakaraila, 2020-08-04:
Actually there are two ways:
1) Unsubscribe that will "delete/remove" subscription that can contain multiple monitored items
2) DeleteMonitoredItem (earlier this was terminate) that will "delete" item you have "subscribed"

I expect you want to use the second one...

henjoe, 2020-08-04:
Thanks Mika! 

Will try the 2nd one. 

Regards,
henjoe

henjoe, 2020-08-05:
Hello Mika, so edited my function node to be like this:

[image]

But it doesn't unsubscribe, I also tried this action:
[image]

But still it doesn't unsubscribe, the one who can successfully unsubscribe is this: 
[image]

But again, this only works for 1 subscribe item.

Hoping for your kind help.

Thank you.

With kind regards,
Henjoe

mikakaraila, 2020-08-05:
I have to check which action is available/implemented

henjoe, 2020-08-05:
First action is "subscribe".

Here is actually my testing flow:
[image]

May I know what you need from me to troubleshoot it? 

Thank Mika,

With kind regards,
Henjoe

mikakaraila, 2020-08-05:
Correct config:
First subscribe (inject node):
[image]

Second subscribe another one (inject node):
[image]

Third unsubscribe first subscribed item (inject node):
[image]

mikakaraila, 2020-08-05:
I have to look subscription how to delete it. I expect these is minor issue...

henjoe, 2020-08-05:
Thanks Mika!

Will try the updated version of the node. I hope it works now.

Will let you know the results.

With kind regards,
Henjoe

henjoe, 2020-08-06:
Hello Mika, 

So I updated my node to 0.2.74, however upon testing, the problem is still there.
It can only unsubscribe to the first OPC Tag to subscribe in opcuaClient node.

[image]

Here's what my config file to subscribe look like:

[image]

"Expression" tagname is the first OPC tag to subscribe so I can unsubscribe to it.
my "Counter" is at last OPC tag to subscribe.
[image]

When I tried to unsubscribe the Expression, it was successful. Look below my Trend:
[image]

Hoping for your help.

Thank you.

With kind regards,
Henjoe

henjoe, 2020-08-06:
But it;s kinda weird because it works on my testing flow:
[image]

3 subscribe item, I can unsubscribe any of the item.

henjoe, 2020-08-06:
Hello Mika,

Here's what I noticed.
When I subscribe to any OPC tag one by one with at least a second difference between the opc item,
I can successfully unsubscribe to any item:
[image]

But when I to subscribe simultaneously, only the node who subscribe first that I can unsubscribe:

[image]

Any thoughts here? 

Thank Mika,

With kind regards,
Henjoe

henjoe, 2020-08-06:
Hello Mika, 

WIth what I noticed a while ago, I tried to give a delay for each msg.topic that will subscribe to opcUaClient. 

msg.topic = ["ns=3;s=Counter","ns=3;s=Expression",ns=3;s=Random"]

before, I used to send this array 1by1 to uaClient. 
So I just added a delay for each nodeid to send and then it all works fine now with unsubscription.

I just wait for your advice regarding here, but again thank you for your effort.

With kind regards,
Henjoe

mikakaraila, 2020-08-06:
OK, code was still missing delete from internal storage map. I fixed it to v0.2.75.

closed 2020-08-06 by mikakaraila