mikakaraila/node-red-contrib-opcua #177: Question: How to get the timestamps of a value with the READ MULTIPLE Action

Sergi132, 2020-07-21:
Hello,

Is it possible to get the timestamps back when using the READ MULTIPLE Action?

[image]

We have the UaExpert to do the simulation and we wanted to get Source Timestamp:

[image]

But in the return payload using the READ MULTIPLE it doesn't appear. If we use the SUBSCRIBE action we can get it with the "container" of all the variables, not for each one. Nevertheless we need it with the READ MULTIPLE action as per requirement of our interface. Are we missing something or this is not possible?

Thanks and regads,
Sergi

mikakaraila, 2020-07-22:
I expect node-opcua readVariable doesn't support returning timestamps.

henjoe, 2020-07-24:
I think you can do it by only on "Subscription", I tried all the method, only the subscription has the sourceTimestamp property.

closed 2021-06-10 by mikakaraila