mikakaraila/node-red-contrib-opcua #178: OK this is just too weird...

youngkzy, 2020-07-23:
[image]

  I had this version 0.2.61 on the system before I upgraded to 0.2.xx (unsure).  So I also upgraded to the latest version.  0.2.70... It still is having issues with connection to the OPC Server... So I uninstalled 2.70.   Now, oddly I have 0.2.61 still on my system.  OK WHAT?  and it works where 0.2.70 DOES NOT.   I CANNOT UNINSTALL 0.2.61. 

[image]

Here it is after I tried to uninstall it... 

[image]
[image]

mikakaraila, 2020-07-24:
I have not used uninstall with node-red packages. I have always just installed/updated new version.
BTW it is better to inject empty string than timestamp because timestamp is big integer in payload.

youngkzy, 2020-07-24:
@mikakaraila, 
  Thank you for you kind reply and your attention to this.  It looks like I can upgrade via Node-RED Installer.  However, It is not able to uninstall it.  I upgraded through npm and then uninstalled using npm on 0.2.70 and you see that this is the result.  As it stands, it looks like I'm in a catch-22.  I will try to install the same version on top of itself and then uninstall it.  0.2.61 was installed by a previous conversation logged in these issues.  I'll let you know the results soon.

Kind Regards,
@youngkzy

mikakaraila, 2020-07-30:
I modified create session back to same as in 0.2.51 Let´s see if this was causing error.

closed 2020-08-10 by mikakaraila