mikakaraila/node-red-contrib-opcua #179: Cannot connect to ProSys OPC Simulator after upgrade to latest version

henjoe, 2020-07-24:
Dear Mika,

This is a little bit weird. But when I updated to latest version of node, I can no longer connect to my ProSys OPC UA server, before, I can connect to it.

This is actually the error:
[image]

Hoping for your response.

Thank you.

With kind regards,
Henjoe

mikakaraila, 2020-07-24:
Did you accept new certificate? I tested in my environment with OPCUA-TEST-NODES.json and evrything was working.
There seems to be other errors in your flow. Are you sure they don´t cause this error?

henjoe, 2020-07-24:
Hello Mika, I actually downgrade the opcua node for a moment to continue my testing, and it works fine again (no error). 

I will try to upgrade again on the latest and try to accept the certificate on the side of Prosys. But to be honest, I already tried this before (to "trust" the certificate) but my prosys has no response on it, kindly see below snap shot on my prosys. Does this certification causes the error?

[image]

Thanks Mika! 

With kind regards,
Henjoe

mikakaraila, 2020-07-24:
There should be also node-red client certificate. It is missing...

henjoe, 2020-07-24:
Hmm... THis is actually weird, because before, I don't care about the certification on the side of OPC Server. Do you have a snap shot on your OPC Server certificates so I could have an idea?

Thanks and regards,
Henjoe

henjoe, 2020-07-25:
Hello Mika, 

I think I can connect now to the OPC Server, however there is an error, do I need to add the "action" property on the message now? even though I already set the action to "subscribe" on the opcClientNode? 

Here is what I have on Debug:

[image]

and this is the settings on my client node:
[image]

As can be seen on the debug node, I already added the property "action" with the value of "subscribe", but still getting the same error. 

Hoping for your help.

Thank you Mika ! Highly appreciated it.

henjoe, 2020-07-25:
Hello Mika, 

It is okay now, I actually did not notice that the action property should be in the payload not on the MSG level.

Thanks for all your support ! Highly appreciated it.

With kind regards,
Henjoe

closed 2020-07-25 by henjoe