mikakaraila/node-red-contrib-opcua #180: warning: your opc client is sending multiple request

henjoe, 2020-07-26:
Dear Mika,

I have warning when running node-red. Basically, I have opcnode client with 1000 tags of subscription and interval of 1000ms. 
Is this normal warning? 
[image]

When I check my trend, I still get the correct values of the OPC item, I am just worried about the warning it gives to me. 
Does this have an effect on the performance of node-red itself? 

Thank you.

With kind regards,
Henjoe

mikakaraila, 2020-07-26:
It is from node-opcua. It gives warning if message queue is bigger than 10.

henjoe, 2020-07-29:
Hello Mika noted on this.

Is there a way to avoid it? 
Or does warning affects OPC node performance? 

Thanks ! 

Regards,
Henjoe

mikakaraila, 2020-07-29:
You add some delays, but as you just create items and it's warning it should not affect anyway.

If it comes with reading or writing then it could have something to check.

henjoe, 2020-07-29:
I think it comes from reading, I set nodered to just read opcua data. 
I am just wondering why on last version of nodeopcua I didn't recieve this warning. 

Anyways Mika, what do you mean add some delays? You mean I will add the interval rate of my subscription? As I mention before, I have 1000 tags with 1 seconds update rate. 

Thanks Mika, 

With kind regards, 
Henjoe

mikakaraila, 2020-07-29:
No, it comes from adding monitored items. Etienne added this to node-opcua. It wasn't there earlier.
How you add monitored items?
Don't change anything in subscription it doesn't help.

henjoe, 2020-07-29:
Hello Mika, that's actually weird because I never had "monitored" items, I never use the "monitor"as an action to my opcuaClient. 

What I have is just 14 opcUAClient node all of which is "Subcribes" method. 

thanks and regards,
Henjoe

mikakaraila, 2020-07-29:
Subscription used internally items as monitored. If 14 client nodes are run parallel then at start time it exceeds limit of 10 and causes warning.
If you delay 4 some hundred of milli seconds then warning could disappear.

henjoe, 2020-07-29:
Noted on this Mika,

Thank you for clarifiying this to me.

With kind regards,
Henjoe

closed 2020-08-03 by mikakaraila

alexh225, 2020-09-28:
Hi Mika, thank you very much for your work on this library!
One question related to this topic:
As far I understand the subscription works synchronously and subscribe the items one by one.
Would it be possible to enhance the node by "SUBSCRIBE MULTIPLE", like "READ MULTIPLE" and "WRITE MULTIPLE"?
As far I see, there is a method in the Node OPCUA for handling multiple subscribe requests - **ClientMonitoredItemGroup**:
https://github.com/node-opcua/node-opcua/issues/691

Regards!
Alex