mikakaraila/node-red-contrib-opcua #181: Rejected by server

bunicht, 2020-07-27:
Hello sir!
I've been running this for at least 6 months now with both a Siemens s1500 cpu and a Beckhoff twincat OPC server without issues.
After upgrading to 0.2.7.0 and node red ver 1.0.6 nothing is working anymore.
The message in debug window is 

 OPC UA connection error: The connection has been rejected by server,Please check that client certificate is trusted by server.Err = (Transaction has timed out ( timeout = 60000 ms))

Nothing on the server sides has been changed .
I can still connect with UAexpert 
Please help!

mikakaraila, 2020-07-27:
Have you trusted client certificate on server side? Install creates new certificate.

bunicht, 2020-07-28:
No nothing on any of the servers.
Both are wide open even without credentials.

mikakaraila, 2020-07-28:
Can you provide console output with command node-red -v flow.json

bunicht, 2020-07-28:
Here it comes:

sandvik@Ubuntuserver:~$ node-red -v flow.json
28 Jul 14:57:18 - [info]

Welcome to Node-RED
===================

28 Jul 14:57:18 - [info] Node-RED version: v1.1.2
28 Jul 14:57:18 - [info] Node.js  version: v8.10.0
28 Jul 14:57:18 - [info] Linux 5.4.0-42-generic x64 LE
28 Jul 14:57:19 - [info] Loading palette nodes
28 Jul 14:57:19 - [info] Settings file  : /home/sandvik/.node-red/settings.js
28 Jul 14:57:19 - [info] Context store  : 'default' [module=memory]
28 Jul 14:57:19 - [info] User directory : /home/sandvik/.node-red
28 Jul 14:57:19 - [warn] Projects disabled : editorTheme.projects.enabled=false
28 Jul 14:57:19 - [info] Flows file     : /home/sandvik/.node-red/flow.json
28 Jul 14:57:19 - [info] Creating new flow file
28 Jul 14:57:19 - [warn]

---------------------------------------------------------------------
Your flow credentials file is encrypted using a system-generated key.

If the system-generated key is lost for any reason, your credentials
file will not be recoverable, you will have to delete it and re-enter
your credentials.

You should set your own key using the 'credentialSecret' option in
your settings file. Node-RED will then re-encrypt your credentials
file using your chosen key the next time you deploy a change.
---------------------------------------------------------------------

28 Jul 14:57:19 - [info] Starting flows
28 Jul 14:57:19 - [info] Started flows
28 Jul 14:57:19 - [error] Unable to listen on http://127.0.0.1:1880/
28 Jul 14:57:19 - [error] Error: port in use
sandvik@Ubuntuserver:~$

Thanks for helping me !!!

scorbin-ewi, 2020-07-28:
The last two lines indicate that the port you are trying to start the server on, 1880 is already in use.  Perhaps you already have a node-red server running.

You should be able to use the following command to determine what it is:

sudo lsof -i -P -n | grep 1880

The first column is the name of the process using that port.

Shane Corbin

bunicht, 2020-07-29:
Sorry for delay!!
I'm noob at this :)
Node red running in snap on Ubuntu server18.04
I've upgraded node red  to ver 1.06 . Can that be the problem?
Anyway here is the console output

sandvik@Ubuntuserver:~$ node-red -v flow.json
29 Jul 08:17:40 - [info]

Welcome to Node-RED
===================

29 Jul 08:17:40 - [info] Node-RED version: v1.1.2
29 Jul 08:17:40 - [info] Node.js  version: v8.10.0
29 Jul 08:17:40 - [info] Linux 5.4.0-42-generic x64 LE
29 Jul 08:17:41 - [info] Loading palette nodes
29 Jul 08:17:41 - [info] Settings file  : /home/sandvik/.node-red/settings.js
29 Jul 08:17:41 - [info] Context store  : 'default' [module=memory]
29 Jul 08:17:41 - [info] User directory : /home/sandvik/.node-red
29 Jul 08:17:41 - [warn] Projects disabled : editorTheme.projects.enabled=false
29 Jul 08:17:41 - [info] Flows file     : /home/sandvik/.node-red/flow.json
29 Jul 08:17:41 - [info] Creating new flow file
29 Jul 08:17:41 - [warn]

---------------------------------------------------------------------
Your flow credentials file is encrypted using a system-generated key.

If the system-generated key is lost for any reason, your credentials
file will not be recoverable, you will have to delete it and re-enter
your credentials.

You should set your own key using the 'credentialSecret' option in
your settings file. Node-RED will then re-encrypt your credentials
file using your chosen key the next time you deploy a change.
---------------------------------------------------------------------

29 Jul 08:17:41 - [info] Starting flows
29 Jul 08:17:41 - [info] Started flows
29 Jul 08:17:41 - [info] Server now running at http://127.0.0.1:1880/

Regards/Lennart

mikakaraila, 2020-07-29:
I mean flow that will use client node and one item to show more details.

bunicht, 2020-07-29:
I'm lost here.
Don't now what to do actually

mikakaraila, 2020-07-29:
Use your own flow json file with node-red -v option

alimexTSH, 2020-07-29:
I have a similar issue with a Siemens S7-1500 CPU but my OS is Windows Server.
No certificate is used and trusted.

Here is my log file:

Welcome to Node-RED
===================

29 Jul 10:44:42 - [info] Node-RED version: v1.1.2
29 Jul 10:44:42 - [info] Node.js  version: v10.15.3
29 Jul 10:44:42 - [info] Windows_NT 10.0.14393 x64 LE
29 Jul 10:44:43 - [info] Loading palette nodes
29 Jul 10:45:01 - [info] Dashboard version 2.23.0 started at /ui
29 Jul 10:45:02 - [info] Settings file  : \Users\administrator\.node-red\settings.js
29 Jul 10:45:02 - [info] Context store  : 'default' [module=memory]
29 Jul 10:45:02 - [info] User directory : \Users\administrator\.node-red
29 Jul 10:45:02 - [warn] Projects disabled : editorTheme.projects.enabled=false
29 Jul 10:45:02 - [info] Flows file     : C:\Users\administrator\.node-red\flow.json
29 Jul 10:45:02 - [warn]

---------------------------------------------------------------------
Your flow credentials file is encrypted using a system-generated key.

If the system-generated key is lost for any reason, your credentials
file will not be recoverable, you will have to delete it and re-enter
your credentials.

You should set your own key using the 'credentialSecret' option in
your settings file. Node-RED will then re-encrypt your credentials
file using your chosen key the next time you deploy a change.
---------------------------------------------------------------------

29 Jul 10:45:02 - [info] Starting flows
29 Jul 10:45:02 - [info] [OpcUa-Client:3d5a1971.1f33be]         No certificate used.
29 Jul 10:45:02 - [info] [OpcUa-Client:3d5a1971.1f33be] Found node-opcua globally installed path: C:\Users\administrator\AppData\Roaming\npm\node_modules\node-opcua-client
29 Jul 10:45:02 - [info] [OpcUa-Client:3d5a1971.1f33be] Found node-red-contrib-opcua globally installed path: C:\Users\administrator\AppData\Roaming\npm\node_modules\node-red-contrib-opcua
29 Jul 10:45:02 - [info] [OpcUa-Client:3d5a1971.1f33be] Client certificate not used!
29 Jul 10:45:02 - [info] [OpcUa-Client:3d5a1971.1f33be] Connection options:{"securityPolicy":"http://opcfoundation.org/UA/SecurityPolicy#Non","securityMode":1,"endpoint_must_exist":false,"defaultSecureTokenLifetime":40000,"connectionStrategy":{"maxRetry":10512000,"initialDelay":2000,"maxDelay":30000},"keepSessionAlive":true}
29 Jul 10:45:02 - [info] [OpcUa-Client:3d5a1971.1f33be] EndPoint: {"id":"fb42e6f7.1eaf2","type":"OpcUa-Endpoint","z":"","_closeCallbacks":[],"_inputCallback":null,"_inputCallbacks":null,"_asyncDelivery":true,"wires":[],"_wireCount":0,"credentials":{},"endpoint":"opc.tcp://192.168.10.11:4840","securityPolicy":"None","securityMode":"None","login":false,"user":null,"password":null}
29 Jul 10:45:02 - [warn] [OpcUa-Client:3d5a1971.1f33be] OpcUaClientNode: Create Client: {"securityPolicy":"http://opcfoundation.org/UA/SecurityPolicy#None","securityMode":1,"endpoint_must_exist":false,"defaultSecureTokenLifetime":40000,"connectionStrategy":{"maxRetry":10512000,"initialDelay":2000,"maxDelay":30000},"keepSessionAlive":true}
29 Jul 10:45:02 - [info] [OpcUa-Client:3d5a1971.1f33be] Client status: create client
29 Jul 10:45:02 - [info] [OpcUa-Client:3d5a1971.1f33be] async series - connecting
29 Jul 10:45:02 - [info] [OpcUa-Client:3d5a1971.1f33be] Client status: connecting
29 Jul 10:45:02 - [info] Started flows
29 Jul 10:45:02 - [info] Server now running at http://127.0.0.1:1880/
08:45:02.534Z :message_builder_base          :153   Error   client0 cannot construct object with nodeID ns=0;i=449
29 Jul 10:46:02 - [error] [OpcUa-Client:3d5a1971.1f33be]  OPC UA connection error: The connection has been rejected by server,Please check that client certificate is trusted by server.Err = (Transaction has timed out ( timeout = 60000 ms))
29 Jul 10:46:02 - [info] [OpcUa-Client:3d5a1971.1f33be] Error: The connection has been rejected by server,Please check that client certificate is trusted by server.Err = (Transaction has timed out ( timeout = 60000 ms))    at _internal_create_secure_channel (C:\Users\administrator\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-client\source\private\client_base_impl.ts:654:27)    at async.series (C:\Users\administrator\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-client\source\private\client_base_impl.ts:583:17)    at err (C:\Users\administrator\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-client\node_modules\async\dist\async.js:2955:19)    at wrapper (C:\Users\administrator\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-client\node_modules\async\dist\async.js:268:20)    at iterateeCallback (C:\Users\administrator\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-client\node_modules\async\dist\async.js:413:21)    at C:\Users\administrator\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-client\node_modules\async\dist\async.js:321:20    at result (C:\Users\administrator\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-client\node_modules\async\dist\async.js:2953:17)    at secureChannel.create (C:\Users\administrator\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-client\source\private\client_base_impl.ts:550:21)    at ClientSecureChannelLayer._performMessageTransaction (C:\Users\administrator\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:1058:13)    at modified_callback (C:\Users\administrator\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:1362:31)
29 Jul 10:46:02 - [info] [OpcUa-Client:3d5a1971.1f33be] Client status: connection error
29 Jul 10:46:02 - [info] [OpcUa-Client:3d5a1971.1f33be] Client disconnected!

mikakaraila, 2020-07-29:
I don't have Siemens S7-1500 to test this.
Can you check what is object ns=0;i=449

mikakaraila, 2020-07-29:
Please check also that Anonymous is accepted on Siemens server settings

alimexTSH, 2020-07-29:
Originally I inserted ns=3;s="AlignmentTable"."General"."BarcodeScanner"."Scanning"."OK" into OPC UA item with boolean type. The ns=0;i=449 is generated automatically and does not change when a different OPC-Object is used...
Here you can see the first dialog from the client config.

[image]

This is the second page of the configuration
[image]

The Anonymous login is working with OPCUA-IIoT-Listener-Node right away. I have tested this with another Siemens S7-1500 CPU which is configured to only allow logins with security "none" and have the same results.
Connect with UaExpert is also working without credentials and certificates on both CPUs.

mikakaraila, 2020-07-29:
Can you test if browse node is working?

bunicht, 2020-07-29:
For me everything is solved by reinstall Node Red.
The old version was 1.0.6 and now 1.1.2

alimexTSH, 2020-07-29:
Browse-Node is throwing errors:
[image]

I have used the test file OPCUA_BROWSE_and_SUBSCRIBE.json.

mikakaraila, 2020-07-29:
Ok I tested with OPCUA-TEST-NODES.json and Prosys UA Simulation server. Did you notice that node-red update solved bunicht environment?

alimexTSH, 2020-07-29:
I am running the latest version of node-red-contrib-opcua and node-red. I had this error before and after the update.

I have tested the connection to CodeSys OPCUA-Server. Even without the use of a certificate I can see the error 
"OPC UA connection error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (Transaction has timed out ( timeout = 60000 ms))"

Siemens has a certificate which is self-signed but Codesys has no certificate at all.

Have you tried to run with Windows Server 2016?

Thank you for the valued feedback!

mikakaraila, 2020-07-29:
I have just windows 10. I can run more tests next week.

mfurla8, 2020-08-03:
Same error on my side
Siemens 1500 OPC UA Server... Any Ideas??

bunicht, 2020-08-03:
I have just connected a s7 1200 OPC Ua Server.

This is the debug message 
no cert no credentials on serverside 

 OPC UA connection error: The connection has been rejected by server,Please check that client certificate is trusted by server.Err = (this transport protocol is currently not supported :opc:)

mfurla8, 2020-08-03:
it is now on s7 1500 the same error about certificate....
How can i downgrade this version?

mikakaraila, 2020-08-03:
Downgrade:
cd "home"/.node-red
npm install node-red-contrib-opcua@0.2.51

mikakaraila, 2020-08-03:
@bunicht  can you explain more?

bunicht, 2020-08-04:
> @bunicht can you explain more?

It was just that I missed a semicolon in endpoint node adress

mikakaraila, 2020-08-04:
OK I will close this one. If somebody has problem with S7-1200 then it will be another issue.

mfurla8, 2020-08-04:
i still didn't solve with 1500

mikakaraila, 2020-08-04:
OK please provide more information as @bunicht got it working. Please read all comments of the issue.

mfurla8, 2020-08-04:
[image]
[image]
[image]

**How can i set my own key? What key?** 
You should set your own key using the 'credentialSecret' option in
your settings file. Node-RED will then re-encrypt your credentials
file using your chosen key the next time you deploy a change.

[image]

alimexTSH, 2020-08-04:
Please see my comment https://github.com/mikakaraila/node-red-contrib-opcua/issues/181#issuecomment-665533712
I have attached a log file from debug parameter. I can confirm the same configuration as mfurla8.

mikakaraila, 2020-08-04:
And you have only one client node in flow? Can you provide verbose log?
Something is different on these environments compared to @bunicht he got it working.

mfurla8, 2020-08-04:
Yes only one client...
[image]

How can i get verbose log?

mfurla8, 2020-08-04:
[1391,452466701 10.0.69.10 S1] Address DB61,REAL94 has value 54.0807991027832 and quality OK
[1391,452590606 10.0.69.10 S1] Address DB61,INT98 has value 9 and quality OK
[1391,452722636 10.0.69.10 S1] Address DB61,REAL100 has value 4085.6708984375 and quality OK
[1391,452846333 10.0.69.10 S1] Address DB61,DINT104 has value 2538 and quality OK
[1391,452975811 10.0.69.10 S1] Address DB61,S108.40 has value  and quality OK
[1391,453105133 10.0.69.10 S1] Address DB61,INT150 has value 1 and quality OK
[1391,453195653 10.0.69.10 S1] We are calling back our readDoneCallback.
4 Aug 11:10:19 - [error] [OpcUa-Client:2691c73c.b1465]  OPC UA connection error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (Transaction has timed out ( timeout = 60000 ms))
4 Aug 11:10:19 - [error] [OpcUa-Client:OPC LINE] OPC LINE OPC UA connection error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (Transaction has timed out ( timeout = 60000 ms))
4 Aug 11:10:19 - [error] [OpcUa-Client:323e6517.a5b692]  OPC UA connection error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (Transaction has timed out ( timeout = 60000 ms))
[1393,449459944 10.0.69.10 S1] Reading All Items (readAllItems was called)
[1393,449787910 10.0.69.10 S1] Calling SRP from RAI
[1393,449937336 10.0.69.10 S1] SendReadPacket called
[1393,450595820 10.0.69.10 S1] Sending Read Packet
[1393,452259504 10.0.69.10 S1] Received 146 bytes of S7-data from PLC.  Sequence number is 2
[1393,452548200 10.0.69.10 S1] ReadResponse called
[1393,452869447 10.0.69.10 S1] Time is 0 seconds and 2.8 ms.

mikakaraila, 2020-08-04:
And if you use empty string on inject node on the left instead of timestamp?

mfurla8, 2020-08-04:
> 
> 
> Use start command:
> node-red -v yourflow.json
> 
> And if you use empty string on inject node on the left instead of timestamp?

it is the same...

alimexTSH, 2020-08-04:
Here is my log:
PS C:\Users\administrator\.node-red> node-red -v flow_new.json
4 Aug 10:24:29 - [info]

Welcome to Node-RED
===================

4 Aug 10:24:29 - [info] Node-RED version: v1.1.2
4 Aug 10:24:29 - [info] Node.js  version: v10.15.3
4 Aug 10:24:29 - [info] Windows_NT 10.0.14393 x64 LE
4 Aug 10:24:30 - [info] Loading palette nodes
4 Aug 10:24:37 - [info] Dashboard version 2.23.0 started at /ui
4 Aug 10:24:38 - [info] Settings file  : \Users\administrator\.node-red\settings.js
4 Aug 10:24:38 - [info] Context store  : 'default' [module=memory]
4 Aug 10:24:38 - [info] User directory : \Users\administrator\.node-red
4 Aug 10:24:38 - [warn] Projects disabled : editorTheme.projects.enabled=false
4 Aug 10:24:38 - [info] Flows file     : \Users\administrator\.node-red\flow_new.json
4 Aug 10:24:38 - [info] Creating new flow file
4 Aug 10:24:38 - [warn]

---------------------------------------------------------------------
Your flow credentials file is encrypted using a system-generated key.

If the system-generated key is lost for any reason, your credentials
file will not be recoverable, you will have to delete it and re-enter
your credentials.

You should set your own key using the 'credentialSecret' option in
your settings file. Node-RED will then re-encrypt your credentials
file using your chosen key the next time you deploy a change.
---------------------------------------------------------------------

4 Aug 10:24:38 - [info] Starting flows
4 Aug 10:24:38 - [info] Started flows
4 Aug 10:24:38 - [info] Server now running at http://127.0.0.1:1880/
4 Aug 10:27:35 - [info] Modifizierte Nodes werden gestoppt
4 Aug 10:27:35 - [info] Geänderte Nodes gestoppt
4 Aug 10:27:35 - [info] Modifizierte Nodes werden gestartet
4 Aug 10:27:35 - [info] [OpcUa-Client:f93bc86b.73bdf]   No certificate used.
4 Aug 10:27:35 - [info] [OpcUa-Client:f93bc86b.73bdf] Found node-opcua globally installed path: C:\Users\administrator\A
ppData\Roaming\npm\node_modules\node-opcua-client
4 Aug 10:27:35 - [info] [OpcUa-Client:f93bc86b.73bdf] Found node-red-contrib-opcua globally installed path: C:\Users\adm
inistrator\AppData\Roaming\npm\node_modules\node-red-contrib-opcua
4 Aug 10:27:35 - [info] [OpcUa-Client:f93bc86b.73bdf] Client certificate not used!
4 Aug 10:27:35 - [info] [OpcUa-Client:f93bc86b.73bdf] Connection options:{"securityPolicy":"http://opcfoundation.org/UA/
SecurityPolicy#None","securityMode":1,"endpoint_must_exist":false,"defaultSecureTokenLifetime":40000,"connectionStrategy
":{"maxRetry":10512000,"initialDelay":2000,"maxDelay":30000},"keepSessionAlive":true}
4 Aug 10:27:35 - [info] [OpcUa-Client:f93bc86b.73bdf] EndPoint: {"id":"395713b7.82d04c","type":"OpcUa-Endpoint","z":"","
_closeCallbacks":[],"_inputCallback":null,"_inputCallbacks":null,"_asyncDelivery":true,"wires":[],"_wireCount":0,"creden
tials":{},"endpoint":"opc.tcp://192.168.10.11:4840","securityPolicy":"None","securityMode":"None","login":false,"user":n
ull,"password":null}
4 Aug 10:27:35 - [warn] [OpcUa-Client:f93bc86b.73bdf] OpcUaClientNode: Create Client: {"securityPolicy":"http://opcfound
ation.org/UA/SecurityPolicy#None","securityMode":1,"endpoint_must_exist":false,"defaultSecureTokenLifetime":40000,"conne
ctionStrategy":{"maxRetry":10512000,"initialDelay":2000,"maxDelay":30000},"keepSessionAlive":true}
4 Aug 10:27:35 - [info] [OpcUa-Client:f93bc86b.73bdf] Client status: create client
4 Aug 10:27:35 - [info] [OpcUa-Client:f93bc86b.73bdf] async series - connecting
4 Aug 10:27:35 - [info] [OpcUa-Client:f93bc86b.73bdf] Client status: connecting
4 Aug 10:27:35 - [info] Gestartete geänderte Nodes
08:27:35.604Z :message_builder_base          :153   Error   client0 cannot construct object with nodeID ns=0;i=449
4 Aug 10:28:13 - [error] [OpcUa-Client:f93bc86b.73bdf]  OPC UA connection error: The connection has been rejected by ser
ver,
Please check that client certificate is trusted by server.
Err = (_socket has been disconnected by third party)
4 Aug 10:28:13 - [info] [OpcUa-Client:f93bc86b.73bdf] Error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (_socket has been disconnected by third party)
    at _internal_create_secure_channel (C:\Users\administrator\.node-red\node_modules\node-red-contrib-opcua\node_module
s\node-opcua-client\source\private\client_base_impl.ts:654:27)
    at async.series (C:\Users\administrator\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-client
\source\private\client_base_impl.ts:583:17)
[... 3 lines cut ...]
_modules\async\dist\async.js:268:20)
    at iterateeCallback (C:\Users\administrator\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-cl
ient\node_modules\async\dist\async.js:413:21)
    at C:\Users\administrator\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-client\node_modules\
async\dist\async.js:321:20
    at result (C:\Users\administrator\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-client\node_
modules\async\dist\async.js:2953:17)
    at secureChannel.create (C:\Users\administrator\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcu
a-client\source\private\client_base_impl.ts:550:21)
    at ClientSecureChannelLayer._performMessageTransaction (C:\Users\administrator\.node-red\node_modules\node-red-contr
ib-opcua\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:1058:13)
    at modified_callback (C:\Users\administrator\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-s
ecure-channel\source\client\client_secure_channel_layer.ts:1362:31)
4 Aug 10:28:13 - [info] [OpcUa-Client:f93bc86b.73bdf] Client status: connection error
4 Aug 10:28:13 - [info] [OpcUa-Client:f93bc86b.73bdf] Client disconnected!

[image]
[image]
[image]
[image]
[image]

I don't know why it tries to connect to the nodeID ns=0;i=449.

As you can see I have also used the empty string to inject the flow.

mfurla8, 2020-08-04:
> 
> 
> > Use start command:
> > node-red -v yourflow.json
> > And if you use empty string on inject node on the left instead of timestamp?
> 
> it is the same...

i don't have a clue... how to solve this... but it works perfect till upgrade...

mikakaraila, 2020-08-04:
Can you downgrade back to old 0.2.51 and do same verbose log to compare what happens?

mfurla8, 2020-08-04:
pi@raspberrypi:/home $ sudo npm install node-red-contrib-opcua@0.2.51
npm WARN notice [SECURITY] lodash has the following vulnerability: 1 high. Go he      re for more details: https://www.npmjs.com/advisories?search=lodash&version=4.17      .19 - Run `npm i npm@latest -g` to upgrade your npm version, and then `npm audit      ` to get more info.

> deasync@0.1.20 install /home/node_modules/deasync
> node ./build.js

gyp WARN EACCES user "root" does not have permission to access the dev dir "/roo      t/.node-gyp/8.11.4"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/tmp/.node-gyp"
gyp ERR! configure error
gyp ERR! stack Error: EACCES: permission denied, mkdir '/home/node_modules/deasy      nc/build'
gyp ERR! System Linux 4.14.62-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gy      p/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/node_modules/deasync
gyp ERR! node -v v8.11.4
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
Build failed
npm WARN enoent ENOENT: no such file or directory, open '/home/package.json'
npm WARN home No description
npm WARN home No repository field.
npm WARN home No README data
npm WARN home No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! deasync@0.1.20 install: `node ./build.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the deasync@0.1.20 install script.
npm ERR! This is probably not a problem with npm. There is likely additional log      ging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-08-04T11_46_56_153Z-debug.log
pi@raspberrypi:/home $

mfurla8, 2020-08-04:
WTFF???

mikakaraila, 2020-08-04:
You should install under .node-red folder and perhaps --unsafe option

mikakaraila, 2020-08-04:
Found some discussion related to this problem:
https://groups.google.com/forum/#!searchin/freeopcua/ns$3D0$3Bi$3D449%7Csort:date/freeopcua/4NJdqXE6oCg/Zf76IcHb8XcJ
https://groups.google.com/forum/#!searchin/freeopcua/ns$3D0$3Bi$3D449%7Csort:date/freeopcua/xSrXiTP2eSM/ylKduOEdAwAJ

mfurla8, 2020-08-04:
[image]
anyone saw like that... it freezess???

alimexTSH, 2020-08-04:
I got it working with 0.2.51. Had to uninstall with npm uninstall node-red-contrib-opcua --save and PowerShell with administrative rights.
After that I tried to update to 0.2.72 and have the certificate problem again. So between this two version something changed that caused the error. Maybe I'll need to update with npm instead of node-red's integrated palette update function.

Thanks for pointing out that the old version might be better!

mfurla8, 2020-08-05:
[image]
here is the problem... with installing certificates...

mfurla8, 2020-08-05:
[image]

mikakaraila, 2020-08-05:
Thanks, finally somebody found root cause.
I suppose Etienne has changed this, I will report problem to him -> node-opcua.

mikakaraila, 2020-08-05:
Did you try to install with option: --unsafe-perms

mfurla8, 2020-08-05:
> 
> 
> Did you try to install with option: --unsafe-perms

it's the same

mikakaraila, 2020-08-05:
I got instructions from Etienne and new typescript based API needs one extra parameter, this is causing rejected by server error.
It is needed and it is causing problems now.
I will fix it today. Another issue is why postinstall fails.

mfurla8, 2020-08-05:
i am happy that we found the problem

mikakaraila, 2020-08-05:
@mfurla8 Etienne needs some more info, look issue on node-opcua

mikakaraila, 2020-08-05:
New version 0.2.73 available, please test.

I don´t have Siemens S7-1500 somebody should do this testing.

mfurla8, 2020-08-06:
[image]
it freezes here?

hmmm???

mikakaraila, 2020-08-06:
Can be something with to do with Internet connection or you have to use command:
npm install node-red-contrib-opcua --unsafe-perms

mfurla8, 2020-08-06:
i made out but...

[image]

alimexTSH, 2020-08-06:
Same experience as mfurla8. The version installed correctly. I also removed the client object and opc endpoint to ensure that all old settings were overwritten. 
[image]

mikakaraila, 2020-08-06:
Fixed, but npmjs.org timeouts. I will try to publish later or tomorrow morning.

I made some not needed changes and cleaned those.

mfurla8, 2020-08-07:
i install 0.2.76.... but still same error.... ???

mikakaraila, 2020-08-07:
Can you connect with UaExpert to same endpoint?

[9 more replies]

closed 2020-08-10 by mikakaraila