mikakaraila/node-red-contrib-opcua #182: "Browse done" not really useful

mgaceanu, 2020-07-29:
Setting the status at the current place results in setting the status after each processed line (see https://github.com/mikakaraila/node-red-contrib-opcua/blob/30cd5c1efcedabbf29c1756f4b8bbbfa364d189d/opcua/102-opcuaclient.js#L1293 ).
I would like to use a status node to know when the browsing is complete. Is there any other way to do this?

mikakaraila, 2020-07-30:
I added own status setting internally that will report when whole browse is done:
[image]

mikakaraila, 2020-07-30:
New version 0.2.71 includes this

mgaceanu, 2020-08-03:
Thanks for the really quick response and for the fix!

closed 2020-08-03 by mikakaraila