mikakaraila/node-red-contrib-opcua #185: How to get "quality" of the OPC Tags?

henjoe, 2020-08-08:
Hello Mika, 

I noticed that on OPCUaClient node with a subscribe method, there is no "quality" property of the OPC Tag?

[image]

Is there a way how to get the quality? right now what I can get is the listed property above.

Thank Mika,

With kind regards,
Henjoe

mikakaraila, 2020-08-10:
I will take a look, it will need enhancement. I would not to break existing structure so I should add it to msg.

mikakaraila, 2020-08-10:
I added statusCode to msg object:
[image]

mikakaraila, 2020-08-10:
Available in v0.2.78