mikakaraila/node-red-contrib-opcua #186: OpcUA-Item Node to work with Variables

Bono1969, 2020-08-10:
Hi,
is there a way (or exists a way) to work with a variable in the Node?
To explain: i have a lot of OPC Servers where i have to browse every time the same values. To be able to work with subflows and not need to edit every single node. In the picture i add, the field in the red frame changes from server to server. In my case from S-201 to S-260.
[image]

Bono1969, 2020-08-10:
in addition, as Part of the Payload? For Example "msg.machine"?

mikakaraila, 2020-08-10:
No, much easier to use mustache (template) and update needed topic string with it.
I can provide some example later.

Bono1969, 2020-08-10:
An example is fantastic👍🏻

mikakaraila, 2020-08-10:
This way:
[image]

mikakaraila, 2020-08-10:
I hope this is enough. Only ugly thing is that topic = character is encoded to &#x3D;  and must be replaced by back to =

Bono1969, 2020-08-11:
At the moment i'm not able to connect to the OPC Servers (IT Issues).
Next Week i should be able to Test with your Example.
Have a nice time.

mikakaraila, 2021-01-18:
Not anymore active? Closed.

closed 2021-01-18 by mikakaraila