mikakaraila/node-red-contrib-opcua #187: Connection to server times out, doesn't reconnect...

sb934, 2020-08-10:
For simulation purposes, I have a server file written in python which refreshes new values every few seconds. I am trying to connect to this server running on my local-host, to read node values. The communication is SignedAndEncrypted (Basic256Sha256) policy ; where the server has it's own copy of certificate and private key attached and meanwhile I am using the default certificate, key pair of NodeRed OPCUA. I am able to connect and retrieve values for the first few minutes, however the connection fails after a few minutes and fails to reconnect. I am attaching the error log...

#### Error Log received on server.py
Exception raised while parsing message from client, closing
Traceback (most recent call last):
  File "/home/saksham/.local/lib/python3.7/site-packages/opcua/server/binary_server_asyncio.py", line 75, in _process_data
    ret = self.processor.process(hdr, buf)
  File "/home/saksham/.local/lib/python3.7/site-packages/opcua/server/uaprocessor.py", line 83, in process
    msg = self._connection.receive_from_header_and_body(header, body)
  File "/home/saksham/.local/lib/python3.7/site-packages/opcua/common/connection.py", line 309, in receive_from_header_and_body
    chunk = MessageChunk.from_header_and_body(self.security_policy, header, body)
  File "/home/saksham/.local/lib/python3.7/site-packages/opcua/common/connection.py", line 49, in from_header_and_body
    decrypted = crypto.decrypt(data.read(len(data)))
  File "/home/saksham/.local/lib/python3.7/site-packages/opcua/crypto/security_policies.py", line 172, in decrypt
    return self.Decryptor.decrypt(data)
AttributeError: 'NoneType' object has no attribute 'decrypt'

#### Error on node-red terminal

21:51:10.710Z :message_builder_base          :109   Error   client1 _decrypt_MSG : Sign and Encrypt : Invalid packet signature

I have no idea what exactly is the issue because the connection works and communicates initially and then fails...

mikakaraila, 2020-08-11:
As node-opcua is used a lot, I am expecting that Python OPC UA Server contains some leak/bug as it is most probably used in this way so much.

Error message Invalid packet signature is coming from node-opcua. 
Most probably Python library is not calculating signature correctly or decrypt:ing it right way.
You should make issue to Python OPCUA.
If you need more deeply information from the node-opcua message Etienne will give you advises how to debug/trace this more deeply.

I will close this one as it is not actually related to node-red-contrib-opcua.

mikakaraila, 2020-08-24:
You can get more details and full message trace if you set first:
set DEBUG=ALL 
Then start
node-red -v flowfile.json

sb934, 2020-08-27:
Alright Mika, I shifted to Node-JS SDK for OPC-UA and it worked perfectly fine with Node-red-contrib-opcua...

Meanwhile, I have been facing issues while connecting to a commercial OPC-UA server - Kaiser-Raman. I'll open a new issue for the same.

closed 2020-08-27 by sb934