mikakaraila/node-red-contrib-opcua #188: Subscribe returns repeated values

JoeZeng133, 2020-08-11:
for some reason, when I subscribe to a node, OPC UA Client will return the same object multiple times at the published interval (see below).
[image]

mikakaraila, 2020-08-11:
Can you check on server with UaExpert if source timestamp is updated

JoeZeng133, 2020-08-11:
Yes, both source and server timestamps are changing
[image]
[image]

mikakaraila, 2020-08-11:
Then it causes update even value is not changing

JoeZeng133, 2020-08-12:
so is this a problem with the server?

mikakaraila, 2020-08-12:
OPC UA standard says that in the subscription value change or status code / server or source timestamp change will be enough that server will publish new update.

JoeZeng133, 2020-08-12:
I ran verbose logging and found that in the logs (below) there are 2 messages, 

12 Aug 09:32:21 - [info] [OpcUa-Client:73ca5013.a694f]  Status-Code:Good#InfoTypeDataValue|Overflow (0x0000480)
12 Aug 09:32:24 - [info] [OpcUa-Client:73ca5013.a694f] Client status: active subscribed
12 Aug 09:32:24 - [info] [OpcUa-Client:73ca5013.a694f] ns=2;s=Simulation Examples.Functions.Ramp1 value has changed to 79
12 Aug 09:32:24 - [info] [OpcUa-Client:73ca5013.a694f] { /* DataValue */
   value: Variant(Scalar<Int32>, value: 79)
   statusCode:      Good#InfoTypeDataValue|Overflow (0x0000480)
   serverTimestamp: 2020-08-12T14:32:23.661Z $ 874.100.000
   sourceTimestamp: 2020-08-12T14:32:23.661Z $ 874.100.000
}
12 Aug 09:32:24 - [info] [OpcUa-Client:73ca5013.a694f]  Status-Code:Good#InfoTypeDataValue|Overflow (0x0000480)
12 Aug 09:32:24 - [info] [OpcUa-Client:73ca5013.a694f] Client status: active subscribed
12 Aug 09:32:24 - [info] [OpcUa-Client:73ca5013.a694f] ns=2;s=Simulation Examples.Functions.Ramp1 value has changed to 91
12 Aug 09:32:24 - [info] [OpcUa-Client:73ca5013.a694f] { /* DataValue */
   value: Variant(Scalar<Int32>, value: 91)
   statusCode:      Good#InfoTypeDataValue|Overflow (0x0000480)
   serverTimestamp: 2020-08-12T14:32:24.166Z $ 559.300.000
   sourceTimestamp: 2020-08-12T14:32:24.166Z $ 559.300.000
}

but in the debug output from node-red it shows the last message repeated twice.

[image]

JoeZeng133, 2020-08-12:
this seems to me a race condition

JoeZeng133, 2020-08-12:
just a quick fix I found: I changed node.send(msg) to node.send(dataValue) 
[image]

mikakaraila, 2020-08-12:
Ok thanks have to check if someone needs whole message not just value. Topic is needed perhaps to save data to database.

mikakaraila, 2020-08-12:
If you have debug node connect it shows msg second time as verbose above shows it first

JoeZeng133, 2020-08-13:
> If you have debug node connected it shows msg second time as verbose above shows it first

I don’t understand your reply. This should be a bug instead of a question because
1. Each “on” event updates the same msg object concurrently, that is a race condition 
2. If I use dataValue, which is not shared between events, to send to next nodes, then no repeated values are spot in the debug console 

 If I had a misunderstanding, please correct me

mikakaraila, 2020-08-13:
Neither do I understand this completely.
Can you provide simple example flow and instructions to repeat this?

JoeZeng133, 2020-08-13:
this is the flow I am using and the inject configuration
[image]
[image]

initially, I was getting message that has repeated messages
[image]

and then I changed the following code
https://github.com/mikakaraila/node-red-contrib-opcua/blob/c8ba57e0db11d8f2437eed0423f641833603ab0d/opcua/102-opcuaclient.js#L1018-L1047

to this

          monitoredItem.on("changed", function (dataValue) {
          let msgToSend = {};

          set_node_status_to("active subscribed");
          verbose_log(msg.topic + " value has changed to " + dataValue.value.value);
          verbose_log(dataValue.toString());
          if (dataValue.statusCode === opcua.StatusCodes.Good) {
            verbose_log("\tStatus-Code:" + (dataValue.statusCode.toString(16)).green.bold);
          } else {
            verbose_log("\tStatus-Code:" + dataValue.statusCode.toString(16));
          }

          msgToSend.statusCode = dataValue.statusCode;
          msgToSend.topic = msg.topic;

          // Check if timestamps exists otherwise simulate them
          if (dataValue.serverTimestamp != null) {
            msgToSend.serverTimestamp = dataValue.serverTimestamp;
            msgToSend.serverPicoseconds = dataValue.serverPicoseconds;
          } else {
            msgToSend.serverTimestamp = new Date().getTime();;
            msgToSend.serverPicoseconds = 0;
          }

          if (dataValue.sourceTimestamp != null) {
            msgToSend.sourceTimestamp = dataValue.sourceTimestamp;
            msgToSend.sourcePicoseconds = dataValue.sourcePicoseconds;
          } else {
            msgToSend.sourceTimestamp = new Date().getTime();;
            msgToSend.sourcePicoseconds = 0;
          }

          msgToSend.payload = dataValue.value.value;
          node.send(msgToSend);
        });

and then the messages look like this

[image]

basically, in the original code, msg is shared between threads, so when two "changed" event fires, they can write to it at the same time and cause a race condition.

mikakaraila, 2020-08-14:
Cannot repeat problem on my computer, only the first value comes twice then normally:
[image]

JoeZeng133, 2020-08-14:
interesting, I am not sure why yours is normal after the 1 message, but mine is pretty consistent across several machines. One thing you can try is to make publishing interval longer such that more events get published at one time which increase the likelihood of race conditions.

JoeZeng133, 2020-08-14:
also, I just noticed node-red changes messages sending to be asynchronous starting from v1.0 last year. If yours is an older version, then race condition won't exist.
https://nodered.org/blog/2019/08/16/going-async

mikakaraila, 2020-08-14:
Hmm, I have these installed:
C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua>node --version
v12.16.2

C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua>npm --version
6.14.4

C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua>node-red --version
14 Aug 16:07:36 - [info]

Welcome to Node-RED
===================

14 Aug 16:07:36 - [info] Node-RED version: v1.0.6
14 Aug 16:07:36 - [info] Node.js  version: v12.16.2
14 Aug 16:07:36 - [info] Windows_NT 10.0.17763 x64 LE
14 Aug 16:07:37 - [info] Loading palette nodes

JoeZeng133, 2020-08-14:
I just noticed in your example, the 3rd and 4th messages get published at different times. My guess for your result is that you have a publishing interval of 1 second and the counter increased by 1 each second, so the server might only fire 1 event each time.
Can you increase the publishing interval to, let's say, 5 seconds so that the server fire multiple events each time and see if you receive repeated messages.

mikakaraila, 2020-08-15:
What about changing code this way:
node.send(msg.clone());

JoeZeng133, 2020-08-15:
were you able to repeat the problem on your end? node.send(msg.clone()) is done by default in node-red after v1.0, so I doubt it's going to work.
https://nodered.org/blog/2019/09/13/cloning-messages

mikakaraila, 2020-08-17:
Just testing, now I got double values.

JoeZeng133, 2020-08-19:
I am very confident it is a race condition. See the code I provided earlier, where I use a temporary msgToSend variable instead of the msg variable, which is shared between events. If it were not a race condition, using a temporary variable would still lead to repeated values.

JoeZeng133, 2020-08-24:
We’re you able to solve the issue using the code I provided? Let me know if you need help.

mikakaraila, 2020-08-24:
I tested more and as I wrote client will get correct amount debug outputs.
If server makes publish 3 times/second I will get 3 messages.
If server makes publish 5 times/second I will get 5 messages.
I still cannot see race condition...

JoeZeng133, 2020-08-24:
so did you get 3/5 repeated messages? In one of your comments, you said you get repeated values

mikakaraila, 2020-08-24:
Perhaps my comment was unclear, but for 3 publish == 3 messages, 5 publish == 5 messages.
Works exactly as it should.

JoeZeng133, 2020-08-24:
OK, I am trying to understand the setup you have. Were you saying you change the publishing interval to 3 times/second? Or you make the tags in the server change 3 times/second? In the first case you will get correct values while in the 2nd case you will get repeated values

Here is what I use for the experiment. I use a Kepware OPC server and the Ramp2 tag which changes every 300ms.
[image]

then I subscribe to this tag with a publishing interval of 3 seconds, queue size of 4 and sampling interval of 500 ms. Then I get 4 identical messages every 3 seconds.

Then I change my code to avoid the race condition, I get 4 messages that are correct.

Let me know if this is clear for you and whether you are able to reproduce this on your side.

mikakaraila, 2020-08-24:
I got correct values every time, no extra values.

JoeZeng133, 2020-08-24:
would you mind sharing your configuration with me? I want to see if there is something I did wrong on my side.

mikakaraila, 2020-09-02:
My friend pointed out this, msg.payload will be reused and your code fixes it by creating new object by this line:
let msgToSend = {};

It is not "race" condition, same object is reused and overwritten.
I will fix my code, sorry that I didn´t understant this immediately.

JoeZeng133, 2020-09-02:
hi mika, this is exactly the definition of race condition: two memory operations conflict if they access the same location and at least one of them is a write operation. Anyway, I am glad this is to be fixed.

reference:  

Adve, Sarita & Hill, Mark & Miller, Barton & H. B. Netzer, Robert. (1991). Detecting Data Races on Weak Memory Systems. ACM SIGARCH Computer Architecture News. 19. 234-243. 10.1109/ISCA.1991.1021616.

mikakaraila, 2020-09-02:
OK now new version available. There were actually two places to fix same case.

closed 2020-09-24 by mikakaraila