mikakaraila/node-red-contrib-opcua #189: samplingInterval and queueSize from msg

JoeZeng133, 2020-08-12:
I notcied that the samplintInterval is configured using msg.payload, would make more sense to use something like msg.samplingInterval or msg.payload.samplingInterval

https://github.com/mikakaraila/node-red-contrib-opcua/blob/c8ba57e0db11d8f2437eed0423f641833603ab0d/opcua/102-opcuaclient.js#L978

the queueSize is 10 all the time, would be nice to have it set up by the user, e.g., using msg.queueSize or msg.payload.queueSize

https://github.com/mikakaraila/node-red-contrib-opcua/blob/c8ba57e0db11d8f2437eed0423f641833603ab0d/opcua/102-opcuaclient.js#L1002

mikakaraila, 2020-08-13:
Good idea, I will add this change to next version.

mikakaraila, 2020-08-14:
Added:
```javascript
 if (msg.payload && msg.payload.interval && Number(msg.payload.interval > 100)) {
          interval = convertAndCheckInterval(msg.payload.interval);
        }
        if (msg.payload && msg.payload.queueSize && Number(msg.payload.queueSize > 0)) {
          queueSize = msg.payload.queueSize;
        }
```

JoeZeng133, 2020-08-14:
cool, you also might want to consider adding it to this function
https://github.com/mikakaraila/node-red-contrib-opcua/blob/7f62bc9df94fb0c1bd1cb5318142eb83bf175c57/opcua/102-opcuaclient.js#L1070

mikakaraila, 2020-08-14:
OK, no problem.

mikakaraila, 2020-08-25:
Feature implemented.

closed 2020-08-25 by mikakaraila

mikakaraila, 2020-08-26:
NOTE: moved both to msg level: msg.interval and msg.queueSize