mikakaraila/node-red-contrib-opcua #190: Reading/Writing a lot of variables from OPC UA Server

Mikoga92, 2020-08-13:
Hello,

I'm absolutely new to this whole topic of node-red and not experienced in programming also. As I'm a student and working on a solution to read and write variables form an OPC UA Server into a PLC, my question is:

What ways are there to program the subscription or reading/writing of a lot of variables? Do I have to put in for every NodeID a new OPC UA Item-Node? Or can I write the NodeID's into an array or something?

Additionally, the question of performance. How can I improve the performance? I think the answer lies somewhere on the OPC UA Server and the communication with it, but I don't find it.

Thank you in advance! And sorry for my english as I'm not a native speaker.

mikakaraila, 2020-08-13:
1) Read configuration from the file
2) Convert line to msg that contains action subscription (most effective), use function node.
    Another actions that could be used is readmultiple but causes load to server if too frequent reads.

Remember that OPC UA Server capability can limit amount & subscription parameters (like publish).

Mikoga92, 2020-08-13:
Thank you for the fast answer. I'm really sorry but I don't understand. What configuration from which file do you mean?

[image]

The whole thing is as follows: I get a read command and a article-number from the plc into the opc ua server. These are subscribed via client node. If node-red observes that the read command is true, it makes a sql command with the article-number in it, to get data from an sql-database.
The data from the db is now to be written on the opc ua server.

On the picture are not that much variables, but afaik some customers want a few hundred of variables read. What do you recommend?

mikakaraila, 2020-08-17:
I have tested node-opcua based server/client with 10000 variables and UaExpert performance test updates values in ~300ms.
node-red can be bit slower but in principle bottle neck is most probably on database level depending more how transactions are used.

Mikoga92, 2020-08-18:
Well that is really strange. I'm giving a command from the PLC with a Bool variable. That variable is subscribed via opc-ua-client-node. If it is true, it will be set back to false and it is doing the database query and writing the given values back to the opc-ua-server. Node-RED measured time is aroun 5-60ms. Mostly around 8ms, so really low.

But when I'm measuring the time inside the PLC, from the command to get the variables from the database until they are written. Then it is always between 250-300ms. And when I'm adding more variables, let's say 25 on top of that, I have +3,3s until they are written on the PLC.

Are you using an external opc-ua-server? I'm using the one from the S7-1200 here. Gonna try it with arrays and also with the S7-1500...

mikakaraila, 2020-08-18:
I am using other servers and clients. They are not node-red based.
As I said database will produce some own overhead.
Another way to measure server performance is to use UaExpert. Just to make sure real performance without any other components.

Mikoga92, 2020-08-18:
[image]

I measured it with UaExpert in Performance View, on the picture are the results. Just 18 variables and it's already taking up to 350ms...

I'll try it with another PLC and then see whether there is a difference. I think the Siemens PLC is slow.

MattPaulissen, 2020-09-18:
Your performance problem is on the PLC side.  The siemens OPC UA server on the 1200 and 1500 really cares about how you decide to grab the data.  It supports grabbing entire UDTs and Arrays, and in fact, those methods are several orders of magnitude faster.  You should stay away from single reads, but if you have to then you'd have to do registered reads.

Additionally, subscriptions are even more taxing.  There is also a limit to how many tags each PLC will support with subscriptions.  The 1510 and 1511 are ~1000 I believe.

Oxelmaster, 2021-04-28:
Hi, MattPaulissen mentioned in his post above me that you should stay away from single reads and instead use registred reads!
In the client node i can`t see a option for this reading/writing. 
Is it even possible with the node client to use registred read/write or is this feature not implemented?

mikakaraila, 2021-04-28:
It is easy to add registerNodes() and then just read as usually. BUT it depends on server if it will use/support it.

mikakaraila, 2021-04-28:
I published v0.2.118 client node contains now register action.

Here is sample flow, please test and report results.
[REGNODES.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/6394207/REGNODES.txt)

mikakaraila, 2021-04-28:
I will add after testing unregister and perhaps similar input msg handler as with readMultiple

Oxelmaster, 2021-04-28:
Thanks for your fast reply!
Will test it tomorrow and report here

mikakaraila, 2021-04-29:
I implemented also unregisterNodes() for client, but I will wait your results before publishing it.

Oxelmaster, 2021-04-29:
Just tested the Registered reading and its working without a problem on a S7-1500.
Can you please tell me what will happen if i try to register/subscribe the same variable multiple times? Will i get multiple register or subscriptions on the server or will the server ignor the requests?

mikakaraila, 2021-04-29:
You have to test it, I don´t have server that supports register/unregister nodes. If it is done smart way it should accept & register each even they are registered one by one. Same with unregister.

closed 2021-06-10 by mikakaraila

nohsanghyun, 2021-12-15:
> It is easy to add registerNodes() and then just read as usually. BUT it depends on server if it will use/support it.
> 
> Took just few minutes to add it: [image]

I checked in the official document,
If REGISTER is executed once, the effect is effective only in that session.
When the session ends, the REGISTER is also terminated.
But, in your example, there are two client nodes then also there are two sessions.

your opc-ua nodes are really useful to me.
Could you please reply,
thank you.

my sample :
[register_node.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7716168/register_node.txt)