mikakaraila/node-red-contrib-opcua #192: delete subscription doesn't allow re-subscribe

JoeZeng133, 2020-08-13:
when I inject deletesubscription, all monitored items stopped sending. But I am unable to subscribe them again and have to restart the whole session again. My guess is the terminated event wasn't fired when deleteSubscriptions is called, so I manually call subscription.terminate() after deleteSubscriptions and that solves the issues.

https://github.com/mikakaraila/node-red-contrib-opcua/blob/c8ba57e0db11d8f2437eed0423f641833603ab0d/opcua/102-opcuaclient.js#L1219-L1238

JoeZeng133, 2020-08-13:
btw, there is a typo "subscribtion", should be subscription

https://github.com/mikakaraila/node-red-contrib-opcua/blob/c8ba57e0db11d8f2437eed0423f641833603ab0d/opcua/102-opcuaclient.js#L500-L501

https://github.com/mikakaraila/node-red-contrib-opcua/blob/c8ba57e0db11d8f2437eed0423f641833603ab0d/opcua/102-opcuaclient.js#L1219-L1220

mikakaraila, 2020-08-13:
Good thanks, I will take these to next version

mikakaraila, 2020-08-14:
Fixed (also UI texts and other functions with typo), but code still allows old flows with wrong typo to work.

JoeZeng133, 2020-08-14:
cool

mikakaraila, 2020-08-14:
I expect this works now as it should have been working.

closed 2020-08-14 by mikakaraila