mikakaraila/node-red-contrib-opcua #193: node-opcua-client - 0.2.78 - Error: get-installed-path: module not found

apfelsaft87, 2020-08-13:
I am using node-opcua-client node of 0.2.78 and I always get this message:

"Error: get-installed-path: module not found "node-opcua-client" in path /usr/local/addons/redmatic/var/node_modules/node-red-contrib-opcua/node-opcua-client"

Can someone help me?

Thanks in advance!

mikakaraila, 2020-08-14:
Hmm, your path seems to be different than standard node-red installation.
Most probably it can cause error. Have tried to install this manually?
"get-installed-path": "^4.0.8",

apfelsaft87, 2020-08-14:
It is a redmatic installation on a CCU3.
I've tried to install with npm i get-installed-path but 

Unhandled rejection Error: EROFS: read-only file system, mkdir '/root/.npm'0846

┌───────────────────────────────────────────────────┐
│              npm update check failed              │
│        Try running with sudo or get access        │
│       to the local update config store via        │
│ sudo chown -R $USER:$(id -gn $USER) /root/.config │
└───────────────────────────────────────────────────┘
npm ERR! cb() never called!

npm ERR! This is an error with npm itself. Please report this error at:
npm ERR!     <https://github.com/npm/npm/issues>

apfelsaft87, 2020-08-14:
The opc-ua-browser node works fine but not opc-ua-client

mikakaraila, 2020-08-14:
Hmm, it is different also other ways. Did you try to install from command line with sudo ?

apfelsaft87, 2020-08-14:
Sure but 

`-sh: sudo: not found`

mikakaraila, 2020-08-14:
Read installation from the command line. 

Ich glaube kein probleme mit diese zu folgen:
https://github.com/rdmtc/RedMatic/wiki/Node-Installation#installation-via-npm-auf-der-kommandozeile

apfelsaft87, 2020-08-17:
Thank you but it also failed.

I've installed node-red-contrib-opcua and "get-installed-path": "^4.0.8" using that instruction. But still getting the message mentioned in the first posting.

mikakaraila, 2020-08-17:
Let´s start again, please run "node-red -v flowfile.json" to see verbose log.

mikakaraila, 2020-09-02:
This will be problematic to support as it is not "normal" node-red and needs something special... sorry.

Access29, 2020-09-20:
I got the same issue.

Would it be possible to make this node-pack work for RedMatic?

Thanks in advance for the info!

mikakaraila, 2020-09-20:
Can you install get-installed-path to normal npm global registry?

Problem lies in not standard npm installation path, get-installed-path works:

Get installation path where the given package is installed. **Works for globally and locally installed packages**

RedMatic seems to use this one:

/usr/local/addons/redmatic/var/node_modules/

Access29, 2020-09-26:
I tried to install the pack via the consol.
The issue is, that i have no raspian/depian os. So the consol commands are not working.

I am trying to use the opcua pack on a homematic IP CCU3. To install additional software on the CCU i need to use special addons which are only available for the ccu. Therefore i am not using the stanard NodeRed but instead RedMatic which is an addon for the ccu.

So it seams like i am not able to use the opc ua pack unless it can support different install directories.

mikakaraila, 2020-09-26:
Ok I can try to add that addon directory to search path

mikakaraila, 2020-09-30:
I have this change coming, can you test it before I will publish this?

```javascript
// Check finally local installation
    if (clientPkg == null) {
      clientPkg = installedPath.getInstalledPathSync('node-opcua-client', {
        paths: [
        path.join(__dirname, '..'),
        path.join(__dirname, '../..'),
        path.join(process.cwd(), './node_modules'),
        path.join(process.cwd(), '../node_modules'), // Linux installation needs this
        path.join(process.cwd(), '.node-red/node_modules'),
        "/usr/local/addons/redmatic/var/node_modules" // Red-matic package installation folder
        ],
      });
      verbose_log("Found locally installed path: " + clientPkg);
    }
```

mikakaraila, 2020-09-30:
I added this to 0.2.86 test with it.

Access29, 2020-09-30:
I tried to install it but it seams to get stuck somehow in the install screen like this:
[image]
[image]

mikakaraila, 2020-10-01:
Enough disk space?

mikakaraila, 2020-10-13:
Question did you install OPC DA as in first screenshot? There should be some installation log... it would help to solve this.

apfelsaft87, 2020-10-13:
I tested the latest release 0.2.87 but still get this error:

```
13.10.2020, 20:57:15node: OPC UA Client
msg : error
"Error: get-installed-path: module not found "node-opcua-client" in path /usr/local/addons/redmatic/var/node_modules/node-red-contrib-opcua/node-opcua-client"
```

apfelsaft87, 2020-10-13:
I reinstalled now but still same error message

mikakaraila, 2020-10-14:
Then this path that I put into the code is wrong:
/usr/local/addons/redmatic/var/node_modules/node-red-contrib-opcua/node-opcua-client

Can you check where node-opcua-client is installed on redmatic?

apfelsaft87, 2020-10-14:
Sure! How can I find it out?

mikakaraila, 2020-10-14:
Login to redmatic and use 
cd /var/local/addons/redmatic/var/node_modules
Then command
ls
and next folder to find out actual installation path.
Or is it under ?
/usr/local/addons/redmatic/var/node_modules/node-red-contrib-opcua/**node_modules/**node-opcua-client

apfelsaft87, 2020-10-14:
Okay I've tried:
`# cd /var/local/addons/redmatic/var/node_modules
-sh: cd: can't cd to /var/local/addons/redmatic/var/node_modules: No such file or directory`

And then:
`# cd /usr/local/addons/redmatic/var/node_modules/node-red-contrib-opcua/
# ls
LICENSE
Readme.md
function connect_opcua_client code.txt
images
node_modules
opcua
package.json
test.bat
`

mikakaraila, 2020-10-14:
OK it is under /usr not /var, I will fix path to code:
/usr/local/addons/redmatic/var/node_modules/node-red-contrib-opcua/node_modules

apfelsaft87, 2020-10-14:
Okay thank you!

As soon as I can update or I can test something for you just tell me.

Thank you for your great help!!!

motomoto-254, 2020-10-14:
please help, getting error:-
 Error: get-installed-path: module not found "node-opcua-client" in path C:\Users\hp\.node-red\node_modules\node-red-contrib-opcua\node-opcua-client

mikakaraila, 2020-10-14:
v0.2.88 contain fix.

closed 2020-10-14 by mikakaraila