mikakaraila/node-red-contrib-opcua #194: v0.2.78 Client node does not connect (S7-1500, Prosys OPC UA Server or Python OPC UA Server)

AndreasHeine, 2020-08-14:
Tested with:
S7-1500 OPC UA Server (latest firmware 2.6)
Prosys Simulation Server
Python OPC UA

Same result with all of them...

---

Node v12.14.0
Node-Red Version v1.1.3
node-red-contrib-opcua v0.2.78 (in v0.2.51 it works)

---
[image]

after flow restart

[image]

[image]

---

Error:
`06:25:31.558Z :message_builder_base          :153   Error   client1 cannot construct object with nodeID ns=0;i=44`

[{"id":"863ac4c3.d861a8","type":"OpcUa-Client","z":"80de491e.3f6f28","endpoint":"6242c95e.dead98","action":"read","deadbandtype":"a","deadbandvalue":1,"time":10,"timeUnit":"s","certificate":"n","localfile":"","securitymode":"None","securitypolicy":"None","name":"","x":580,"y":100,"wires":[[]]},{"id":"779ddd8.9af52a4","type":"OpcUa-Item","z":"80de491e.3f6f28","item":"ns=0;i=2267","datatype":"Byte","value":"","name":"","x":370,"y":100,"wires":[["863ac4c3.d861a8"]]},{"id":"e3ef3154.a7d2f8","type":"inject","z":"80de491e.3f6f28","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","x":190,"y":100,"wires":[["779ddd8.9af52a4"]]},{"id":"6242c95e.dead98","type":"OpcUa-Endpoint","z":"","endpoint":"opc.tcp://127.0.0.1:4840/","secpol":"None","secmode":"None","login":false}]

---

WireShark:
[image]

mikakaraila, 2020-08-14:
Run node-red -v flowfile.json
Client should verbose then re-connecting and other messages as (if) it tries to reconnect to server.

```javascript
 function create_opcua_client(callback) {
      node.client = null;
      verbose_warn("Create Client: " + JSON.stringify(connectionOption));
      try {
        node.client = opcua.OPCUAClient.create(connectionOption);
        node.client.on("connection_reestablished", function () {
          verbose_warn(" !!!!!!!!!!!!!!!!!!!!!!!!  CONNECTION RE-ESTABLISHED !!!!!!!!!!!!!!!!!!!");
        });
        node.client.on("backoff", function (attempt, delay) {
          verbose_warn("backoff  attempt #" + attempt + " retrying in " + delay / 1000.0 + " seconds");
        });
        node.client.on("start_reconnection", function () {
          verbose_warn(" !!!!!!!!!!!!!!!!!!!!!!!!  Starting Reconnection !!!!!!!!!!!!!!!!!!!");
        });
    
      }
      catch(err) {
        node_error("Cannot create client, check connection options: " + JSON.stringify(connectionOption));
      }
      items = [];
      node.items = items;
      set_node_status_to("create client");
      if (callback) {
        callback();
      }
    }
```

AndreasHeine, 2020-08-14:
C:\Users\andre>node-red -v flowfile.json
14 Aug 08:52:37 - [info]

Welcome to Node-RED
===================

14 Aug 08:52:37 - [info] Node-RED version: v1.1.3
14 Aug 08:52:37 - [info] Node.js  version: v12.14.0
14 Aug 08:52:37 - [info] Windows_NT 10.0.18363 x64 LE
14 Aug 08:52:37 - [info] Loading palette nodes
14 Aug 08:52:38 - [debug] logger Node logging turning on logging next 111 log points
14 Aug 08:52:38 - [info] logger Node Copyright 2020 Jaroslav Peter Prib
14 Aug 08:52:38 - [debug] logger Node test active
14 Aug 08:52:40 - [info] Dashboard version 2.21.0 started at /ui
14 Aug 08:52:40 - [info] Settings file  : \Users\andre\.node-red\settings.js
14 Aug 08:52:40 - [info] Context store  : 'default' [module=memory]
14 Aug 08:52:40 - [info] User directory : \Users\andre\.node-red
14 Aug 08:52:40 - [info] Server now running at http://127.0.0.1:1880/
14 Aug 08:52:40 - [info] Active project : opcua
14 Aug 08:52:40 - [info] Flows file     : \Users\andre\.node-red\projects\opcua\flow.json
14 Aug 08:52:40 - [info] Starting flows
14 Aug 08:52:40 - [info] [OpcUa-Client:863ac4c3.d861a8]         No certificate used.
14 Aug 08:52:40 - [info] [OpcUa-Client:863ac4c3.d861a8] Node-opcua is not globally installed, checking node-red-contrib-opcua next
14 Aug 08:52:40 - [info] [OpcUa-Client:863ac4c3.d861a8] Node-red-contrib-opcua is not globally installed, checking local folders next
14 Aug 08:52:41 - [info] [OpcUa-Client:863ac4c3.d861a8] Found locally installed path: C:\Users\andre\.node-red\node_modules\node-opcua-client
14 Aug 08:52:41 - [info] [OpcUa-Client:863ac4c3.d861a8] Client certificate not used!
14 Aug 08:52:41 - [info] [OpcUa-Client:863ac4c3.d861a8] Connection options:{"securityPolicy":"http://opcfoundation.org/UA/SecurityPolicy#None","securityMode":1,"clientName":"","endpoint_must_exist":false,"defaultSecureTokenLifetime":200000,"connectionStrategy":{"maxRetry":10,"initialDelay":5000,"maxDelay":30000},"keepSessionAlive":true}
14 Aug 08:52:41 - [info] [OpcUa-Client:863ac4c3.d861a8] EndPoint: {"id":"a0ba7822.7d3f1","type":"OpcUa-Endpoint","z":"","_closeCallbacks":[],"_inputCallback":null,"_inputCallbacks":null,"_asyncDelivery":true,"wires":[],"_wireCount":0,"credentials":{},"endpoint":"opc.tcp://LAPTOP-G69QK496:4840/OPCUA/SimulationServer","securityPolicy":"None","securityMode":"None","login":false,"user":null,"password":null}
14 Aug 08:52:41 - [info] [OpcUa-Client:863ac4c3.d861a8] UserIdentity: {"type":0}
14 Aug 08:52:41 - [warn] [OpcUa-Client:863ac4c3.d861a8] OpcUaClientNode: Create Client: {"securityPolicy":"http://opcfoundation.org/UA/SecurityPolicy#None","securityMode":1,"clientName":"","endpoint_must_exist":false,"defaultSecureTokenLifetime":200000,"connectionStrategy":{"maxRetry":10,"initialDelay":5000,"maxDelay":30000},"keepSessionAlive":true}
14 Aug 08:52:41 - [info] [OpcUa-Client:863ac4c3.d861a8] Client status: create client
14 Aug 08:52:41 - [info] [OpcUa-Client:863ac4c3.d861a8] Connecting to opc.tcp://LAPTOP-G69QK496:4840/OPCUA/SimulationServer
14 Aug 08:52:41 - [info] [OpcUa-Client:863ac4c3.d861a8] Client status: connecting
14 Aug 08:52:41 - [info] Started flows
14 Aug 08:52:47 - [warn] [OpcUa-Client:863ac4c3.d861a8] OpcUaClientNode: backoff  attempt #0 retrying in 5 seconds
14 Aug 08:52:54 - [warn] [OpcUa-Client:863ac4c3.d861a8] OpcUaClientNode: backoff  attempt #1 retrying in 10 seconds
14 Aug 08:53:06 - [warn] [OpcUa-Client:863ac4c3.d861a8] OpcUaClientNode: backoff  attempt #2 retrying in 20 seconds

mikakaraila, 2020-08-14:
I tested also and client reconnected correctly:
[image]

AndreasHeine, 2020-08-14:
[image]

mikakaraila, 2020-08-14:
NOTE: maxRetry is 10 if break is long enough then client does not try anymore

AndreasHeine, 2020-08-14:
> I tested also and client reconnected correctly:

did you use different versions then me?

> NOTE: maxRetry is 10 if break is long enough then client does not try anymore

yeah never mind thats not what i am worried about ;)

AndreasHeine, 2020-08-14:
if i drop a server in node red and deploy it crashes...
```

14 Aug 09:02:41 - [red] Uncaught Exception:
14 Aug 09:02:41 - Error:
    at Object.assert (C:\Users\andre\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-assert\source\index.ts:14:21)
    at ReaderState.finish [as _finish] (C:\Users\andre\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-address-space\source\loader\load_nodeset2.ts:889:25)
    at ReaderState._on_finish (C:\Users\andre\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-xml2json\source\xml2json.ts:212:18)
    at ReaderState._on_endElement (C:\Users\andre\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-xml2json\source\xml2json.ts:265:18)
    at SaxLtx.<anonymous> (C:\Users\andre\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-xml2json\source\xml2json.ts:459:36)
    at SaxLtx.emit (events.js:210:5)
    at SaxLtx._handleTagOpening (C:\Users\andre\.node-red\node_modules\ltx\lib\parsers\ltx.js:40:12)
    at SaxLtx.write (C:\Users\andre\.node-red\node_modules\ltx\lib\parsers\ltx.js:159:18)
    at C:\Users\andre\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-xml2json\source\xml2json.ts:395:24
    at FSReqCallback.readFileAfterClose [as oncomplete] (internal/fs/read_file_context.js:61:3)

C:\Windows\system32>
```

mikakaraila, 2020-08-14:
Can you check which node-opcua you have? I am using mid version 2.9.1 latest one is 2.10.0

AndreasHeine, 2020-08-14:
C:\Users\andre\.node-red\node_modules\node-opcua\package.json

{
  "_from": "node-opcua@^2.8.1",
  "_id": "node-opcua@2.10.0",
  "_inBundle": false,
  "_integrity": "sha512-7om8or0lplOBoTF9rnkDVCKwng+zuWCheClyuoqoNa+u0w8TdfcDzcYSopG1nw7OFvpnLLSqZCeczFXjIDXLiQ==",
  "_location": "/node-opcua",
  "_phantomChildren": {},
  "_requested": {
    "type": "range",
    "registry": true,
    "raw": "node-opcua@^2.8.1",
    "name": "node-opcua",
    "escapedName": "node-opcua",
    "rawSpec": "^2.8.1",
    "saveSpec": null,
    "fetchSpec": "^2.8.1"
  },
  "_requiredBy": [
    "/node-red-contrib-opcua"
  ],
  "_resolved": "https://registry.npmjs.org/node-opcua/-/node-opcua-2.10.0.tgz",
  "_shasum": "a7aaa636725a32f92abdedc82811636849a07243",
  "_spec": "node-opcua@^2.8.1",
  "_where": "C:\\Users\\andre\\.node-red\\node_modules\\node-red-contrib-opcua",
  "author": {
    "name": "Etienne Rossignon"
  },
  "bugs": {
    "url": "https://github.com/node-opcua/node-opcua/issues"
  },
  "bundleDependencies": false,
  "dependencies": {
    "chalk": "^4.1.0",
    "node-opcua-address-space": "2.10.0",
    "node-opcua-address-space-for-conformance-testing": "2.10.0",
    "node-opcua-aggregates": "2.10.0",
    "node-opcua-assert": "2.10.0",
    "node-opcua-basic-types": "2.10.0",
    "node-opcua-binary-stream": "2.10.0",
    "node-opcua-certificate-manager": "2.10.0",
    "node-opcua-client": "2.10.0",
    "node-opcua-client-crawler": "2.10.0",
    "node-opcua-client-proxy": "2.10.0",
    "node-opcua-common": "2.10.0",
    "node-opcua-constants": "2.9.0",
    "node-opcua-crypto": "^1.3.5",
    "node-opcua-data-access": "2.10.0",
    "node-opcua-data-model": "2.10.0",
    "node-opcua-data-value": "2.10.0",
    "node-opcua-debug": "2.10.0",
    "node-opcua-enum": "2.10.0",
    "node-opcua-factory": "2.10.0",
    "node-opcua-hostname": "2.10.0",
    "node-opcua-nodeid": "2.10.0",
    "node-opcua-nodesets": "2.6.4",
    "node-opcua-numeric-range": "2.10.0",
    "node-opcua-packet-analyzer": "2.10.0",
    "node-opcua-secure-channel": "2.10.0",
    "node-opcua-server": "2.10.0",
    "node-opcua-server-discovery": "2.10.0",
    "node-opcua-service-browse": "2.10.0",
    "node-opcua-service-call": "2.10.0",
    "node-opcua-service-discovery": "2.10.0",
    "node-opcua-service-endpoints": "2.10.0",
    "node-opcua-service-filter": "2.10.0",
    "node-opcua-service-history": "2.10.0",
    "node-opcua-service-node-management": "2.10.0",
    "node-opcua-service-query": "2.10.0",
    "node-opcua-service-read": "2.10.0",
    "node-opcua-service-register-node": "2.10.0",
    "node-opcua-service-secure-channel": "2.10.0",
    "node-opcua-service-session": "2.10.0",
    "node-opcua-service-subscription": "2.10.0",
    "node-opcua-service-translate-browse-path": "2.10.0",
    "node-opcua-service-write": "2.10.0",
    "node-opcua-status-code": "2.10.0",
    "node-opcua-transport": "2.10.0",
    "node-opcua-types": "2.10.0",
    "node-opcua-utils": "2.10.0",
    "node-opcua-variant": "2.10.0",
    "node-opcua-vendor-diagnostic": "2.10.0",
    "semver": "^7.3.2"
  },
  "deprecated": false,
  "description": "pure nodejs OPCUA SDK - module ",
  "devDependencies": {
    "should": "^13.2.3"
  },
  "engines": {
    "node": ">=8.10"
  },
  "funding": {
    "url": "https://github.com/sponsors/erossignon"
  },
  "gitHead": "f1c467c28f8bef1cc04ab54c57a555bd4c95ee45",
  "homepage": "http://node-opcua.github.io/",
  "keywords": [
    "OPCUA",
    "opcua",
    "m2m",
    "iot",
    "opc ua",
    "internet of things"
  ],
  "license": "MIT",
  "main": "./dist/index.js",
  "name": "node-opcua",
  "repository": {
    "type": "git",
    "url": "git://github.com/node-opcua/node-opcua.git"
  },
  "scripts": {
    "build": "tsc -b",
    "clean": "node -e \"require('rimraf').sync('dist');\""
  },
  "types": "./dist/index.d.ts",
  "version": "2.10.0"
}

AndreasHeine, 2020-08-14:
fresh install by the way via node-red!

mikakaraila, 2020-08-14:
Yours is the latest one. I have to update, perhaps something is changed with nodesets (XML format) as reading seems to crash.

mikakaraila, 2020-08-14:
Something else works with the latest node-opcua@2.10.0

14 Aug 10:45:25 - [warn] [OpcUa-Server:8eadc6ab.6fb768] OpcUaServerNode: node set:C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\opcua\public\vendor\opc-foundation\xml\Opc.Ua.NodeSet2.xml,C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\opcua\public\vendor\opc-foundation\xml\Opc.ISA95.NodeSet2.xml
JS folder: C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\opcua\node_modules
CWD:  C:\Users\karaimi\.node-red\.node-red
14 Aug 10:45:25 - [warn] [OpcUa-Server:8eadc6ab.6fb768] OpcUaServerNode: create Server from XML ...
14 Aug 10:45:25 - [info] [OpcUa-Server:8eadc6ab.6fb768] Using server certificate C:\Users\karaimi\.node-red\node_modules\node-opcua-server\certificates\server_selfsigned_cert_2048.pem
14 Aug 10:45:25 - [info] [OpcUa-Server:8eadc6ab.6fb768] Server options:{"certificateFile":"C:\\Users\\karaimi\\.node-red\\node_modules\\node-opcua-server\\certificates\\server_selfsigned_cert_2048.pem","privateKeyFile":"C:\\Users\\karaimi\\.node-red\\node_modules\\node-opcua-server\\certificates\\PKI\\own\\private\\private_key.pem","port":53880,"maxAllowedSessionNumber":1000,"maxConnectionsPerEndpoint":10,"nodeset_filename":["C:\\Users\\karaimi\\.node-red\\node_modules\\node-red-contrib-opcua\\opcua\\public\\vendor\\opc-foundation\\xml\\Opc.Ua.NodeSet2.xml","C:\\Users\\karaimi\\.node-red\\node_modules\\node-red-contrib-opcua\\opcua\\public\\vendor\\opc-foundation\\xml\\Opc.ISA95.NodeSet2.xml"],"serverInfo":{"applicationName":{"text":"Node-RED OPCUA"}},"serverCapabilities":{"maxBrowseContinuationPoints":10,"maxHistoryContinuationPoints":10,"operationLimits":{"maxNodesPerBrowse":10,"maxNodesPerHistoryReadData":6,"maxNodesPerHistoryReadEvents":10,"maxNodesPerHistoryUpdateData":10,"maxNodesPerRead":10,"maxNodesPerWrite":10}},"isAuditing":false,"registerServerMethod":3,"buildInfo":{"productName":" OPC UA server for node-red","buildNumber":"0.2.78","buildDate":"2020-08-10T23:00:00"}}
14 Aug 10:45:25 - [warn] [OpcUa-Server:8eadc6ab.6fb768] OpcUaServerNode: init next...
14 Aug 10:45:25 - [info] [OpcUa-Server:8eadc6ab.6fb768]
registering server to :opc.tcp://H7Q8Q13:4840/UADiscovery
14 Aug 10:45:25 - [info] Started modified flows
14 Aug 10:45:25 - [warn] [OpcUa-Server:8eadc6ab.6fb768] OpcUaServerNode: Server add VendorName ...
14 Aug 10:45:25 - [warn] [OpcUa-Server:8eadc6ab.6fb768] OpcUaServerNode: Server add MyVariable2 ...
14 Aug 10:45:25 - [warn] [OpcUa-Server:8eadc6ab.6fb768] OpcUaServerNode: Server add FreeMemory ...
14 Aug 10:45:25 - [warn] [OpcUa-Server:8eadc6ab.6fb768] OpcUaServerNode: Server add Counter ...
14 Aug 10:45:25 - [info] [OpcUa-Server:8eadc6ab.6fb768] Next server start...
14 Aug 10:45:25 - [info] [OpcUa-Server:8eadc6ab.6fb768] server initialized
14 Aug 10:45:25 - [warn] [OpcUa-Server:8eadc6ab.6fb768] OpcUaServerNode: Server is now listening ... ( press CTRL+C to stop)
14 Aug 10:45:25 - [info] [OpcUa-Server:8eadc6ab.6fb768] Server endpointUrl(s): opc.tcp://H7Q8Q13.VSTAGE.CO:53880 securityMode: 1 securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#None
14 Aug 10:45:25 - [info] [OpcUa-Server:8eadc6ab.6fb768] Server endpointUrl(s): opc.tcp://H7Q8Q13.VSTAGE.CO:53880 securityMode: 2 securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
14 Aug 10:45:25 - [info] [OpcUa-Server:8eadc6ab.6fb768] Server endpointUrl(s): opc.tcp://H7Q8Q13.VSTAGE.CO:53880 securityMode: 2 securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic256
14 Aug 10:45:25 - [info] [OpcUa-Server:8eadc6ab.6fb768] Server endpointUrl(s): opc.tcp://H7Q8Q13.VSTAGE.CO:53880 securityMode: 2 securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
14 Aug 10:45:25 - [info] [OpcUa-Server:8eadc6ab.6fb768] Server endpointUrl(s): opc.tcp://H7Q8Q13.VSTAGE.CO:53880 securityMode: 3 securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
14 Aug 10:45:25 - [info] [OpcUa-Server:8eadc6ab.6fb768] Server endpointUrl(s): opc.tcp://H7Q8Q13.VSTAGE.CO:53880 securityMode: 3 securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic256
14 Aug 10:45:25 - [info] [OpcUa-Server:8eadc6ab.6fb768] Server endpointUrl(s): opc.tcp://H7Q8Q13.VSTAGE.CO:53880 securityMode: 3 securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
contacting discovery server backoff  opc.tcp://H7Q8Q13.VSTAGE.CO:4840  attempt # 0  retrying in  2  seconds
contacting discovery server backoff  opc.tcp://H7Q8Q13.VSTAGE.CO:4840  attempt # 1  retrying in  4  seconds
contacting discovery server backoff  opc.tcp://H7Q8Q13.VSTAGE.CO:4840  attempt # 2  retrying in  8  seconds

AndreasHeine, 2020-08-14:
reinstalled node (lts version) npm v6.14.7 node v12.18.3
reinstalled node-red v1.1.3
installed node-red-contrib-opcua v0.2.78 (from node-red)

client still dont want to connect...
server somehow dont crash anymore!

any ideas? 
in "node-red-contrib-opcua v0.2.51" everything was fine i used it a lot, no issues at all!
[image]
[image]

AndreasHeine, 2020-08-14:
dont know exacly what it was uninstalled all node related stuff deleted modules manually, reboot, reinstalled all (node / node-red / opcua all in latest version) now it works .... i could bite in the table (-_-)° 
thx for the support!

closed 2020-08-14 by AndreasHeine