mikakaraila/node-red-contrib-opcua #196: unclear how subscription frequency works

mhechthz, 2020-08-25:
Hi,

I've now running the opc-ua client stable for certain months. Since the data frequency of my over 600 variables looked ok, I never checked this in detail.

Now I have a test setting with the free Prosys OPC-UA server and see the following:

If I inject thw following to the client node:
```
subs = {
       'topic': "ns=3;s=Sawtooth",
       'datatype':"Float",
       'payload': 2000, // subscription frequency
       'name': "Sawtooth"
}
return subs;
```
and set the interval in the client node to `500 ms`, then I get the sawtooth value every 500 ms. If I set `'payload':500` and the interval to `2000 ms`, then I get every 2 seconds 4 equal values from sawtooth with identical timestamps. This is not the bahaviour I expected to get.

Is this ok?

mikakaraila, 2020-08-25:
Sorry, there is code change. payload is not used anymore.
Use msg.payload.interval and msg.payload.queueSize 

        var interval = 100; // Set as default if no payload
        var queueSize = 10;
        if (msg.payload && msg.payload.interval && Number(msg.payload.interval > 100)) {
          interval = convertAndCheckInterval(msg.payload.interval);
        }
        if (msg.payload && msg.payload.queueSize && Number(msg.payload.queueSize > 0)) {
          queueSize = msg.payload.queueSize;
        }

mikakaraila, 2020-08-25:
And of course if server publish value with 500ms resolution you will get values 4 times in 2sec.
As server timestamp is updated or value has changed.

mhechthz, 2020-08-26:
> 
> 
> And of course if server publish value with 500ms resolution you will get values 4 times in 2sec.
> As server timestamp is updated or value has changed.

Hmmm ... the server changes the value every 200 ms. So if I set the interval in the client node to 2000 ms and the payload for the subscription (wich still seems to work in my version) to 500 ms, I expect to get 4 different values every 2000 ms. Is this no good idea?

mhechthz, 2020-08-26:
> Sorry, there is code change. payload is not used anymore.
> There was feature request: #189

So you made a change, that is not backward compatible??

I had prefered to have `msg.interval` in parallel to `msg.payload`, but not changing payload from a number to an object.

I currently have version 0.2.58 - is this already active here?

Thank you for your patience ;-)

mikakaraila, 2020-08-26:
Good point, I will move those to msg level.

mikakaraila, 2020-08-26:
New version available with msg.interval & msg.queueSize

mikakaraila, 2020-08-26:
From this link you can look how subscription works:
http://documentation.unified-automation.com/uasdkc/1.4.2/html/L2UaSubscription.html

mhechthz, 2020-08-27:
> 
> 
> New version available with msg.interval & msg.queueSize

Is there a listing of the currently exisiting parameters and how they work?

mikakaraila, 2020-08-27:
I will update readme and add there subscription parameters and monitoredItem parameters that can be used to tune communitaion.

mhechthz, 2020-08-29:
> > And of course if server publish value with 500ms resolution you will get values 4 times in 2sec.
> > As server timestamp is updated or value has changed.
> 
> Hmmm ... the server changes the value every 200 ms. So if I set the interval in the client node to 2000 ms and the payload for the subscription (wich still seems to work in my version) to 500 ms, I expect to get 4 different values every 2000 ms. Is this no good idea? What I get are 4 identical values (even the source timestamp).
> 
> [Settings for Prosys OPC-UA server](https://pasteboard.co/Jo7dQq3.jpg)

Hi Mika. This is still unanswered.

mikakaraila, 2020-08-30:
OK, interval == samplingInterval not the publish interval. Server will decide publishing.
Client cannot effect to it, it can request it but final decision is inside server.
Did you look this one?
http://documentation.unified-automation.com/uasdkc/1.4.2/html/L2UaSubscription.html

Monitoring can be used to filter values, but on the right server will do publish.

mhechthz, 2020-08-31:
Hmmm ... I'm still confused. If I get more values or less values then demanded in the subscription I would be fine, as the server has the final decision.

BUT:

If I set the interval value in the client node to 2 sec and the subscription interval to 500 ms I get 4 values every 2 seconds which is ok. But **these 4 values are totally identical** - also timestamps (!!!) - even if I set the change frequency on the OPC-UA sever to 200 ms, i.e. there are definitely more frequent updates on the server. I absolutely expect to get **4 different values**.

I think the link doesn't explain this. I would swear that the server log indicates, that all the 4 values sent are different  

So do I understand something wrong, or is there a bug in the node?

What I expect to get from Prosys simulation server (the only I have that I can control myself):
opc-ua node interval 1 sec - value interval (payload) 0.5 sec --> every second two different values
opc-ua node interval 1 sec - value interval (payload) 2 sec --> every two seconds one value

What I get from Prosys simulation server  (the only I have that I can control myself):
opc-ua node interval 1 sec - value interval (payload) 0.5 sec --> every second two identical values (value and timestamps)
opc-ua node interval 1 sec - value interval (payload) 2 sec --> every second sometimes one value, sometimes two identical values

mhechthz, 2020-08-31:
It's now even more confusing.

I just installed the new version of your software. Now payload doesn't work any longer. What I get now is - with interval 2 sec in the client node and 200 ms updates on the server - **10 identical values every 2 sec**!!

Setting 'interval' in the message on 1 sec, **I get every 2 sec 3 identical values**.

Nothing is as expected.

mikakaraila, 2020-09-01:
Hmm, I will do some testing. I can modify code to accept still old payload as interval.

mhechthz, 2020-09-01:
Ok, but it is probably not the problem of 'payload' or 'interval'. Since the subscription seems to do - in some way - the right thing according to the number of values transmitted. What is - in my opinion - wrong, is that n times totally **identical** values are transmitted and not the **different** values that are available at the OPC-UA server.

So I think either the server sends n times the same value - this would be really strange - or the OPC-UA-Client-Node fails to distinguish the n transmitted different values and takes at each notification the same (probably the first) value.

mikakaraila, 2020-09-01:
I have used ns=3;s=Counter and value is changing as expected. 

[image]

mikakaraila, 2020-09-01:
Welcome to Node-RED
===================

1 Sep 13:14:26 - [info] Node-RED version: v1.0.6
1 Sep 13:14:26 - [info] Node.js  version: v12.16.2
1 Sep 13:14:26 - [info] Windows_NT 10.0.17763 x64 LE

mhechthz, 2020-09-01:
Hi Mika, 

I got meanwhile this information: https://forum.prosysopc.com/forum/opc-ua-simulation-server/problems-with-node-red-contrib-opcua/#p4959 

My flow:
`[
    {
        "id": "69371cfb.c00f34",
        "type": "OpcUa-Client",
        "z": "49ef4b6b.5f3184",
        "endpoint": "a2742222.82122",
        "action": "subscribe",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": "5000",
        "timeUnit": "ms",
        "certificate": "n",
        "localfile": "",
        "name": "",
        "x": 1180,
        "y": 1000,
        "wires": [
            [
                "89cf021a.db6ad"
            ]
        ]
    },
    {
        "id": "edcac04d.256b9",
        "type": "inject",
        "z": "49ef4b6b.5f3184",
        "name": "",
        "topic": "ns=4;s=Sawtooth/4",
        "payload": "",
        "payloadType": "date",
        "repeat": "",
        "crontab": "",
        "once": true,
        "onceDelay": "1",
        "x": 710,
        "y": 1000,
        "wires": [
            [
                "62edca0e.6d8724"
            ]
        ]
    },
    {
        "id": "89cf021a.db6ad",
        "type": "debug",
        "z": "49ef4b6b.5f3184",
        "name": "",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "x": 1410,
        "y": 1000,
        "wires": []
    },
    {
        "id": "62edca0e.6d8724",
        "type": "function",
        "z": "49ef4b6b.5f3184",
        "name": "subscribe vars",
        "func": "subs = {\n       'topic': \"ns=3;s=Sinusoid\",\n       'datatype':\"Float\",\n       'interval': 1000, // subscription frequency\n       'name': \"Sinusoid\",\n       //'deadbandtype':'a',\n       //'deadbandvalue':0.2\n}\n\nreturn subs;",
        "outputs": 1,
        "noerr": 0,
        "x": 940,
        "y": 1000,
        "wires": [
            [
                "69371cfb.c00f34"
            ]
        ]
    },
    {
        "id": "a2742222.82122",
        "type": "OpcUa-Endpoint",
        "z": "",
        "endpoint": "opc.tcp://TTSM02:53530/OPCUA/SimulationServer",
        "secpol": "Basic256Sha256",
        "secmode": "SignAndEncrypt",
        "login": true
    }
]`

with update frequency of 200 ms at the server results in this:

[image]

mikakaraila, 2020-09-02:
Etienne helped me and there is actually error in my code.
See this one and last comment by me:
https://github.com/mikakaraila/node-red-contrib-opcua/issues/188

mikakaraila, 2020-09-02:
New version available. 
There was actually another error in handling msg.interval and I added old payload can be still used to set interval.
Now this should work, please test.

mhechthz, 2020-09-03:
I will test it as soon as possible, but I'm currently on vacation ...

mhechthz, 2020-09-09:
Ok, now I updated to the latest version. I didn't change anything for the flow. I now get the error "Invalid Endpoint" with the right (unchanged) address of the server :-(

With the older versions the connection is still ok, but the new version results in this error.

mikakaraila, 2020-09-09:
What is older version? Really old javascript based or some older 0.2.5x?

mhechthz, 2020-09-09:
Installed it in March this year - so not too old. I've running the node in two docker containers. The one where I test, I hat your last version running (after change from payload to interval - here the connection was ok).

mhechthz, 2020-09-09:
So finally I think the problem is solved. THANK YOU!

(The `deadbandtype` and `deadbandvalue` question has its own issue)

closed 2020-09-09 by mikakaraila