mikakaraila/node-red-contrib-opcua #197: issue is still unresolved.

youngkzy, 2020-08-25:
@mikakaraila,
I'm very sorry to inform you that the issue around connecting to a local OPC Server with NO SECURITY still does not work.

[image]

The error exists where it does not find the valid endpoint.
I am still having to use version 0.2.51 in order to connect to OPC servers using that version of node-opcua referenced by 0.2.51.

Kind Regards,

mikakaraila, 2020-08-25:
Strange as I have tested it has been working. Which server you are using?

youngkzy, 2020-08-27:
I uninstalled the old version and then installed the newer one.  I can tell you that I can graduate up the versions up from 0.2.51 to a point where they break.  I have tried 0.2.6?? but if I install a newer version say 0.2.7?? and then go back to 0.2.6?? it still is broke.  I am at the working version of 0.2.51 which is using a different version of node-opcua. 

 Kepware KepServer.V6 installed locally and the sample nodes installed on the server.   

I have node-red-contrib-opcua-iiot-opcua installed and it works no matter which version of node-red-contrib-opcua i install or uninstall.  The problem with node-red-contrib-opcua-iiot-opcua is that is is deprecated.

mikakaraila, 2020-08-27:
Both are using older node-opcua that is based on JavaScript.
I will try to find time to install KepServer and look what is wrong.

You have added new client to trusted clients?
Look page 7: https://www.kepware.com/getattachment/ccefc1a5-9b13-41e6-99d9-2b00cc85373e/opc-ua-client-server-easy-guide.pdf

youngkzy, 2020-09-08:
@mikakaraila, 
 This is my Trusted Servers tab.
I currently do not have the current version installed as it does not work.
Are you needing my information logs from this point of view?  this install?   Let me know. 

[image]

Tesla76, 2020-09-15:
Hello mikakaraila,
first of all, I would like to thank you for your excellent work.

I can confirm this issue. I have a similar constellation with a Windows server 2016, on which several instances of Node-Red (V1.0.3) and a Kepserver are running. I'm using node-red-contrib-iiot-opcua (V3.12.0) on one instance. Since this is unfortunately deprecated, I want to switch to node-red-contrib-opcua(V0.2.84). I didn't manage to connect to the Kepserver (Invalid Endpoint issue). Then i installed node-red-contrib-iiot-opcua  in another instance on which node-red-contrib-iiot-opcua was not installed and the connection works straight away. So it looks to me as if the two packages are not compatible.

mikakaraila, 2020-09-15:
You have same comments, but no debug log available?

I have had problems to install KepServer. Now I have separated vmware to test it.

Tesla76, 2020-09-15:
Sorry, i have no debug log because on the productive system debugging is off and i cant't restart it so easy to turn it on ...
Perhaps i can clone the productive system and try to debug the new instance if that helps !?

mikakaraila, 2020-09-15:
I installed KepServer and configured it and node-red flow works:

[image]

mikakaraila, 2020-09-15:
You must have something wrong in your environment and I cannot help without any further information.

Tesla76, 2020-09-15:
Ist the package node-red-contrib-iiot-opcua (V3.12.0) installed and in use also at your Node-Red enviroment ?
(I have already written about that issue)

"Then i installed node-red-contrib-iiot-opcua in another instance on which node-red-contrib-iiot-opcua was not installed and the connection works straight away. So it looks to me as if the two packages are not compatible"

mikakaraila, 2020-09-24:
If node-red-contrib-iiot-opcua is installed after node-red-contrib-opcua it can update some packages.
Code is only on npm not available github, so bit strange. Cannot place any issue for it.

Can you check this one? Instructions below:

Verbose log actually does not help at all.
It has something bit deeper, can you set first:

set MAXARRAYLENGTH=1000

set NODEOPCUADEBUG=TRACE-REQUEST-RESPONSE-PERF-STATS

node-red yourflow.json

Then I can see what is really happening on client protocol level.

youngkzy, 2020-09-25:
@Tesla76 and @mikakaraila, 
  Thank you for looking into this.  I have node-red-contrib-iiot-opcua installed on my laptop. And perhaps you have a point that it may interfere.  I am making a new installation on a VM for a system to deploy soon.  I have not loaded any OPCUA nodes as of yet.  I will give this a try as I am loading the system now.  I'll let you know as soon as I can test it.

Kindest Regards,
@youngkzy

napest-droid, 2020-10-13:
Dear Mikakaraila,

I would like to ask you, wheather there is any progres in thin issue - present still in version 0.2.87. I assume that the issue is associated with anonymous access. Unfortunately my server doesnt allow me to specify credentials, thus I have to use annonymous login, but the "invalid endpoint" error is received. 

Thanks in advance.

mikakaraila, 2020-10-13:
I am not able to reproduce this "invalid endpoint" and as long as nobody gives me good debug log I cannot solve this.

mikakaraila, 2020-10-13:
Can you check server endpoints?
https://github.com/node-opcua/node-opcua/issues/693

napest-droid, 2020-10-14:
I made what you mentioned above to prepare the debug log. I made a mistake in myflow.json file name. All flows disapeared, thus I checked the functionality of OPC client and everything works. Maybe it deleted some configuration nodes. I retrieved the name of correct flow.json file and again it works as well. 

set MAXARRAYLENGTH=1000
set NODEOPCUADEBUG=TRACE-REQUEST-RESPONSE-PERF-STATS
node-red yourflow.json

Now I'am not able to reprouce the error again. Sory for my bad english.

mikakaraila, 2020-10-14:
Look my comment on #202 

Node-opcua checks server endpoint will match exactly.

closed 2020-10-18 by mikakaraila

youngkzy, 2021-01-06:
Problem Resolved!  
Conflict with obsoleted "node-red-contrib-iiot-opcua"  

removed obsoleted above node(s).  Upgraded all on my test system Node-RED, Node.js, and its working OK.

Sorry it took so long.  Holiday and work schedule...