mikakaraila/node-red-contrib-opcua #199: Unable to connect to Kaiser-Raman

sb934, 2020-08-27:
Hi, I am trying to connect to an OPCUA-server using Basic256/Basic128Rsa15 security policy using Node-Red's inbuilt certificates and key. My flow works perfectly fine when working on a simulation server I've created using Node-JS SDK. 

I'm receiving the following error code with inbuilt credentials. 

- Error : client1 Security token has expired : tokenId 1

I replaced the certificate and key after going through this thread - S7-1500 secure conversation is closed with simple_client.js  #549 (https://github.com/node-opcua/node-opcua/issues/549)
And am getting the following error...

- Error : internal error expecting valid derivedKeys

Can someone suggest something to get this to work?

mikakaraila, 2020-09-02:
Can you provide debug log?
set DEBUG=all
node-red -v flow.json
or
set DEBUG=trace

Environment setting will provide low level message dump to console that can help to solve actual problem.

Shakti060, 2020-09-10:
Any updates on this @mikakaraila

mikakaraila, 2020-09-10:
I have asked debug log. I cannot do anything without some log...

mikakaraila, 2020-09-15:
Error message refers to node-opcua library, are you using node-opcua-client certificate files?

In your comment you refers to node-red certificate? It can cause this problem...

[image]

mikakaraila, 2020-09-24:
Verbose log actually does not help at all.
It has something bit deeper, can you set first:

set MAXARRAYLENGTH=1000

set NODEOPCUADEBUG=TRACE-REQUEST-RESPONSE-PERF-STATS

node-red yourflow.json

Then I can see what is really happening on client protocol level.

mikakaraila, 2020-10-13:
Can you check server endpoints?
https://github.com/node-opcua/node-opcua/issues/693

closed 2020-10-18 by mikakaraila