mikakaraila/node-red-contrib-opcua #202: Version 0.2.83 invalid endpoint

nearxos, 2020-09-08:
i am trying to setup an OPC UA client in Node-red and what i get is an invalid endpoint message

[image]
[image]

anyybody else experience the same issue?

mikakaraila, 2020-09-08:
Known issue, I need debug log.
What's server?

nearxos, 2020-09-10:
> Known issue, see #197 I need debug log.
> What's server?

Server is a raspberry pi with Codesys license on it. i am using it as a PLC

mikakaraila, 2020-09-10:
set DEBUG=all
node-red -v flowfile.json

I will need something to study this otherwise I cannot start to fix this.

mikakaraila, 2020-09-15:
Can you connect to same endpoint with UaExpert?

lbee11, 2020-09-15:
> Can you connect to same endpoint with UaExpert?

I have the same problem and Yes, I can connect with UaExpert without any issues.

Usually my OPC UA Client node will just sit there saying "Connecting" and then eventually fall to "Invalid Endpoint", but sometimes it jumps straight to "Invalid Endpoint". Here is my log file:

> 15 Sep 20:43:00 - [info] Starting flows
15 Sep 20:43:00 - [info] [OpcUa-Client:e1ae1439.3804a] Local certificate file client_selfsigned_cert_2048.pem
15 Sep 20:43:00 - [info] [OpcUa-Client:e1ae1439.3804a] Node-opcua is not globally installed, checking node-red-contrib-opcua next
15 Sep 20:43:00 - [info] [OpcUa-Client:e1ae1439.3804a] Node-red-contrib-opcua is not globally installed, checking local folders next
15 Sep 20:43:00 - [info] [OpcUa-Client:e1ae1439.3804a] Found locally installed path: /home/liam/.node-red/node_modules/node-opcua-client
15 Sep 20:43:00 - [info] [OpcUa-Client:e1ae1439.3804a] Using client certificate /home/liam/.node-red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_2048.pem
15 Sep 20:43:00 - [info] [OpcUa-Client:e1ae1439.3804a] Connection options:{"securityPolicy":"http://opcfoundation.org/UA/SecurityPolicy#Basic256","securityMode":3,"serverCertificate":{"type":"Buffer","data":[48,130,4,125,48,130,3,101,160,3,2,1,2,2,20,61,128,173,234,132,103,211,0,14,140,119,222,87,34,252,209,96,56,220,1,48,13,6,9,42,134,72,134,247,13,1,1,11,5,0,48,108,49,11,48,9,6,3,85,4,6,19,2,70,82,49,12,48,10,6,3,85,4,8,12,3,73,68,70,49,14,48,12,6,3,85,4,7,12,5,80,97,114,105,115,49,18,48,16,6,3,85,4,10,12,9,78,111,100,101,79,80,67,85,65,49,18,48,16,6,3,85,4,11,12,9,78,111,100,101,79,80,67,85,65,49,23,48,21,6,3,85,4,3,12,14,78,111,100,101,79,80,67,85,65,45,84,69,83,84,48,30,23,13,50,48,48,57,49,53,49,57,51,53,48,55,90,23,13,51,53,48,57,49,50,49,57,51,53,48,55,90,48,108,49,11,48,9,6,3,85,4,6,19,2,70,82,49,12,48,10,6,3,85,4,8,12,3,73,68,70,49,14,48,12,6,3,85,4,7,12,5,80,97,114,105,115,49,18,48,16,6,3,85,4,10,12,9,78,111,100,101,79,80,67,85,65,49,18,48,16,6,3,85,4,11,12,9,78,111,100,101,79,80,67,85,65,49,23,48,21,6,3,85,4,3,12,14,78,111,100,101,79,80,67,85,65,45,84,69,83,84,48,130,1,34,48,13,6,9,42,134,72,134,247,13,1,1,1,5,0,3,130,1,15,0,48,130,1,10,2,130,1,1,0,180,216,123,56,78,216,122,202,100,142,92,110,114,103,93,214,223,21,223,0,210,185,49,64,166,72,140,72,241,99,112,160,210,26,69,82,19,143,15,157,64,183,146,63,154,213,3,142,50,129,74,223,160,132,176,13,87,15,215,196,126,209,92,142,154,222,90,131,24,173,189,84,200,206,189,23,98,75,149,212,130,22,221,227,64,34,64,45,87,233,12,125,85,149,212,59,227,44,35,222,189,255,255,31,209,3,187,48,55,139,175,111,248,104,133,184,90,38,156,150,152,253,108,93,14,43,114,241,54,178,130,53,71,230,209,85,105,90,64,22,181,208,212,26,56,61,55,167,200,74,130,100,63,12,101,36,211,5,10,235,81,160,156,217,214,221,225,91,134,207,188,200,168,26,113,96,52,41,187,18,220,226,120,77,42,225,206,118,100,103,205,216,218,251,189,190,32,61,49,193,19,18,141,247,99,53,44,125,26,148,239,25,226,253,0,21,129,162,103,46,134,192,81,55,253,144,48,99,237,175,221,188,48,171,220,18,33,76,48,183,78,168,212,217,48,181,45,136,58,239,148,188,119,16,158,183,2,3,1,0,1,163,130,1,21,48,130,1,17,48,29,6,3,85,29,14,4,22,4,20,147,69,128,168,89,23,153,20,64,37,170,191,21,72,3,108,185,64,55,83,48,31,6,3,85,29,35,4,24,48,22,128,20,147,69,128,168,89,23,153,20,64,37,170,191,21,72,3,108,185,64,55,83,48,12,6,3,85,29,19,1,1,255,4,2,48,0,48,11,6,3,85,29,15,4,4,3,2,2,244,48,29,6,3,85,29,37,4,22,48,20,6,8,43,6,1,5,5,7,3,2,6,8,43,6,1,5,5,7,3,1,48,82,6,9,96,134,72,1,134,248,66,1,13,4,69,22,67,83,101,108,102,45,115,105,103,110,101,100,32,99,101,114,116,105,102,105,99,97,116,101,32,103,101,110,101,114,97,116,101,100,32,98,121,32,78,111,100,101,45,79,80,67,85,65,32,67,101,114,116,105,102,105,99,97,116,101,32,117,116,105,108,105,116,121,48,65,6,3,85,29,17,4,58,48,56,134,37,117,114,110,58,85,98,117,110,116,117,45,78,111,100,101,45,82,101,100,58,78,111,100,101,45,79,80,67,85,65,45,83,101,114,118,101,114,130,15,85,98,117,110,116,117,45,78,111,100,101,45,82,101,100,48,13,6,9,42,134,72,134,247,13,1,1,11,5,0,3,130,1,1,0,63,60,112,210,251,120,23,174,60,124,233,194,124,116,135,168,157,65,217,71,2,113,219,70,57,93,63,112,156,174,186,249,37,112,101,191,222,19,248,113,233,58,127,208,28,194,250,69,245,19,20,242,111,230,143,2,188,67,189,127,60,59,81,161,27,48,10,91,201,169,250,210,218,42,245,211,134,139,140,134,203,151,250,251,36,1,91,227,40,194,41,110,95,101,89,79,151,207,18,152,238,31,201,128,250,230,11,251,138,168,222,10,103,171,255,28,229,170,18,220,139,17,192,241,241,209,88,112,44,130,100,236,17,103,175,116,203,49,254,42,47,43,174,79,81,108,88,58,252,216,241,137,97,72,85,6,41,186,25,28,142,217,211,104,219,54,166,151,64,3,219,32,29,230,124,82,82,219,84,8,233,178,134,122,136,245,122,215,134,203,249,231,9,3,182,173,193,219,52,62,73,167,122,114,183,204,221,193,153,107,126,212,195,94,102,109,196,136,4,42,124,229,43,157,118,195,154,87,172,17,22,164,196,190,249,245,134,218,62,173,87,35,97,178,228,127,27,65,9,17,125,148,133,4,189,190]},"certificateFile":"/home/liam/.node-red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_2048.pem","privateKeyFile":"/home/liam/.node-red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem","clientName":"","endpoint_must_exist":false,"defaultSecureTokenLifetime":200000,"connectionStrategy":{"maxRetry":10,"initialDelay":5000,"maxDelay":30000},"keepSessionAlive":true}
[... cut ...]
15 Sep 20:45:35 - [warn] [OpcUa-Client:e1ae1439.3804a] OpcUaClientNode: backoff  attempt #6 retrying in 30 seconds
15 Sep 20:46:05 - [warn] [OpcUa-Client:e1ae1439.3804a] OpcUaClientNode: backoff  attempt #7 retrying in 30 seconds
15 Sep 20:46:35 - [warn] [OpcUa-Client:e1ae1439.3804a] OpcUaClientNode: backoff  attempt #8 retrying in 30 seconds
15 Sep 20:47:05 - [warn] [OpcUa-Client:e1ae1439.3804a] OpcUaClientNode: backoff  attempt #9 retrying in 30 seconds
15 Sep 20:47:35 - [info] [OpcUa-Client:e1ae1439.3804a] Client status: invalid endpoint ocp.tcp://192.168.100.150:4840

mikakaraila, 2020-09-16:
OK, client works as it should, have you checked that server has client certificate on trusted folder?

There has been discussion and one comment says that AnsiC based servers will put certificate to rejected.
https://github.com/node-opcua/node-opcua/issues/425

digital-lion, 2020-09-24:
I just posted an other issue, but now I have the same trouble with this node.
Background is: I wanted to test if issue #205 is only related to version 0.2.83 as reported. 

After updating from 0.2.83 to 0.2.84 I got also the "invalid Endpoint". The same flow was working before and is attached at issue #205.

[image]

mikakaraila, 2020-09-24:
Trace log actually does not help at all.
It has something bit deeper, can you set first:

set MAXARRAYLENGTH=1000

set NODEOPCUADEBUG=TRACE-REQUEST-RESPONSE-PERF-STATS

node-red yourflow.json

Then I can see what is really happening on client protocol level.

mikakaraila, 2020-10-13:
Here is another way to check server endpoints:
https://github.com/node-opcua/node-opcua/issues/693

closed 2020-10-18 by mikakaraila

FMauroy, 2020-10-19:
Hi,

I've been trying to use your OPC-UA connector for NodeRed but encounter the same issue (invalid endpoint). According to what I found in this thread and linked ones, the endpoint should match exactly what the server provides, is that right? It doesn't seem to help in my case. I use Prosys OPC Simulator 3, and did a copy/paste of the full endpoint URL. I checked that the Docker image Node-Red runs under can resolve the server name. I also have other images on the same system that can talk to the OPC server without issue.

Log file:

> 19 Oct 13:59:24 - [info] [OpcUa-Client:Client Browse] Client status: create client
19 Oct 13:59:24 - [info] [OpcUa-Client:Client Browse] Connecting to opc.tcp://edge:53530/OPCUA/SimulationServer
19 Oct 13:59:24 - [info] [OpcUa-Client:Client Browse] Client status: connecting
19 Oct 13:59:24 - [info] [OpcUa-Client:Client Browse] No client to connect...
19 Oct 13:59:24 - [info] [OpcUa-Client:Client Browse] Client status: invalid endpoint opc.tcp://edge:53530/OPCUA/SimulationServer

What else could I try?

Thanks a lot!
fred

FMauroy, 2020-10-19:
Me again. Just tried with a standalone install of NodeJS on my Windows laptop (to compare with my previous try running Node-RED in a Docker image), copy/pasted the flow into the new instance, and it worked on the first try.
So, there must be something off with my Docker installation, or the image is maybe running an older version of Node-RED or something.
Sorry for the bother ;-)
fred