mikakaraila/node-red-contrib-opcua #203: Adding additional information is no longer possible

mhechthz, 2020-09-09:
Hi Mika,

it must be a hard job to maintain such an important node. You do a very good job. I hope you don't lose your energy.

Up to the last node update, I used an additional custom value in the msg object to add the shortname (for later storage in data base) of a subscribed item:

```
subs = {
 'topic': "ns=3;s=Sinusoid",
 'datatype':"Float",
 'interval': 1000, // subscription frequency
 'name':'Sinusoid'
}

return subs;
```

The result was, that the OPC-UA Client node gave back the name together with all necessary values. The keeps me from parsing hte `topic` information to get the variable name:
```
{
  "topic":"ns=3;s=Sinusoid",
  "datatype":"Float",
  "payload":-6.211478258430595,
  "name":"Sinusoid",
  "_msgid":"9c0537dc.d77d78",
  "serverTimestamp":"2020-09-09T15:32:57.049Z",
  "serverPicoseconds":0,
  "sourceTimestamp":"2020-09-09T15:32:56.800Z",
  "sourcePicoseconds":0
}
```

After the last update, `name` cannot be used any longer, since there is a collision with the server name (I guess). For the older versions, it is no problem to change from `name` to `itemname` (e.g.).

**BUT**: For the new version, additional values are not available any longer but obviously are discarded.

So I'm forced to parse the topic, e.g. "ns=2;s=source.subspace.machine.itemname" or comparable if there is a more complex structure. 

Why did you remove the really helpful feature in the new version? 

OR

Can the old behaviour be added again?

mikakaraila, 2020-09-09:
I expect this change came as I changed subscription to create new object instead of using the same msg object again.
Have to decide how to copy object either by assign or JSON parse & stringify.

mhechthz, 2020-09-09:
Maybe you could use a dedicated `user` property to keep specific user-defined information.

mikakaraila, 2020-09-09:
msg object can contain also other properties defined by other users... not so easy to fulfill everybody´s wishes

mhechthz, 2020-09-10:
I understand. But if you currently throw away the old msg object, it is not necessary to care about name conflicts, is it?

mikakaraila, 2020-09-10:
I added code to clone original msg to msgToSend, I will publish new code. Please test!

mhechthz, 2020-09-11:
Ok, works. Mission accomplished :-)

Thank you.

closed 2020-09-11 by mikakaraila