mikakaraila/node-red-contrib-opcua #204: Status Code value = 0

henjoe, 2020-09-18:
Dear Mika,

I hope you find this message well and fine.
I asked before to add the "status" of the OPC tags on a "Subscribe" method. 
I have this OPC tags which it's quality is always "good" but I received this status code to be Zero (0).
[image]

Does zero value means "good"? what will be the value if it is bad or uncertain? 
Hoping for you guidance here.

Thank you very much.

With kind regards,
Henjoe

mikakaraila, 2020-09-20:
Yes, but it can contain optional info bits:

/** Good: No Error */

 static Good: ConstantStatusCode =  new ConstantStatusCode({ name:'Good', value: 0, description:'No Error' });

henjoe, 2020-09-20:
Thanks Mika, 

but on the payload object I'll show, there is no ConstantStatusCode property ? 
Is this not available on the subscribe method? 

With kind regards,
Henjoe

mikakaraila, 2020-09-20:
Look this the client node code:

```javascript
 if (dataValue.statusCode === opcua.StatusCodes.Good) {
            verbose_log("\tStatus-Code:" + (dataValue.statusCode.toString(16)).green.bold);
          } else {
            verbose_log("\tStatus-Code:" + dataValue.statusCode.toString(16));
          }
          
          msgToSend.statusCode = dataValue.statusCode;
          msgToSend.topic = msg.topic;
```

henjoe, 2020-09-20:
Hello Mika,

Can I input this code in a function node? 
Where should I put it? 

Never knew about this, this is cool.

With kind regards,
Henjoe

mikakaraila, 2020-09-20:
Here is simple example:

[image]

closed 2020-09-22 by mikakaraila

henjoe, 2020-09-25:
Dear Mika, 

I know this was already close, but I need your suggestions here.
I can get the String now on the statusCode, however there is a "(0x000)" string besides it. 

[image]

Is there any way to remove that string? only the "Good", "Bad" and "Uncertain" should be in my item. 
I am thingking of using the JS functions like string.replace? or string.contains? 

Thank you.

With kind regards,
Henjoe