mikakaraila/node-red-contrib-opcua #206: Error: Invalid variant arrayType: Scalar dataType: Boolean value:true

lfabrizio95, 2020-09-28:
I'm trying to call an opc ua method, but I get the following error:

"Error: Invalid variant arrayType: Scalar dataType: Boolean value:true"

I don't know if I'm doing something wrong, I have to admit, but calling the same method from OPC UA Expert it's actually working!

I'll attach some images, tell me if you need something else
[image]
[image]
[image]

[image]

mikakaraila, 2020-09-28:
Sorry but your node belongs to node-red-contrib-iiot-opcua that is not available at the moment. Only old package is still in the npm, but no github page. Klaus Landsdorf made it.

Not mine one... sorry.

lfabrizio95, 2020-09-28:
Thanks for the kind reply, the client is able to call methods in your package? I'm exploring it right now, but I can't see how (if it's possible)

mikakaraila, 2020-09-28:
I could implement it, needs just some time to do it. I have been using methods calls. Typescript API is good.

lfabrizio95, 2020-09-28:
That would be awesome!

mikakaraila, 2020-09-29:
I got simplified version implemented. How many input arguments you will need?
I will add some to node itself otherwise function node should be used.

lfabrizio95, 2020-09-30:
In this specific scenario, we have 9 input arguments (string and boolean)

lfabrizio95, 2020-09-30:
It's ok also with the function node if there is some documentation that we can check!

mikakaraila, 2020-09-30:
I have METHOD.json flow as an example (above). I will try to publish new version later today. Still some other work to do...

mikakaraila, 2020-09-30:
Example ok:

[image]

lfabrizio95, 2020-10-01:
We tried the new node, but it takes only 3 arguments maximum? Is it corrrect? 
In our example we have 9 input arguments

mikakaraila, 2020-10-01:
Look examples and use function node to set needed amount of input arguments.

closed 2020-10-01 by mikakaraila