mikakaraila/node-red-contrib-opcua #208: uncatchable error on read client

JoeZeng133, 2020-09-29:
trying to read a tag that does not exist on the OPC UA server. It logs the error to the debug console, but the catch node does not catch it. Below is the simple flow to generate this issue.
[image]

JoeZeng133, 2020-09-29:
btw, the error message comes from this line (if I comment it out, the error message disappears). hope it helps
https://github.com/mikakaraila/node-red-contrib-opcua/blob/5a2e86896150a8b931b0d2c0be5a7aed82e16c59/opcua/102-opcuaclient.js#L566

mikakaraila, 2020-09-29:
Yes, caused by dataValue === null then it will crash on accessing DataType[]. Fixed!

mikakaraila, 2020-09-30:
OK test with 0.2.86

EliasReuter, 2020-12-02:
Hi @mikakaraila and @JoeZeng133 

I am running a flow reading almost 500 variables on the version 0.2.89 of the node and still have the same problem, an its now logging "where" the problem occurs. It is Blank, compared to Joe's screenshot, where it describe that the problem comes from the "Read" function.
We are currently using the "Read Multiple" Function, but used to use the "Read" function. The problem occurred in both cases.
[image]

Is it possible that there is still an issue with this line that Joe found?
Is it possible that this could lead to restart every ~10/15mins of the node red service? (I can open a new issue regarding to this unexpected restarts if prefered, but mentioning this issue, cause it may be a possible cause)

Thank you!

mikakaraila, 2020-12-02:
Ok, I will take a look, but most probably not on this week. Bit busy at work...

EliasReuter, 2020-12-09:
Hi @mikakaraila 
Thank you for the support.

Just to add some more info, before this mentioned error shows up, this message appears:
[image]

(Not sure if there's a direct relation between these errors, but it might be useful for you in order to debug)

mikakaraila, 2020-12-09:
I check code and somehow it seems to relate to connection is lost and node.client is null. 

I need more info why connection is broken. Any messages on server side?

This issue has changed, error in now on different part of code, not in the original place as in the title.
Reading 500 variables will cause load on server side, subscribe is more efficient.

rcarreir, 2020-12-11:
Hi @mikakaraila !
I am having the same issue as @EliasReuter, but in my case there are just a few variables (less than 40) and also the "UA stucks at some point and only gets back to work again after reseting the whole flow. I can't understand what is the problem, but as the entire flow crashes my variables stops to upload to my server. I think in this case subscribe may not be the best option.

In parallel, while I was writing down this comment I found an other error in the node labeled as "invalid endpoint" and checking the server status in "UAExpert" I can see that the OPC server shuted down for some time and after geting up and working again, the node still gets stuck at that error and only goes back to active reading after reseting the flow. So it seems that it has no auto reset or auto reconnect afer a OPC server error.

[image]

[image]

Thanks for your support!

Regards,
Rodrigo

mikakaraila, 2020-12-11:
Etienne fixed most probably something related to this today

EliasReuter, 2020-12-11:
Hi @mikakaraila,

1) "I check code and somehow it seems to relate to connection is lost and node.client is null.
I need more info why connection is broken. Any messages on server side?"

We are connection to a Matrikon OPC UA Wrapper Server that is connecting to a SCADA system via OPC DA. 
Following, description and screenshots of what we are using: (+ log files)

Driver OIGatteway Wonderware
[image]

Driver Matrikon OPC US/DA
[image]

Servidor OPC
[image]

Log files from wrapper
[MatrikonOpcUaWrapper20201005-0040.log.zip](https://github.com/mikakaraila/node-red-contrib-opcua/files/5681035/MatrikonOpcUaWrapper20201005-0040.log.zip)

Hope this is useful information for you

2) "This issue has changed, error in now on different part of code, not in the original place as in the title."

Ok. Any other information you need us to provide you in order to find the bug?

3) "Reading 500 variables will cause load on server side, subscribe is more efficient."

Our application needs to get the value of these 500 variables in order to send them in jsons to other system with https, where 
 deterministic interval between every measure is needed. (Even if value is the same)
So we think subscription is not what would apply to our needs. So "Inject Interval + MultipleRead" seems to be the best 
 solution at the time for us. 
But we are open to suggestions.

Thank you!

mikakaraila, 2020-12-13:
I have a feeling that my client can disconnect if response time is longer than session timeout.
Can you check how much time from the inject (get all 500 variables) it will take time until client gets values?
Sometimes server cannot collect data and build message immediately as it will run also control actions.

EliasReuter, 2020-12-14:
Hi @mikakaraila,

Thank you for the Support!

**Reading method: READMULTIPLE**
~ 80% of the variables
- Inject Interval = 5 seconds
- Time between values = 4 to 6 seconds with inconsistencies up to 10 seconds (description)

**Reading method: READMULTIPLE**
~ 20% of the variables
- Inject Interval = 15 seconds
- Time between values = 4 to 6 seconds with inconsistencies up to 10 seconds (description)

What do you mean by "control actions"? Writing? 
We are only reading values. This Matrikon OPC UA Wrapper is dedicated to read values from node red

**Inconsistency problem**
I was thinking in opening a new issue for this after this progresses, but it seems to be related, after your question (but if prefered we can open an new issue an refer to this one)
We have an non-deterministic interval between measures as described above. 
Around 70% of the time the values are equal-distance, but then some "lakes" may appear, that are caused because the read (or read multiple) action is taking longer than "immediate" read. Even up to 10 to 20 seconds without response.

Here is an graph we made to show this behavior:
[image]
The points at the left is showing how there is an value every ~5 seconds , but then an lake of ~10 or ~20 seconds appears with no data (we check that this non-deterministic interval between values is caused by the delay of the read action in the opc ua client as comment. we located that problem surrounding the node with debug nodes)

Hope this is information is useful for you and even related to this issue. But if needed, we can open an new issue related to this (or once its solved, for future visitors).

Thank you, mika!

mikakaraila, 2020-12-14:
OK, my feeling was correct, client uses 10s session timeout:

        node.session.timeout = opcuaBasics.calc_milliseconds_by_time_and_unit(10, "s");

It should be 60s or even 5min so it will not disconnect and got o re-connect loop. Will you change it and test before I modify it?

EliasReuter, 2020-12-15:
Hi @mikakaraila,

I changed that line in the mentioned file, first to (60, "s") and then to (300, "s"), and restarted NodeRed Service (in order to make sure it takes the updated js). But I still get the same error.

[image]

Am I forgetting something to test this right?
What else can I test for you?

mikakaraila, 2020-12-16:
Ack, I will study bit more. It can be that timeout is set in wrong place.

mikakaraila, 2020-12-16:
Try this, same file and look line numbers:

[image]

EliasReuter, 2020-12-16:
Hi @mikakaraila,

If I copy those lines, "timeout" (line 139 in your image) is an undefined variable in my context. Client Node doesn't even shows connection status and following error shows up: (i guess its not working at all with this error)
[image]

So, I commented the line where timeout variable is assigned, leaving the following:
[image]

With this, same error of this issue appears after a while.
[image]

Where and how should I define that timeout variable? Are you using it somewhere else before or after that image?
If preferred, sent me the complete "102-opcuaclient.js" file and I'll replace it in my environment (I'll back up the original one), in order to test your fixes.

Thank you 👍

mikakaraila, 2020-12-16:
Just testing, library uses for some reason 60s timeout, trying to solve it now:
[image]

mikakaraila, 2020-12-16:
I expect Prosys UA Simulation server decides this. I request it to be 300s == 5min but server limits session timeout to 60s.
Rename .txt to .js and test again.

[102-opcuaclient.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/5702906/102-opcuaclient.txt)

EliasReuter, 2020-12-16:
Renamed the original file to "102-opcuaclient_old.js" and got this new "102-opcuaclient.js" you sent me.

After restarting node red service by "sudo systemctl restart nodered.service" same message appears:
[image]

mikakaraila, 2020-12-16:
Then the problem is not in the session timeout. I need more debug output. Can you get console output from the node-red?

EliasReuter, 2020-12-18:
Hi,
Following status appears now: 
[image]

Node Red's log files:

mikakaraila, 2020-12-18:
Log files missing...

EliasReuter, 2020-12-18:
[logNodeRedForOpc.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/5716076/logNodeRedForOpc.txt)
[logNodeRedForOpc1.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/5716078/logNodeRedForOpc1.txt)
[logNodeRedForOpc2.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/5716080/logNodeRedForOpc2.txt)
[logNodeRedForOpc3.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/5716081/logNodeRedForOpc3.txt)

EliasReuter, 2020-12-18:
[logNodeRedForOpcUa.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/5716174/logNodeRedForOpcUa.txt)

mikakaraila, 2020-12-19:
OK I will study these:

18 Dec 11:18:31 - [error] [OpcUa-Client:f5b707d4.9b7e1] Client node error on:
dispose when pendingTransactions is not empty

xxxxx  <<<<<< _on_message_received for unknown or timeout request  8                   ServiceFault BadSessionIdInvalid (0x80250000) 1524777

18 Dec 11:18:47 - [error] [OpcUa-Client:f5b707d4.9b7e1] Client node error on:
dispose when pendingTransactions is not empty
xxxxx  <<<<<< _on_message_received for unknown or timeout request  9                   ServiceFault BadSessionIdInvalid (0x80250000) 1524772

Then it stays:
dispose when pendingTransactions is not empty
dispose when pendingTransactions is not empty
dispose when pendingTransactions is not empty
dispose when pendingTransactions is not empty
dispose when pendingTransactions is not empty
dispose when pendingTransactions is not empty

mikakaraila, 2020-12-19:
Found one new one:
18 Dec 11:19:48 - [error] [OpcUa-Client:d7233715.316068] Client node error on:
ERROR !!! callback has thrown en error  Error
    at Object.assert (/home/ldc/.node-red/node_modules/node-opcua-assert/dist/index.js:13:21)
    at /home/ldc/.node-red/node_modules/node-opcua-client/dist/private/client_base_impl.js:414:45
    at ClientSecureChannelLayer.<anonymous> (/home/ldc/.node-red/node_modules/node-opcua-client/dist/private/client_base_impl.js:575:24)
    at Object.modified_callback [as callback] (/home/ldc/.node-red/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:953:31)
    at ClientSecureChannelLayer.cancelPendingTransactions (/home/ldc/.node-red/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:401:29)
    at ClientSecureChannelLayer.close (/home/ldc/.node-red/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:419:14)
    at OPCUAClientImpl.disconnect (/home/ldc/.node-red/node_modules/node-opcua-client/dist/private/client_base_impl.js:691:24)
    at OPCUAClientImpl.disconnect (/home/ldc/.node-red/node_modules/thenify/index.js:65:46)
    at OPCUAClientImpl.disconnect (/home/ldc/.node-red/node_modules/thenify/index.js:65:46)
    at close_opcua_client (/home/ldc/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:307:21)
18 Dec 11:19:48 - [error] [OpcUa-Client:d7233715.316068] Client node error on:

mikakaraila, 2020-12-19:
Client cannot disconnect before all transactions has been finished.

How are you storing data?

EliasReuter, 2020-12-21:
Hi @mikakaraila,

There is no storage in Node-Red.
In Node-Red, once I've got the values for the OPC UA Tags I build a new JSON (variable_id + value + timestamp) and send it to a web service via https from node red. So, there is no storage from Node Red Side (only a few global/flow variables with frequent dumps into files to avoid nulls and errors when NodeRed restarts as I mentioned before in this thread, maybe caused by the error of this issue).

There is no storage at the Server Side.
As I've mentioned, the Matrikon OPC UA Wrapper Server that is connecting to a SCADA system (Wonderware Archestra) via OPC DA and providing us with variables via OPC UA. 

I'm not sure if I answered your question with this. If not, what do you mean by "How are you storing data?".
How and why could storage strategies affect the performance of OPC UA connectivity/session?

mikakaraila, 2020-12-21:
I was just expecting if storing values takes long time or it is more than 10s it could causes problems if it blocks client.

Storing values to JSON files can cause something... couple of checks:
1) Check save value time if it is more than 10s
2) Check server session count with UaExpert server diagnostics will it incease?

After that we could perhaps modify flow to read values again after 10s values are saved. If interval is too fast then it could cause queue of read values and later crash flow.

EliasReuter, 2020-12-22:
Hi @mikakaraila,

The flow is not saving into JSON Files, just building the object in a function node and then sending it via https, as I described. No storing action but those context variables from node red.

I edited every inject node with "ReadMultiple" action to 20seconds, to test what you are saying, but the messages appears either way. So maybe fast interval is not the case?

It's not so easy for me at the moment to install UaExpert Server Diagnostics in that Computer, but I can give it a try if it's really needed.

Should I still use your .js file? or should i get back to the original?

mikakaraila, 2020-12-22:
Follow first CurrentSessionCount:

[image]

I will do some other investigation what will happen with read & session timeout with different parameters.

mikakaraila, 2020-12-22:
OK, I found two places and added error handling. I got error repeated by deploying flow again.

Seems to continue, I will run more tests, but at least not crashing anymore:
22 Dec 22:31:19 - [warn] [OpcUa-Item:c7d7e4aa.ee8468] OpcUaClientNode: Setting value to
dispose when pendingTransactions is not empty
22 Dec 22:31:19 - [info] [OpcUa-Client:84353b80.d69fb8] Session closed
22 Dec 22:31:19 - [info] [OpcUa-Client:84353b80.d69fb8] Client status: session closed
22 Dec 22:31:19 - [info] [OpcUa-Client:84353b80.d69fb8] Client status: closed
22 Dec 22:31:19 - [error] [OpcUa-Client:84353b80.d69fb8] Client node error on:  error: " error at active reading: Session has been closed and should not be used to perform a transaction anymore"
22 Dec 22:31:19 - [info] [OpcUa-Client:84353b80.d69fb8] Client status: error
22 Dec 22:31:19 - [warn] [OpcUa-Item:c7d7e4aa.ee8468] OpcUaClientNode: Setting value to
22 Dec 22:31:19 - [info] [OpcUa-Client:84353b80.d69fb8] Client disconnected!
22 Dec 22:31:19 - [warn] [OpcUa-Client:84353b80.d69fb8] OpcUaClientNode: Create Client: {"securityPolicy":"http://opcfoundation.org/UA/SecurityPolicy#None","securityMode":1,"requestedSessionTimeout":300000,"applicationName":"","clientName":"","endpoint_must_exist":false,"defaultSecureTokenLifetime":200000,"connectionStrategy":{"maxRetry":10,"initialDelay":5000,"maxDelay":30000},"keepSessionAlive":true,"certificateFile":"C:\\Users\\karaimi\\.node-red\\node_modules\\node-red-contrib-opcua\\node_modules\\node-opcua-client\\certificates\\client_selfsigned_cert_2048.pem","privateKeyFile":"C:\\Users\\karaimi\\.node-red\\node_modules\\node-red-contrib-opcua\\node_modules\\node-opcua-client\\certificates\\PKI\\own\\private\\private_key.pem"}
22 Dec 22:31:19 - [info] [OpcUa-Client:84353b80.d69fb8] Client status: create client
22 Dec 22:31:19 - [info] [OpcUa-Client:84353b80.d69fb8] Connecting to opc.tcp://H7Q8Q13.mshome.net:53530/OPCUA/SimulationServer
22 Dec 22:31:19 - [info] [OpcUa-Client:84353b80.d69fb8] Client status: connecting
22 Dec 22:31:19 - [info] [OpcUa-Client:84353b80.d69fb8] Exact endpointUrl: opc.tcp://H7Q8Q13.mshome.net:53530/OPCUA/SimulationServer hostname: H7Q8Q13
22 Dec 22:31:19 - [warn] [OpcUa-Item:c7d7e4aa.ee8468] OpcUaClientNode: Setting value to
22 Dec 22:31:19 - [info] [OpcUa-Client:84353b80.d69fb8] Client disconnected!
22 Dec 22:31:19 - [info] [OpcUa-Client:84353b80.d69fb8] Connected to opc.tcp://H7Q8Q13.mshome.net:53530/OPCUA/SimulationServer
22 Dec 22:31:19 - [info] [OpcUa-Client:84353b80.d69fb8] Connected to opc.tcp://H7Q8Q13.mshome.net:53530/OPCUA/SimulationServer
22 Dec 22:31:19 - [error] [OpcUa-Client:84353b80.d69fb8] Client node error on:  error: "Cannot read endpoints: TypeError: Cannot read property 'getEndpoints' of null"
22 Dec 22:31:19 - [info] [OpcUa-Client:84353b80.d69fb8] Create session ...
22 Dec 22:31:19 - [info] [OpcUa-Client:84353b80.d69fb8] Create session with userIdentity: {"userName":"","password":"","type":0}
22 Dec 22:31:19 - [error] [OpcUa-Client:84353b80.d69fb8] Client node error on:  error: "Client not yet created, cannot create session"
22 Dec 22:31:19 - [info] [OpcUa-Client:84353b80.d69fb8] Client status: connection error

mikakaraila, 2020-12-22:
I just published new version 0.2.90 that contains fixes to error handling. Test it please.

EliasReuter, 2020-12-24:
hi @mikakaraila,

I downloaded the new version, but now following error appears after a while (now with information):
[image]

mikakaraila, 2020-12-25:
What is the first error you will get?

s4ke, 2021-01-04:
Could it be that the opc ua client is not properly cleaned up and retried if it dies during connection? I tested the following:

1. Active connection in Node-RED
2. Killed OPC-UA Server
3. Tried fetching data via inject node, died with this error: "Client node error on: READ Client for <SERVER NAME> Server error: "READ Client for  <SERVER NAME> Server error at active reading: Session has been closed and should not be used to perform a transaction anymore""
4. Restarted OPC-UA Server, injection does not work anymore

mikakaraila, 2021-01-04:
I tested in the same way with Prosys UA Simulator. What server did you use? Client should go to re-connect itself.
Is your flow sending inject with interval to read values? What is interval 1s?

I will re-test again, can be some timing problem and thus client is not connecting correctly.

s4ke, 2021-01-04:
For this particular test, we used https://github.com/flopach/opc-ua-sensor-simulator

mikakaraila, 2021-01-04:
Aah, it is using Free OPCUA (Python based). OK I will use it to test again.

I expect there can be something that perhaps causes re-connect is not working...

mikakaraila, 2021-01-04:
Yes, there seems to be something that Python server makes different way...

[image]

mikakaraila, 2021-01-04:
Seems to be problem to UaExpert also to re-connect to server:

[image]

mikakaraila, 2021-01-04:
For me problem seems to be somehow inside Python server. I will check & compare this to Prosys UA Simulator how it handles session after server is re-started and old id is used to create session. As you see UaExpert fails to re-connect for the same reason.

As new session is created and then client tries to connect there seems to come error, this causes error in creating connection:
ERROR:asyncua.server.uaprocessor:sending service fault response: "The server does not support the requested service." (BadServiceUnsupported)

INFO:asyncua:Starting server!
WARNING:asyncua.server.server:Endpoints other than open requested but private key and certificate are not set.
INFO:asyncua.server.internal_server:starting internal server
INFO:asyncua.server.binary_server_asyncio:Listening on 127.0.0.1:4840
INFO:asyncua.server.binary_server_asyncio:New connection from ('127.0.0.1', 51953)
INFO:asyncua.server.uaprocessor:Activate session request (None)
INFO:asyncua.server.uaprocessor:request to activate non-existing session (None)
ERROR:asyncua.server.uaprocessor:sending service fault response: "The session id is not valid." (BadSessionIdInvalid)
INFO:asyncua.server.uaprocessor:Create session request (None)
INFO:asyncua.server.internal_session:Created internal session ('127.0.0.1', 51953)
INFO:asyncua.server.internal_session:Create session request
INFO:asyncua.server.internal_server:get endpoint
INFO:asyncua.server.uaprocessor:Activate session request (None)
INFO:asyncua.server.internal_session:activate session
INFO:asyncua.server.internal_session:Activated internal session ('127.0.0.1', 51953) for user User(role=<UserRole.User: 3>, name=None)
INFO:asyncua.server.uaprocessor:Read request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.uaprocessor:Browse request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.uaprocessor:Browse request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.uaprocessor:Browse request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.uaprocessor:translate browsepaths to nodeids request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.address_space:element RelativePathElement(ReferenceTypeId:TwoByteNodeId(i=46), IsInverse:False, IncludeSubtypes:True, TargetName:QualifiedName(0:OptionSetValues)) was not found in node FourByteNodeId(i=17588)
INFO:asyncua.server.address_space:element RelativePathElement(ReferenceTypeId:TwoByteNodeId(i=46), IsInverse:False, IncludeSubtypes:True, TargetName:QualifiedName(0:OptionSetValues)) was not found in node FourByteNodeId(i=288)
INFO:asyncua.server.address_space:element RelativePathElement(ReferenceTypeId:TwoByteNodeId(i=46), IsInverse:False, IncludeSubtypes:True, TargetName:QualifiedName(0:OptionSetValues)) was not found in node FourByteNodeId(i=20998)
INFO:asyncua.server.address_space:element RelativePathElement(ReferenceTypeId:TwoByteNodeId(i=46), IsInverse:False, IncludeSubtypes:True, TargetName:QualifiedName(0:OptionSetValues)) was not found in node FourByteNodeId(i=289)
INFO:asyncua.server.address_space:element RelativePathElement(ReferenceTypeId:TwoByteNodeId(i=46), IsInverse:False, IncludeSubtypes:True, TargetName:QualifiedName(0:OptionSetValues)) was not found in node FourByteNodeId(i=11737)
INFO:asyncua.server.uaprocessor:Read request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.uaprocessor:Read request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.uaprocessor:Browse request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.uaprocessor:Read request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.uaprocessor:Browse request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.uaprocessor:Read request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.uaprocessor:Browse request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.uaprocessor:Read request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.uaprocessor:Browse request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.uaprocessor:Read request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.uaprocessor:Browse request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.uaprocessor:Read request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.uaprocessor:Browse request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.uaprocessor:create subscription request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.subscription_service:create subscription
INFO:asyncua.server.uaprocessor:create monitored items request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.subscription_service:create monitored items
INFO:asyncua.server.monitored_item_service.78:request to subscribe to events for node FourByteNodeId(i=2253) and attribute 12
INFO:asyncua.server.uaprocessor:create subscription request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.subscription_service:create subscription
INFO:asyncua.server.uaprocessor:create monitored items request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.subscription_service:create monitored items
INFO:asyncua.server.monitored_item_service.79:request to subscribe to datachange for node FourByteNodeId(ns=2;i=3) and attribute 13
INFO:asyncua.server.monitored_item_service.79:request to subscribe to datachange for node FourByteNodeId(ns=2;i=4) and attribute 13
[... cut ...]
INFO:asyncua.server.internal_session:Activated internal session ('127.0.0.1', 51966) for user User(role=<UserRole.User: 3>, name=None)
WARNING:asyncua.server.uaprocessor:Unknown message received FourByteNodeId(i=841) (User(role=<UserRole.User: 3>, name=None))
ERROR:asyncua.server.uaprocessor:sending service fault response: "The server does not support the requested service." (BadServiceUnsupported)
INFO:asyncua.server.uaprocessor:re-publish request (User(role=<UserRole.User: 3>, name=None))
INFO:asyncua.server.uaprocessor:re-publish request (User(role=<UserRole.User: 3>, name=None))

mikakaraila, 2021-01-04:
And with Prosys UA Simulation Server node-red OPCUA client and UaExpert can re-connect:

[image]

mikakaraila, 2021-01-04:
Node-red flow console shows re-connect warn messages:

[image]

mikakaraila, 2021-01-04:
It is somehow related to this issue:

https://github.com/FreeOpcUa/python-opcua/issues/594

mikakaraila, 2021-01-04:
Node-red client (node-opcua) fails to reconnect, server disconnects client by error:

OpcUaClientNode:  !!!!!!!!!!!!!!!!!!!!!!!!  Starting Reconnection !!!!!!!!!!!!!!!!!!!
4 Jan 16:55:13 - [warn] [OpcUa-Client:b5077895.6fbab8] OpcUaClientNode:  !!!!!!!!!!!!!!!!!!!!!!!!  Starting Reconnection !!!!!!!!!!!!!!!!!!!
OpcUaClientNode: backoff  attempt #0 retrying in 5 seconds
4 Jan 16:55:14 - [warn] [OpcUa-Client:b5077895.6fbab8] OpcUaClientNode: backoff  attempt #0 retrying in 5 seconds
OpcUaClientNode: backoff  attempt #1 retrying in 10 seconds
4 Jan 16:55:20 - [warn] [OpcUa-Client:b5077895.6fbab8] OpcUaClientNode: backoff  attempt #1 retrying in 10 seconds
OpcUaClientNode:  !!!!!!!!!!!!!!!!!!!!!!!!  CONNECTION RE-ESTABLISHED !!!!!!!!!!!!!!!!!!!
4 Jan 16:55:30 - [warn] [OpcUa-Client:b5077895.6fbab8] OpcUaClientNode:  !!!!!!!!!!!!!!!!!!!!!!!!  CONNECTION RE-ESTABLISHED !!!!!!!!!!!!!!!!!!!
OpcUaClientNode:  !!!!!!!!!!!!!!!!!!!!!!!!  Starting Reconnection !!!!!!!!!!!!!!!!!!!
4 Jan 16:55:30 - [warn] [OpcUa-Client:b5077895.6fbab8] OpcUaClientNode:  !!!!!!!!!!!!!!!!!!!!!!!!  Starting Reconnection !!!!!!!!!!!!!!!!!!!
Disconnected following reconnection failure Connection Break

EliasReuter, 2021-01-04:
I tried @s4ke procedure to test it but with UaAnsiCServer OPC UA Test Server and following error appeared:
[image]

mikakaraila, 2021-01-04:
Did you check opc-ua-server.py, it needs to modify if you use Python version > 3.6

Server code seems to miss modify subscription and thus cannot response to client request to change subscription to new session... problem is located to server and node-opcua library that handles error response to modify subscription.

UaExpert seems to delete subscription and create a new one as node-opcua stops it to "BadServiceUnsupported". I will ask Etienne to check this, but it can take longer time as with commercial servers re-connect is working correctly.

mikakaraila, 2021-01-04:
@EliasReuter can you check what happens on UaAnsiCServer server? As I have commented this case, I have been looking server  console/log and node-red console. Debug tab tells only error that server returns on response and if the response does not contain any extra info then we cannot see actual error & solve this case.

Also testing this problem with UaExpert could show some extra info, can you make same testing with UaExpert with UaAnsiCServer?

mikakaraila, 2021-01-04:
I just downloaded and tested this case with Ansi C OPC UA Server and with subscription node-red client re-connects and everything works as it should.

mikakaraila, 2021-01-04:
Only readmultiple action did not show actual error in case of diagnostics was undefined.

I modified node_error to show actual error to user as in your debug tab it was empty, here is error it should report:
[image]

EliasReuter, 2021-01-05:
My test with UaAnsiCServer is, once connection is established and the node is reading multiple sample values provided by the server, I close UAAnsiCServer's Console, in order to test a complete disconnection. 
Then that {} empty error is logged. 
It seems that it reconnects immediately (without RECONNECTING log) and doesn't make the whole Node Red Server restart or freeze as before.
Am I testing it wrong? Is that behaviour correct? 
How have you been simulating disconnection?

This are tests in my local environment. 
I'm currently not able to test this in the production environment I've been mentioning with the Matrikon Server. Once I have access to that Server I'll inform results and testing this last 0.2.91 version.

EliasReuter, 2021-01-05:
I am able again to test in the Productive server I've mentioned (sooner than expected) and get the following error message:
[image]

Quote to an earlier comment from @mikakaraila:
> I expect Prosys UA Simulation server decides this. I request it to be 300s == 5min but server limits session timeout to 60s.
> Rename .txt to .js and test again.
> 
> [102-opcuaclient.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/5702906/102-opcuaclient.txt)

Is this error related? Is there something on the Server side to configure?

mikakaraila, 2021-01-05:
You can check sessions from the Prosys Server:
[image]

Client-node name is used in the sessionName:
[image]

mikakaraila, 2021-01-05:
I studied more wrapper log and found some interesting errors:

[2020-12-11 15:15:23.113] [StackTrace      , 092] ***EXCEPTION*** BadCommunicationError 'Could not create COM server 'f8582cf2-88fb-11d0-b850-00c0f0104305' on host 'localhost'. Reason: Class not registered.' Could not create COM server 'f8582cf2-88fb-11d0-b850-00c0f0104305' on host 'localhost'. Reason: Class not registered.
[2020-12-11 15:15:23.113] [Error           , 092] : Could not connect to server (opc.com://localhost/Matrikon.OPC.Simulation/f8582cf2-88fb-11d0-b850-00c0f0104305). BadCommunicationError 'Could not create COM server 'f8582cf2-88fb-11d0-b850-00c0f0104305' on host 'localhost'. Reason: Class not registered.'

========================================
Id: BadCommunicationError
Description: Could not create COM server 'f8582cf2-88fb-11d0-b850-00c0f0104305' on host 'localhost'. Reason: Class not registered.
>>> Could not create COM server 'f8582cf2-88fb-11d0-b850-00c0f0104305' on host 'localhost'. Reason: Class not registered.
---    at Opc.Ua.Com.ComUtils.CreateInstance1(Guid clsid, String hostName, UserIdentity identity)
---    at Opc.Ua.Com.ServerFactory.CreateServer(Uri uri, UserIdentity identity)
---    at Opc.Ua.Com.Client.ComClient.CreateInstance()
========================================

[2020-12-11 15:15:23.113] [Information     , 092] Calling OPC Classic API IOPCHDA_Server.GetStatus.
[2020-12-11 15:15:23.113] [Information     , 079] Communication with COM server failed. Disposing server and reconnecting.
[2020-12-11 15:15:23.113] [Error           , 092] : IOPCHDA_Server.GetStatus ObjectDisposedException 'Cannot access a disposed object.
Object name: 'The COM server has been disposed.'.'

========================================
Id: Bad
Description: Cannot access a disposed object.
Object name: 'The COM server has been disposed.'.
>>> Cannot access a disposed object.
Object name: 'The COM server has been disposed.'.
---    at Opc.Ua.Com.Client.ComObject.BeginComCall[T](String methodName, Boolean throwOnError)
---    at Opc.Ua.Com.Client.ComHdaClient.GetStatus()
========================================

[2020-12-11 15:15:23.113] [StackTrace      , 079] ***EXCEPTION*** BadCommunicationError 'Could not create COM server 'f8582cf2-88fb-11d0-b850-00c0f0104305' on host 'localhost'. Reason: Class not registered.' Could not create COM server 'f8582cf2-88fb-11d0-b850-00c0f0104305' on host 'localhost'. Reason: Class not registered.
[2020-12-11 15:15:23.113] [StackTrace      , 050] ***EXCEPTION*** BadCommunicationError 'Could not create COM server 'f8582cf2-88fb-11d0-b850-00c0f0104305' on host 'localhost'. Reason: Class not registered.' Could not create COM server 'f8582cf2-88fb-11d0-b850-00c0f0104305' on host 'localhost'. Reason: Class not registered.
[2020-12-11 15:15:23.113] [Error           , 079] : Could not connect to server (opc.com://localhost/Matrikon.OPC.Simulation/f8582cf2-88fb-11d0-b850-00c0f0104305). BadCommunicationError 'Could not create COM server 'f8582cf2-88fb-11d0-b850-00c0f0104305' on host 'localhost'. Reason: Class not registered.'

========================================
Id: BadCommunicationError
Description: Could not create COM server 'f8582cf2-88fb-11d0-b850-00c0f0104305' on host 'localhost'. Reason: Class not registered.
>>> Could not create COM server 'f8582cf2-88fb-11d0-b850-00c0f0104305' on host 'localhost'. Reason: Class not registered.
---    at Opc.Ua.Com.ComUtils.CreateInstance1(Guid clsid, String hostName, UserIdentity identity)
---    at Opc.Ua.Com.ServerFactory.CreateServer(Uri uri, UserIdentity identity)
---    at Opc.Ua.Com.Client.ComClient.CreateInstance()
========================================

[2020-12-11 15:15:23.113] [Information     , 079] Calling OPC Classic API IOPCEventServer.GetStatus.
[2020-12-11 15:15:23.113] [Error           , 079] : IOPCEventServer.GetStatus ObjectDisposedException 'Cannot access a disposed object.
Object name: 'The COM server has been disposed.'.'

========================================
Id: Bad
Description: Cannot access a disposed object.
Object name: 'The COM server has been disposed.'.
>>> Cannot access a disposed object.
Object name: 'The COM server has been disposed.'.
---    at Opc.Ua.Com.Client.ComObject.BeginComCall[T](String methodName, Boolean throwOnError)
---    at Opc.Ua.Com.Client.ComAeClient.GetStatus()
========================================

[2020-12-11 15:15:23.113] [Information     , 079] OPC Classic API IOPCEventServer.GetStatus call completed.
[2020-12-11 15:15:23.113] [Information     , 092] OPC Classic API IOPCHDA_Server.GetStatus call completed.
[2020-12-11 15:15:23.113] [Error           , 050] : Could not connect to server (opc.com://localhost/Matrikon.OPC.Simulation/f8582cf2-88fb-11d0-b850-00c0f0104305). BadCommunicationError 'Could not create COM server 'f8582cf2-88fb-11d0-b850-00c0f0104305' on host 'localhost'. Reason: Class not registered.'

========================================
Id: BadCommunicationError
Description: Could not create COM server 'f8582cf2-88fb-11d0-b850-00c0f0104305' on host 'localhost'. Reason: Class not registered.
>>> Could not create COM server 'f8582cf2-88fb-11d0-b850-00c0f0104305' on host 'localhost'. Reason: Class not registered.
---    at Opc.Ua.Com.ComUtils.CreateInstance1(Guid clsid, String hostName, UserIdentity identity)
---    at Opc.Ua.Com.ServerFactory.CreateServer(Uri uri, UserIdentity identity)
---    at Opc.Ua.Com.Client.ComClient.CreateInstance()
[... cut ...]
[2020-12-11 15:15:23.128] [Error           , 050] : IOPCServer.GetStatus ObjectDisposedException 'Cannot access a disposed object.
Object name: 'The COM server has been disposed.'.'

========================================
Id: Bad
Description: Cannot access a disposed object.
Object name: 'The COM server has been disposed.'.
>>> Cannot access a disposed object.
Object name: 'The COM server has been disposed.'.
---    at Opc.Ua.Com.Client.ComObject.BeginComCall[T](String methodName, Boolean throwOnError)
---    at Opc.Ua.Com.Client.ComDaClient.GetStatus()

mikakaraila, 2021-01-05:
One reason can be that wrapper is not fast enough when connection break occurs and it runs original Classic OPC and tries to return server status and restore all Items from the server back to OPC UA.  Can you test with just 5-10 items so server will response faster and re-connect will not timeout?

In case of Free OPCUA server does not provide needed re-connect service (Unsupported error).

EliasReuter, 2021-01-06:
I let the Productive Server with ~15 items for 30 minutes and no error message showed up. 
Then I added items in a total of ~60 items  for 30 minutes and no error messages showed up.
Then I added items in a total of ~150 items for 120 minutes and no error messages showed up.
Then I added items in a total of ~250 items for 20 minutes and THEN same error messages showed up.

Attached the log for you.

mikakaraila, 2021-01-06:
Hmm, server does not accept longer session timeout and re-connect takes too long time with 250 items.

I expect there is nothing that I can do, but can you run parallel 150 items / client node?
Try & test what happens then with server...

EliasReuter, 2021-01-06:
[logNodeRedForOpc.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/5777551/logNodeRedForOpc.txt)

[20 more replies]

closed 2021-01-18 by mikakaraila