mikakaraila/node-red-contrib-opcua #209: Alarms and Events reading

henjoe, 2020-10-01:
Hello Mika,

Not an issue but a question regarding OPC A&E, 
As I noticed, most of the user of node-red-contrib-opcua only read the data of a specific tags.
I am wondering if it is possible to read alarms and events too? 
I've seen this method under opcuaClient:
[image]

Does this read an A&E data? 

Thanks Mika!

With kind regards,
Henjoe

mikakaraila, 2020-10-01:
Yes, but as I tested it there are some bugs. Typescript API needs some changes... fixing.

mikakaraila, 2020-10-01:
Only few lines and works:

[image]

mikakaraila, 2020-10-01:
Fixed, look events.json from examples

closed 2020-10-01 by mikakaraila

henjoe, 2020-10-01:
Thank you Mika, highly appreciated ✔️

henjoe, 2020-10-02:
Hello Mika, 

Just follow up questions. How did you get A&E simulation data on Prosys Simulation Server? 
I can't get any when I copied your flow.

[image]

Thanks Mika!

With kind regards,
Henjoe

henjoe, 2020-10-02:
Hello Mika,

I am getting the output now.
But I wondering on the side of OPCUA Event, What does it mean is "i=2253" as the root node?
[image]

When I see my Prosys, I can't see any nodeid regarding"i=2253"? I just want to ask this because I want to get the concept for this.
[image]

Thanks Mika!

With kind regards,
Henjoe

mikakaraila, 2020-10-02:
It is server object as it should always get all event notifications.

[image]

henjoe, 2020-10-02:
Oh I see! Thanks Mika! I've forgot to use the OPCUA Expert to see this. 

With kind regards,
Henjoe