mikakaraila/node-red-contrib-opcua #210: Node-Red OPC-Client is stuck when the connection to the Server is lost for over a minute and reconnects

EddyITE, 2020-10-07:
I have an OPC-Client in Node Red which is subscribed to an OPC-Server using freeOPCUA Python.
When unplugging the internet from the Raspberry Pi with the Server and plugging it in again after around 2 minutes. The client reconnects but is stuck and stops giving me new values. The node-red -v console gives me the following message.
[image]

It is spammed with keep-alive messages and does not stop.
Trying to subscribe does not work, but redeploying the flow or using msg.action=reconnect and then subscribe again works.
The OPC-Server process on the Raspberry PI jumps from 3% to around 18% and stay that high until I redeploy or restart the client.
When either using another Client or another Server, the problem can't be reproduced. I am guessing that both the Server and the Client are having problem with clearing the old session.

mikakaraila, 2020-10-07:
OK I need to reproduce it before I can study more detailed... was FreeOpcUa Server some sample python or modified one?

EddyITE, 2020-10-08:
I have modified it a little bit but the same behaviour can be reproduced with the example from https://github.com/FreeOpcUa/python-opcua/blob/master/examples/server-minimal.py.

You can use the following flow to test this behaviour:
`[{"id":"24ecd56d.6e84ba","type":"inject","z":"76066bb8.dbc0a4","name":"","props":[],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":210,"y":280,"wires":[["b461b124.10ed3"]]},{"id":"67b79f39.ced8f","type":"delay","z":"76066bb8.dbc0a4","name":"","pauseType":"delay","timeout":"1","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":400,"y":320,"wires":[["b461b124.10ed3"]]},{"id":"2666fbf5.1d2414","type":"function","z":"76066bb8.dbc0a4","name":"Clear Message and send empty msg to subscribe again","func":"msg = {};\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":270,"y":380,"wires":[["67b79f39.ced8f"]]},{"id":"7192b5a2.3c8e0c","type":"inject","z":"76066bb8.dbc0a4","name":"Reconnect","props":[{"p":"action","v":"reconnect","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":260,"y":460,"wires":[["4256ca2a.c9e624","2666fbf5.1d2414"]]},{"id":"b461b124.10ed3","type":"OpcUa-Item","z":"76066bb8.dbc0a4","item":"ns=2;i=2","datatype":"Int32","value":"","name":"","x":620,"y":280,"wires":[["4256ca2a.c9e624"]]},{"id":"4256ca2a.c9e624","type":"OpcUa-Client","z":"76066bb8.dbc0a4","endpoint":"a9013ec2.79493","action":"subscribe","deadbandtype":"a","deadbandvalue":1,"time":"1","timeUnit":"s","certificate":"n","localfile":"","securitymode":"None","securitypolicy":"None","name":"","x":820,"y":280,"wires":[["69a6219e.7f8eb"]]},{"id":"69a6219e.7f8eb","type":"debug","z":"76066bb8.dbc0a4","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":990,"y":280,"wires":[]},{"id":"a9013ec2.79493","type":"OpcUa-Endpoint","z":"","endpoint":"opc.tcp://172.24.57.169:4840","secpol":"None","secmode":"None","login":false}]`

mikakaraila, 2020-10-08:
Ack, thanks now it should be possible to fix.

mfurla8, 2020-10-13:
Hi any news on this issue?

mfurla8, 2020-10-13:
> Not yet, busy time at work. I tested first locally and it needs second computer to disconnect network.

Any way around?

Dobite Outlook za Android<https://aka.ms/ghei36>

mikakaraila, 2020-10-13:
Code that solved reconnecting is actually inside node-opcua. It will keep session alive and re-subscribe items.
I´m not sure if free-opcua-server is fully supporting needed opcua services.
Can you test with some other server?

EddyITE, 2020-10-14:
This problem only happens as a result of using the opcua Node and the FreeOPCUA Server. 
When changing the Server to the simulation Server from Prosys the Node works as intented.
When changing the Client to UaExpert, the FreeOPCUA Server works as well and reconnects without problem. The only Error message I get from UaExpert is : 
 Error: Number of acknowledgement results(0) does not match number of acknowledgement (1)
but this message appears even without disconnecting .

mikakaraila, 2020-10-14:
Ok then there is some interoperability with session re-create / re-subscribe with FreeOPCUA.

It might be on node-opcua... then I cannot solve this.

mikakaraila, 2020-12-21:
Please re-test with the latest node-opcua. There is internal change for the keep-alive manager.

mikakaraila, 2020-12-22:
New version 0.2.90 that uses node-opcua@2.28.3 released.

closed 2020-12-22 by mikakaraila