mikakaraila/node-red-contrib-opcua #211: Server stops with maximum number of connections has been reached

wernerpaulin, 2020-10-08:
I'm using the server-function of contrib-opcua and write in a 100ms cycle 2 float values on variables I've previously created.
All fine but after some time I get the error message "OPCUAServerEndPoint#_on_client_connection The maximum number of connection has been reached - Connection is refused" (please see the copy of my terminal output). 
This actually recovers a couple of time with the error message:
OPCUAServerEndPoint#_on_client_connection The maximum number of connection has been reached - Connection is refused
PREVENTING DDOS ATTACK => Closing unused channels

Beside the Client Nodes, I'm using UaExpert to read the data.

What do I do wrong? Thanks!

############### Copy of Terminal output #############################

------------------------------------------------------
            channelId =  134
             timeout  =  180000
        remoteAddress =  ::ffff:169.254.212.163
        remotePort    =  56637

        bytesWritten  =  1802
        bytesRead     =  1846
------------------------------------------------------
            channelId =  135
             timeout  =  180000
        remoteAddress =  ::ffff:169.254.212.163
        remotePort    =  56636

        bytesWritten  =  1802
        bytesRead     =  1846
------------------------------------------------------
            channelId =  137
             timeout  =  180000
        remoteAddress =  ::ffff:169.254.212.163
        remotePort    =  56644

        bytesWritten  =  1802
        bytesRead     =  1846
------------------------------------------------------
OPCUAServerEndPoint#_on_client_connection The maximum number of connection has been reached - Connection is refused

mikakaraila, 2020-10-08:
Are opening new client or session on each write?

This error message is coming from node-opcua package:
OPCUAServerEndPoint#_on_client_connection The maximum number of connection has been reached - Connection is refused
PREVENTING DDOS ATTACK => Closing unused channels

wernerpaulin, 2020-10-08:
Thanks for your very fast reply!
I've added a screenshot which shows that I write a payload via OPC UA Item to a OPC UA Client Node. This happens around every 100ms.
From my understanding I do not open new client sessions. Did I miss something?

[image]

wernerpaulin, 2020-10-09:
It seems this error happens as soon as I connect a OPC UA Item with a OPC UA client node.
Do you have any idea about this?
Thanks!

wernerpaulin, 2020-10-09:
Problem solved. It was actually caused by the start sequence of the server. With this setup it worked and it is much simpler:
[image]
Thanks!

closed 2020-10-09 by wernerpaulin