mikakaraila/node-red-contrib-opcua #213: OPC UA server error v. 0.2.87

Sebalonghi, 2020-10-13:
I did a new installation of Nodered with the last version of the contrib-opcua 0.2.87.
I did a simple OPC UA Server, but I cannot connect UAExpert with it and cannot understand why.
With the version 0.2.70 which I have on another PC, I can do it very easily.
The error that comes from UAExpert is:
"There was no exacy match for the specified URL in the server's endpoints, using the first server certificate found for connecting."
and then
"Error: UaSessionPrivate: ActivateSession - no endpoint matching the session security settings"
It seems to be a problem with the certificates. I use anonymous authentication.

Moreover, several commands to the OPC UA server node as "addVariable", "addFolder", etc. return error messages.

All of these were perfectly functioning in the previous releases.
Thank you for answering and helping.

mikakaraila, 2020-10-13:
I expect this can be caused by node-opcua. There are some breaking changes:
https://github.com/node-opcua/node-opcua/releases/tag/v2.20.0

Can you provide some error logs? It would help to under stand problem.

mikakaraila, 2020-10-14:
Node-opcua check that hostname will match to endpoint so check that it is exactly same (lowercase/uppercase).

mikakaraila, 2020-10-15:
Server endpoints are here:
15 Oct 16:33:31 - [info] [OpcUa-Server:2617f455.77df1c] Server endpointUrl(s): opc.tcp://ieciiot:53880 securityMode: 1 securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#None
15 Oct 16:33:31 - [info] [OpcUa-Server:2617f455.77df1c] Server endpointUrl(s): opc.tcp://ieciiot:53880 securityMode: 2 securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
15 Oct 16:33:31 - [info] [OpcUa-Server:2617f455.77df1c] Server endpointUrl(s): opc.tcp://ieciiot:53880 securityMode: 2 securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic256
15 Oct 16:33:31 - [info] [OpcUa-Server:2617f455.77df1c] Server endpointUrl(s): opc.tcp://ieciiot:53880 securityMode: 2 securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
15 Oct 16:33:31 - [info] [OpcUa-Server:2617f455.77df1c] Server endpointUrl(s): opc.tcp://ieciiot:53880 securityMode: 3 securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
15 Oct 16:33:31 - [info] [OpcUa-Server:2617f455.77df1c] Server endpointUrl(s): opc.tcp://ieciiot:53880 securityMode: 3 securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic256
15 Oct 16:33:31 - [info] [OpcUa-Server:2617f455.77df1c] Server endpointUrl(s): opc.tcp://ieciiot:53880 securityMode: 3 securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256

Which one you tested?
opc.tcp://ieciiot:53880

Sebalonghi, 2020-10-16:
I read the changes done in the OPC UA node  @ https://github.com/node-opcua/node-opcua/releases/tag/v2.20.0 
and understood that now the endpoint should be the alternateHostname, in my case "ieciiot". I was trying to connect to the server with the IP address. If I put "opc.tcp://ieciiot:53880" now I can connect to the server.
I think the issue is then solved.
Thank you.

closed 2020-10-18 by mikakaraila