mikakaraila/node-red-contrib-opcua #215: Node-Red crashes if security policy is not available

mhechthz, 2020-10-19:
Hi,

during a test I set the wrong security policy Basic256Rsa15 which is not provided by OPCUA server. This crashes Node-Red with the message:

```
19 Oct 17:36:29 - [info] Server now running at http://127.0.0.1:1880/
15:36:29.809Z :client_base_impl              :443   fetchServerCertificate Err =  Cannot find an Endpoint matching  security mode: 3 policy: http://opcfoundation.org/UA/SecurityPolicy#Basic256Rsa15
19 Oct 17:36:29 - [red] Uncaught Exception:
19 Oct 17:36:29 - TypeError: Cannot read property 'toString' of undefined
```

mikakaraila, 2020-10-20:
OK, I will take a look later.

mikakaraila, 2020-11-02:
Forgot to fix this, check what new version will set to output.

mhechthz, 2020-11-03:
Also the new node is directly crashing without any status information:
```
11:20:48.311Z :client_base_impl              :443   fetchServerCertificate Err =  Cannot find an Endpoint matching  security mode: 3 policy: http://opcfoundation.org/UA/SecurityPolicy#Basic192
3 Nov 12:20:48 - [red] Uncaught Exception:
3 Nov 12:20:48 - TypeError: Cannot read property 'toString' of undefined
    at async.series (/data/node_modules/node-opcua-client/dist/private/client_base_impl.js:108:103)
    at err (/data/node_modules/async/dist/async.js:2955:19)
    at wrapper (/data/node_modules/async/dist/async.js:268:20)
    at replenish (/data/node_modules/async/dist/async.js:435:29)
    at iterateeCallback (/data/node_modules/async/dist/async.js:424:21)
    at /data/node_modules/async/dist/async.js:321:20
    at Immediate.result (/data/node_modules/async/dist/async.js:2953:17)
    at runCallback (timers.js:705:18)
    at tryOnImmediate (timers.js:676:5)
    at processImmediate (timers.js:658:5)

```

mikakaraila, 2020-11-03:
Cannot reproduce, can you check your node-opcua version?

I got just these to console:
3 Nov 17:01:59 - [info] [OpcUa-Client:c67d31e5.5a4c5] Client status: create client
3 Nov 17:01:59 - [info] [OpcUa-Client:c67d31e5.5a4c5] Connecting to opc.tcp://H7Q8Q13:53530/OPCUA/SimulationServer
3 Nov 17:01:59 - [info] [OpcUa-Client:c67d31e5.5a4c5] Client status: connecting
3 Nov 17:01:59 - [info] [OpcUa-Client:c67d31e5.5a4c5] Exact endpointUrl: opc.tcp://H7Q8Q13:53530/OPCUA/SimulationServer hostname: H7Q8Q13
3 Nov 17:01:59 - [warn] [OpcUa-Client:22cb1512.bb27ba] OpcUaClientNode: Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt securityMode:3 securityPolicy:"http://opcfoundation.org/UA/SecurityPolicy#Basic192"
3 Nov 17:01:59 - [warn] [OpcUa-Client:22cb1512.bb27ba] OpcUaClientNode: Case B: UserName & password does not match to server (needed by Sign):
3 Nov 17:01:59 - [info] [OpcUa-Client:22cb1512.bb27ba] Client status: invalid endpoint opc.tcp://H7Q8Q13:53530/OPCUA/SimulationServer
3 Nov 17:01:59 - [warn] [OpcUa-Client:c67d31e5.5a4c5] OpcUaClientNode: Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt securityMode:3 securityPolicy:"**http://opcfoundation.org/UA/SecurityPolicy#Basic192**"
3 Nov 17:01:59 - [warn] [OpcUa-Client:c67d31e5.5a4c5] OpcUaClientNode: Case B: UserName & password does not match to server (needed by Sign):
3 Nov 17:01:59 - [info] [OpcUa-Client:c67d31e5.5a4c5] Client status: invalid endpoint opc.tcp://H7Q8Q13:53530/OPCUA/SimulationServer
3 Nov 17:01:59 - [warn] [OpcUa-Client:ORIGINAL] ORIGINAL: Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt securityMode:3 securityPolicy:"http://opcfoundation.org/UA/SecurityPolicy#Basic192"
3 Nov 17:01:59 - [warn] [OpcUa-Client:ORIGINAL] ORIGINAL: Case B: UserName & password does not match to server (needed by Sign):
3 Nov 17:01:59 - [info] [OpcUa-Client:ORIGINAL] Client status: invalid endpoint opc.tcp://H7Q8Q13:53530/OPCUA/SimulationServer

mhechthz, 2020-11-03:
I'm using Node-Red 1.1.3 on Ubuntu Linux.

mikakaraila, 2020-11-03:
Yes, but I asked node-opcua version. Because in your case it seems to come inside it.

mhechthz, 2020-11-04:
In the package file I find:
```

  "_from": "node-opcua@^2.14.0",
  "_id": "node-opcua@2.23.0",
 
```

mikakaraila, 2020-11-04:
OK, I will ask Etienne to check 2.23 if there is something with endpoint checking (as it seems to be).

erossignon, 2020-11-04:
(node-opcua should not crash, there is a bug here we are going to resolve)

This being said the connection cannot happen because the endpoint that you are using to connect to the server does not match one of the endpoints exposed by the server. This extra verification has been added to help us conform to the latest CTT test.

Please refers to https://github.com/node-opcua/node-opcua/releases/tag/v2.20.0 for more information about this change that was introduced on version 2.20.0

mikakaraila, 2020-11-04:
If you look this console output it is coming from node-opcua:
11:20:48.311Z :client_base_impl              :443   fetchServerCertificate Err =  Cannot find an Endpoint matching  security mode: 3 policy: http://opcfoundation.org/UA/SecurityPolicy#Basic192
3 Nov 12:20:48 - [red] Uncaught Exception:
3 Nov 12:20:48 - TypeError: Cannot read property 'toString' of undefined
    at async.series (/data/node_modules/node-opcua-client/dist/private/client_base_impl.js:108:103)
    at err (/data/node_modules/async/dist/async.js:2955:19)
    at wrapper (/data/node_modules/async/dist/async.js:268:20)
    at replenish (/data/node_modules/async/dist/async.js:435:29)
    at iterateeCallback (/data/node_modules/async/dist/async.js:424:21)
    at /data/node_modules/async/dist/async.js:321:20
    at Immediate.result (/data/node_modules/async/dist/async.js:2953:17)
    at runCallback (timers.js:705:18)
    at tryOnImmediate (timers.js:676:5)
    at processImmediate (timers.js:658:5)

erossignon, 2020-11-04:
When you find error telling you that the enpoint cannot be found you can use this script to run externally to figure out what are the valid endpoints exposed by your server.

https://gist.github.com/erossignon/88d73c43bba876012800e0f1c07163ea#file-getendpointbasic-js

mikakaraila, 2020-11-04:
What about this?

[image]

erossignon, 2020-11-04:
Fix is already in https://github.com/node-opcua/node-opcua/commit/3b7d35caf735ac5051522da02496b0f73340a622
The crash was here (in the debuglLog line)  because of the missing `return`
````
   if (!selectedEndpoint) {
            callback(
            return callback(
                new Error(
                    "Cannot find an Endpoint matching " +
                        " security mode: " +
                        securityMode.toString() +
                        " policy: " +
                        securityPolicy.toString()
                )
            );
        }
        debugLog(chalk.bgWhite.red("xxxxxxxxxxxxxxxxxxxxx => selected EndPoint = "), selectedEndpoint.toString());
````

erossignon, 2020-11-04:
Mika, make sure to edit the right file from the designated version of node-opcua 
the error took place in /data/node_modules/node-opcua-client/dist/private/client_base_impl.js:

I guess you are showing the Typescript file instead.

mikakaraila, 2020-11-04:
OK but I have most probably older version and @mhechthz has inter-mediate version that contains bug.
So updating node-opcua should fix this issue.

mikakaraila, 2020-11-04:
Or next coming version as commit is just 22min ago... so after node-opcua v2.24?

erossignon, 2020-11-04:
Updating to node-opcua@2.25.0 ( which is not yet released) will solve the crash issue, however, @mhechthz will still have to enter a valid endpoint to get the connection.

mikakaraila, 2020-11-04:
I expect this was found by accident and then we started to investigate actual cause for this.

Good to improve quality of node-opcua again.

erossignon, 2020-11-05:
node-opcua@2.24.1 has been released and fixes the issue.

mhechthz, 2020-11-06:
> 
> 
> Updating to node-opcua@2.25.0 ( which is not yet released) will solve the crash issue, however, @mhechthz will still have to enter a valid endpoint to get the connection.

Yes, it's clear that I have to use the right endpoint. As mika stated it was accidentially.

How can I now get the update into my docker container?

mikakaraila, 2020-11-07:
Just uninstall and then re-install node-red-contrib-opcua or manually install latest node-opcua.
It is already v2.25.

mhechthz, 2020-11-09:
Hmmm ... uninstall failed by hanging during uninstall process. Might not be your fault. So I did set up a new node-red instance. Now it works - no more error if accitentially chosen a non-existing security policy.

Thank you all.

closed 2020-11-09 by mikakaraila

federicoabait, 2023-06-30:
hello
I am trying to connect via node OPC UA CLient to the Prosys OPC UA Simation Server. My node red is deployed in a Docker container and I am getting the following error.

"Invalid endpoit, has server security policy...."

Version used: node-red-contrib-opcua, Version: 0.2.308

I have tried to connect from UI Expert to the same simulator without problem
Could you help me please?
Thanks

mikakaraila, 2023-06-30:
Perhaps more debug/console output would help me....

federicoabait, 2023-06-30:
there

2023-06-30 19:32:24 30 Jun 19:32:24 - [debug] [OpcUa-Client:14a2f4acec0b46f1] EndPoint: {"id":"c58dcd07ab22c4a1","type":"OpcUa-Endpoint","_closeCallbacks":[],"_inputCallback":null,"_inputCallbacks":null,"wires":[],"_wireCount":0,"credentials":{"user":"test","password":"test"},"endpoint":"opc.tcp://myhost:53530/OPCUA/SimulationServer","securityPolicy":"None","securityMode":"None","login":false,"none":true,"usercert":false,"userCertificate":"","userPrivatekey":"","user":"test","password":"test"}
2023-06-30 19:32:24 30 Jun 19:32:24 - [debug] [OpcUa-Client:14a2f4acec0b46f1] UserIdentity: {"type":0}
2023-06-30 19:32:24 30 Jun 19:32:24 - [debug] [OpcUa-Client:14a2f4acec0b46f1] 1) CREATE CLIENT: [{"securityPolicy":"1","securityMode":1,"clientCertificateManager":"2","req...
2023-06-30 19:32:24 30 Jun 19:32:24 - [debug] [OpcUa-Client:14a2f4acec0b46f1] Client status: create client
2023-06-30 19:32:24 30 Jun 19:32:24 - [debug] [OpcUa-Client:14a2f4acec0b46f1] Connecting to opc.tcp://myhost:53530/OPCUA/SimulationServer
2023-06-30 19:32:24 30 Jun 19:32:24 - [debug] [OpcUa-Client:14a2f4acec0b46f1] Client status: connecting
2023-06-30 19:32:24 30 Jun 19:32:24 - [debug] [OpcUa-Client:14a2f4acec0b46f1] Exact endpointUrl: opc.tcp://myhost:53530/OPCUA/SimulationServer hostname: 080449e845ab
2023-06-30 19:32:25 30 Jun 19:32:25 - [debug] [OpcUa-Client:14a2f4acec0b46f1] 2) Connect using ENDPOINT: opc.tcp://myhost:53530/OPCUA/SimulationServer
2023-06-30 19:32:40 17:32:40.403Z :client_base_impl              :531   Inner create secure channel has failed Transaction has timed out ( timeout = 15000 ms , request = GetEndpointsRequest)
2023-06-30 19:32:40 30 Jun 19:32:40 - [warn] [OpcUa-Client:14a2f4acec0b46f1] OpcUaClientNode: Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt, using securityMode: [1]
2023-06-30 19:32:40 30 Jun 19:32:40 - [warn] [OpcUa-Client:14a2f4acec0b46f1] OpcUaClientNode:         using securityPolicy: ["http://opcfoundation.org/UA/SecurityPolicy#None"]
2023-06-30 19:32:40 30 Jun 19:32:40 - [warn] [OpcUa-Client:14a2f4acec0b46f1] OpcUaClientNode: Case B: UserName & password does not match to server (needed by Sign), check username: undefined and password: XXX
2023-06-30 19:32:40 30 Jun 19:32:40 - [warn] [OpcUa-Client:14a2f4acec0b46f1] OpcUaClientNode: Invalid endpoint parameters: 
2023-06-30 19:32:40 30 Jun 19:32:40 - [debug] [OpcUa-Client:14a2f4acec0b46f1] Client status: Invalid endpoint, has server security policy: ["http://opcfoundation.org/UA/SecurityPolicy#None"]
2023-06-30 19:32:40 30 Jun 19:32:40 - [error] [OpcUa-Client:14a2f4acec0b46f1] Invalid endpoint
2023-06-30 19:32:49 30 Jun 19:32:49 - [debug] [OpcUa-Client:14a2f4acec0b46f1] Using node action: read
2023-06-30 19:32:49 30 Jun 19:32:49 - [debug] [OpcUa-Client:14a2f4acec0b46f1] Not connected, current status:Invalid endpoint, has server security policy: ["http://opcfoundation.org/UA/SecurityPolicy#None"]
2023-06-30 19:32:49 30 Jun 19:32:49 - [warn] [OpcUa-Client:14a2f4acec0b46f1] OpcUaClientNode: can't work without OPC UA Session
2023-06-30 19:32:49 30 Jun 19:32:49 - [debug] [OpcUa-Client:14a2f4acec0b46f1] Client disconnected!
2023-06-30 19:32:49 30 Jun 19:32:49 - [debug] [OpcUa-Client:14a2f4acec0b46f1] 1) CREATE CLIENT: [{"securityPolicy":"1","securityMode":1,"clientCertificateManager":"2","req...
2023-06-30 19:32:49 30 Jun 19:32:49 - [debug] [OpcUa-Client:14a2f4acec0b46f1] Client status: create client
2023-06-30 19:32:49 30 Jun 19:32:49 - [debug] [OpcUa-Client:14a2f4acec0b46f1] Connecting to opc.tcp://myhost:53530/OPCUA/SimulationServer
2023-06-30 19:32:49 30 Jun 19:32:49 - [debug] [OpcUa-Client:14a2f4acec0b46f1] Client status: connecting
2023-06-30 19:32:49 30 Jun 19:32:49 - [debug] [OpcUa-Client:14a2f4acec0b46f1] Exact endpointUrl: opc.tcp://myhost:53530/OPCUA/SimulationServer hostname: 080449e845ab
2023-06-30 19:32:49 30 Jun 19:32:49 - [debug] [OpcUa-Client:14a2f4acec0b46f1] 2) Connect using ENDPOINT: opc.tcp://myhost:53530/OPCUA/SimulationServer
2023-06-30 19:32:51 30 Jun 19:32:51 - [error] [http request:54b462104358e43a] RequestError: connect ECONNREFUSED 192.168.1.129:5023
2023-06-30 19:33:04 17:33:04.134Z :client_base_impl              :531   Inner create secure channel has failed Transaction has timed out ( timeout = 15000 ms , request = GetEndpointsRequest)
2023-06-30 19:33:04 30 Jun 19:33:04 - [warn] [OpcUa-Client:14a2f4acec0b46f1] OpcUaClientNode: Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt, using securityMode: [1]
2023-06-30 19:33:04 30 Jun 19:33:04 - [warn] [OpcUa-Client:14a2f4acec0b46f1] OpcUaClientNode:         using securityPolicy: ["http://opcfoundation.org/UA/SecurityPolicy#None"]
2023-06-30 19:33:04 30 Jun 19:33:04 - [warn] [OpcUa-Client:14a2f4acec0b46f1] OpcUaClientNode: Case B: UserName & password does not match to server (needed by Sign), check username: undefined and password: XXX
2023-06-30 19:33:04 30 Jun 19:33:04 - [warn] [OpcUa-Client:14a2f4acec0b46f1] OpcUaClientNode: Invalid endpoint parameters: 
2023-06-30 19:33:04 30 Jun 19:33:04 - [debug] [OpcUa-Client:14a2f4acec0b46f1] Client status: Invalid endpoint, has server security policy: ["http://opcfoundation.org/UA/SecurityPolicy#None"]
2023-06-30 19:33:04 30 Jun 19:33:04 - [error] [OpcUa-Client:14a2f4acec0b46f1] Invalid endpoint

federicoabait, 2023-07-01:
more infromation
[image]

mikakaraila, 2023-07-01:
Remove client node and add it again.
Config and test.

mikakaraila, 2023-07-01:
Sorry I didn't check title, I have to add few lines of code to exit cleanly.

federicoabait, 2023-07-18:
Hi
The new version (Version: 0.2.309), solves this issue?
Thank you
Regards

mikakaraila, 2023-07-18:
Sorry still on vacation and traveling. Back in August or if it is rainy day then I will check this again...

federicoabait, 2023-09-23:
Hi
have you had time to look at this issue?
Thank you
Regards

mikakaraila, 2023-09-25:
Node-red is NOT crashing, client will not connect. What is your proposal? What code should do?