mikakaraila/node-red-contrib-opcua #216: Trouble with certificates

mhechthz, 2020-10-21:
Hi,

I have currently another problem. If I take the default certificate from node-red (nodejs) than the encrypted connection works well. If I copy the file ` ... node_modules/node-opcua-client/certificates/client_selfsigned_cert_2048.pem` to another location and provide it to the opc-ua-node than the connection fails with "invalid endpoint" notification on the node 8even if the endpoint is ok).

So how can I provide my own certificate?

mikakaraila, 2020-10-21:
Actually "own" certificate is the one that is provided by node-opcua for the client.
Certificate contains urn that must match to client or server. You cannot just generate & use any certificate.
Take a look into certificate creation inside node-opcua. It uses config.js and command line parameters.
I am using similar way in my other OPC UA projects.

mhechthz, 2020-10-21:
This means, that it is not possible to provide a self createt certificate and put it in the parameter `Local file`? Or is it just a question of right configuration (with `urn`). 

If the certificate is not right, would it be possible to give the appropriate message like "wrong certificate" instead of "invalid endpoint"?

mikakaraila, 2020-10-21:
There are different things:
1) You can generate certificate with same urn but different valid time range and security length
    Code is not using Local file at the moment because I realized it is very seldom that users can generate certificates.
2) There have been wrong certificates, I just cannot remember what error code it returns.
    Invalid endpoint comes as hostname or security profile does not match to given one.

mhechthz, 2020-10-21:
I'm not sure if I understand right:

1. You don't use the provided certificate from `Local file`? I just copied **the working pem** file to another directory and pointed to this file in `Local file` and even than - the certificate is obviously right - it didn't work! _"Code is not using Local file at the moment"_ --> Why the node is not longer working if I explicitely point to the original file in the original node-opcua folder in the `Local file` field.
2. Deactivating the `Local file` i.e. using the default file works. By just changing to a `Local file` **but all other configurations the same**  the error message is `invalid endpoint`.

Either 1. nor 2. is clear to me.

mhechthz, 2020-10-21:
Meanwhile I managed to setup my own certificate (wrote a python program for this). I replaced the original `private_key.pem` and  `client_selfsigned_cert_2048.pem` by my own versions and voila - it worked. What I need now is, to point to both files in the OPCUA-Client-Node parameters.

nadwir, 2020-10-23:
I support this proposal!

mikakaraila, 2020-10-27:
At the moment client uses always certificate generated by node-opcua postinstall:

[image]

Just to clarify you want to generate your own client certificate and use it?

nadwir, 2020-10-29:
Yes I want to have the option to create the two files myself. The current option to specify a certificate in the node's gui is misleading the user, since it does not work.

mikakaraila, 2020-10-29:
Ack, I will enable it, let´s see how many will really use.

mhechthz, 2020-10-29:
The same for me. I have working certificates and currently copy these to the node-opcua subdirectories. This is not as intended I think.

mikakaraila, 2020-11-02:
Implemented, I will run more tests...

mikakaraila, 2020-11-02:
New version available, please test.

mhechthz, 2020-11-03:
Ok - it works!

Thank You.

closed 2020-11-03 by mikakaraila