mikakaraila/node-red-contrib-opcua #217: Question: Client Subscription

AndreasHeine, 2020-10-22:
Hey,
i am playing around with the Client-Nodes gathering data.

the default interval on the subscription 10s and i changed it to 1s and it looks like it does not change after deploy!?

[image]

Versions:
OS: Win10 64bit
```
22 Oct 11:41:11 - [info] Node-RED version: v1.1.3
22 Oct 11:41:11 - [info] Node.js  version: v12.19.0
```

> node-red-contrib-opcua 0.2.87

> Server is an S7-1500 FW2.6 OPC UA Server

mhechthz, 2020-10-22:
Waht is the change interval of the variable on the server? If it is 10 sec you don't get it more frequently.

AndreasHeine, 2020-10-22:
its the server time so each second and the server max publish interval is also 1000ms so it should work.
if i sub with node-opcua everything works well! i am not sure if the sub configuration is passed or not...

mhechthz, 2020-10-22:
So how do you subscribe?

AndreasHeine, 2020-10-22:
node-red:
[image]
step1:
[image]
step2:
[image]

node-opcua:
```js
const itemToMonitor = {
         nodeId: opcua.resolveNodeId("ns=0;i=2258"),
         attributeId: opcua.AttributeIds.Value
       };
const monitoringParamaters = {
         samplingInterval: 1000,
         discardOldest: true,
         queueSize: 1
       };
```
```
connected !
-------------------------------------
monitored item changed:  2020-10-22T11:45:51.734Z {
  high_low: [ 30845032, -461697290 ],
  picoseconds: 347800000
}
monitored item changed:  2020-10-22T11:45:52.734Z {
  high_low: [ 30845032, -451697382 ],
  picoseconds: 338600000
}
monitored item changed:  2020-10-22T11:45:53.734Z {
  high_low: [ 30845032, -441697420 ],
  picoseconds: 334800000
}
monitored item changed:  2020-10-22T11:45:54.734Z {
  high_low: [ 30845032, -431697413 ],
  picoseconds: 335500000
}
monitored item changed:  2020-10-22T11:45:55.734Z {
  high_low: [ 30845032, -421697369 ],
  picoseconds: 339900000
}
monitored item changed:  2020-10-22T11:45:56.734Z {
  high_low: [ 30845032, -411697443 ],
  picoseconds: 332500000
}
monitored item changed:  2020-10-22T11:45:57.734Z {
  high_low: [ 30845032, -401697404 ],
  picoseconds: 336400000
}
monitored item changed:  2020-10-22T11:45:58.734Z {
  high_low: [ 30845032, -391697162 ],
  picoseconds: 360600000
}
monitored item changed:  2020-10-22T11:45:59.734Z {
  high_low: [ 30845032, -381697418 ],
  picoseconds: 335000000
}
```

AndreasHeine, 2020-10-22:
[image]

mhechthz, 2020-10-23:
If you add a function node with following code

```
subs = {
       'topic': "ns=0;i=2258",
       'datatype':"Double",
       'interval': 500,
       'itemname': "MyItem",
}

return subs;
```

and put it between inject and opc ua client node instead of item node, what is happening then?

AndreasHeine, 2020-10-23:
it works! 
so the client node does not look into the node-red configuration-tab !?

mhechthz, 2020-10-23:
I'm not sure. I never tried it the 'conventional' way since I had hundreds of variables to subscribe. If this is a bug Mika has to deal with it. I'm just an ambitious user ;)

AndreasHeine, 2020-10-23:
i should learn more about node-red custom nodes! so i might be able to fix things on my own and PR ;)

@mikakaraila maybe you can afford some time for me :) !?

mikakaraila, 2020-10-27:
This is one part of the code that deals with interval. I expect if you inject timestamp as payload it will cause this.
Somebody was using payload for this and I had to keep it in the code.
But if you do as above then it should work with msg.interval.

[image]

AndreasHeine, 2020-10-27:
ah ok thanks :)

closed 2020-11-04 by AndreasHeine