mikakaraila/node-red-contrib-opcua #218: Status message is not always proper

mhechthz, 2020-10-23:
Hi,

beside the problem, that in case of wrong certificate the opcua-client-node status changes to `invalid endpoint` instead of `invalid certificate`, there is another issue.

If the opcua-server is ended (i.e. killed) the node status is still `active subscribed`. In case of crashing of opcua-server or network problems, the node does not recognize or not inform, that the connection is broken. In my opinion this is not the expected behaviour, is it?

By the way: thank you again Mika for your excellent work. I hope you are not too seriously annoyed by the lot of questions, issues, etc.

mikakaraila, 2020-10-23:
Hmm if server is down how it could update status?

Just started my weekend, so I will take a look next week.

mikakaraila, 2020-10-23:
Second thought perhaps I should set status from the node-opcua response

mikakaraila, 2020-11-02:
I have extended status to include in case of error actual error. This will help to see more about what is happening.

mikakaraila, 2020-11-02:
I expect invalid endpoint will be seen with details:

[image]

mikakaraila, 2020-11-02:
More positive findings, helps a lot in error finding immediately:

[image]

mhechthz, 2020-11-03:
Helped me directily to point to the right certificate (had a typo). So this seems to be much better. Beside the long text at `node.status`, maybe also ` node.error` could be applied to show the error on hte debug area.

Thank you.

mikakaraila, 2020-11-03:
Yes, I know. I will modify this perhaps later. Let´s see how other people will react to this feature.

As error is under node it shows immediately node it belongs to.

closed 2020-11-03 by mikakaraila