mikakaraila/node-red-contrib-opcua #219: READ action OPCUA Client

Sebalonghi, 2020-10-24:
Hi Mika,

I have a problem with the READ action of OPCUA Client node reading from a server generated from a SCADA software (Premium HMI).
I want to read data only at certain time and therefore I do not want to subscribe a value.
Then I inject the nodeId and set READ to the action on the OPCUA Client expecting that the client returns the actual value.
Indeed, the first time I read the value is always zero.
So I subscribe that node with UAExpert to check the value and discover that actually is NOT zero.
But the best part comes now: when I read the value again with Nodered with UAExpert still running, the value is correct!
As soon as I stop UAExpert, then the readings on the nodered remains constant at the last value read by UAExpert even if the value on the server changes.

I tried to run nodered in verbose mode, but no errors are shown. Everything seams working perfectly but the values read are not correct! What I am missing?

Thank you,

Sebastiano

AndreasHeine, 2020-10-27:
Hey,
could you try to browse before read? ua expert typically did that on selection/selecting the node in the addressspace! had something similar in the past but with a bad serverimplementation of opcua...
lg Andreas

mikakaraila, 2020-10-27:
Can you provide your flow for testing? Do you use interval or just inject again to read latest value?

Sebalonghi, 2020-10-27:
Hi, 
Thank you for your answers.
Well, the test flow is very easy, Manual Inject->OPCUA Item->OPCUA Client in read mode->Debug
If I add a copy of the same flow as above with the OPCUA Client node in subscribe mode and I subscribe once, then the read work all the time...

As Andreas said, it may be a problem of a bad implementation of the OPCUA server. The OPCUA server is a third party server and sorry I cannot give you the credentials to make a test.

I just was wondering if I was missing something. Thank you.

Sebastiano

AndreasHeine, 2020-10-28:
@Sebalonghi  does browse before read work?

closed 2020-12-26 by mikakaraila