mikakaraila/node-red-contrib-opcua #220: invalid endpoint since package uppdate

ZeRico, 2020-10-27:
hi, 
everything was working fine since I decided to update to 0.2.83 to current (0.2.88) version. 
then nomore connection possible without having a "invalid endppoint" error. 
so I tried to come back to 0.2.83 but no success to get a valid connection too ! 

my OPCUA servers (Wago Codesys PLCs) have no certificate and I connect in none/none security policy. 

is there something to do after updating ? 
thanks

ZeRico, 2020-10-27:
update: I finaly tried to downgrade to 0.2.51 as suggestedd by https://github.com/mikakaraila/node-red-contrib-opcua/issues/151 and it worked ! 

so I came back to 0.2.84 and also worked.. 

strange

mikakaraila, 2020-10-27:
It is actually improvement inside node-opcua that does not allow to use localhost in the endpoint.
endpoint must match with the actual hostname exactly

ZeRico, 2020-10-29:
I don't use any localhost, at least IP address yet preferably FQDN. 
nothing has been changed in the client node config

mikakaraila, 2020-10-29:
Perhaps it is causing the invalid endpoint error. There has been some duplicates and re-making flow has solved this issue.

I cannot reproduce this so if you can provide more details that I could inspect then I could try to do something...

mikakaraila, 2020-11-02:
New version will show more info at status, please test.

erossignon, 2020-11-05:
duplicates #215

erossignon, 2020-11-05:
node-opcua@2.24.1 has been released and fixes the issue.

erossignon, 2020-12-11:
node-opcua@2.28.3 patch addresses a remaining issue.

mikakaraila, 2020-12-22:
New version uses now node-opcua@2.28.3

closed 2020-12-22 by mikakaraila