mikakaraila/node-red-contrib-opcua #221: Client Writing Multiple Tags.

GlobalAutomation, 2020-11-02:
Hello, I have around 40 tags I need to read from PLC and have them WRITE MULTIPLE to Server via client. If I have multiple OPCUA items I hit a limit at 10 Items. If I use a function, same thing, only 10. 

eg in function if I have more I get invalid endpoint error. Maybe I'm going about it wrong to write multiple Tags to server.

Simple test flow works up until msg[10] then fails.
[image]
[brokenflow.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/5473733/brokenflow.txt)
[workingflow.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/5473734/workingflow.txt)

mikakaraila, 2020-11-02:
Hmm not limited in the client code. Did you check server operation limits?

GlobalAutomation, 2020-11-02:
The OPCUA Server is created on another flow in Node Red. Its running OK but I'm not sure about limits. I create the folder and variable on the server in the flow as below.

[image]

mikakaraila, 2020-11-02:
Yes, it limits:

[image]

GlobalAutomation, 2020-11-03:
Thank you. Where is this file and do you think there would be implications if I knocked it up to 40?

closed 2020-11-03 by GlobalAutomation