mikakaraila/node-red-contrib-opcua #222: node-red-contrib-opcua should use sticky version number in package.json

erossignon, 2020-11-04:
Instead of using "node-opcua": "^2.24.0" inside node-red-contrib-opcua package.json issue I would recommend to use a sticky version number by dropping the ^ .

` "node-opcua": "^2.24.0"`  => ` "node-opcua": "2.24.0"` 

This will make sure that node-red-contrib-opcua is installed with the version it has been intended for.

mikakaraila, 2020-11-04:
Yes I know. Normally node-opcua versions have been just improvements without any issue.
It has been my decision to use always latest node-opcua.

mikakaraila, 2021-01-13:
As your latest version contained breaking changes now node-red-contrib-opcua uses now node-opcua v2.28.3

closed 2021-01-13 by mikakaraila

erossignon, 2021-01-13:
How ever I still see that you're not using sticky version numbers ..
you may replace "^2.28.3" with "2.28.3" in package.json

mikakaraila, 2021-01-14:
npm package contains correct package.json

I just pushed the latest code v0.2.93 to github:

[image]