mikakaraila/node-red-contrib-opcua #223: Problem with UPC UA server node

johncaipa, 2020-11-06:
Hi, I have installed this library in my node-red in an instance of ubuntu in AWS, I want to implement an OPC UA server but I have the following problem, could you help me to know what the problem is, also know how to implement OPC security with this node.

"Error: EACCES: permission denied, mkdir '/home/ubuntu/.config/node-opcua-nodejs'"

[image]

mikakaraila, 2020-11-07:
You are most probably root?

Then try to install this way: sudo npm install --unsafe-perm=true node-red-contrib-opcua

mikakaraila, 2020-12-22:
Not node-red-contrib-opcua fault, user rights on AWS EC2 Linux instance.

closed 2020-12-22 by mikakaraila

johncaipa, 2021-03-06:
Sorry to bother you again, try again to install again the nodes and use the OPC UA server example (OPCUAServer_Flow.json), but I not had success in getting it up and running. On ubuntu AWS instance.
[image]
What could I do or try to see what the problem is?

mikakaraila, 2021-03-06:
Did you use the latest version?
I installed and it's working.

johncaipa, 2021-03-06:
[image]
I installed it yesterday, so it is a recent version although you have just released a new version, I do not understand what the problem may be, could you share another example that uses OPC UA server to validate if it is my flow?
Excuse so much annoyance I know that maybe it is not an error of your nodes, but I have not managed to make it work and I need to be able to implement this protocol, it is the only option I have

mikakaraila, 2021-03-06:
Can you start node-red from command line?
node-red -v testflow.json
Just to see if it reports something to console?

mikakaraila, 2021-03-06:
My instance on AWS took some time to start. I installed node-red-contrib-opcua through palette manager.

johncaipa, 2021-03-06:
[image]
This is what resulted from running it as you suggested.

mikakaraila, 2021-03-06:
Check PKI folder access rights.
ls -al
Most likely it has ubuntu or root as owner.

johncaipa, 2021-03-06:
Thanks you were right, change ownership of the folder .config and contents inside it from 'root' to 'ubuntu' and now it works.
[image]

mikakaraila, 2021-03-06:
Yes good ☺️