mikakaraila/node-red-contrib-opcua #224: UaSessionPrivate:activeSession - no endpoint matching the session security settings

k-fyraj, 2020-11-10:
I am running node-red on docker and I am trying to set-up a opc-ua server. I created a simple flow, just to start the server, but when I try to connect to the server with UaExpert I get the follow log:
[image]

Here is the flow:
[image]

and the server settings from UaExpert:
[image]

Finally, I constantly get an log entry from node-red:
`contacting discovery server backoff opc.tcp://fb00fda2d8c6:4840 attempt #10 retrying in 50 seconds`

My system:
Node-RED version: v1.2.2
Node.Js version: v10.22.1

When I run the docker container I have also exposed the port of the opc-ua server. In this case 53880

Is this an issue or have I forgotten to configure something?

mikakaraila, 2020-11-10:
Most probably security check hostname and given endpoint hostname are not matching.

This one: fb00fda2d8c6 and given localhost do not match.

If you want to use discovery server then you should install it or make it from node-opcua sample.
I have implemented one, but cannot distribute it as is.

k-fyraj, 2020-11-10:
Thank your for your reply. I am not trying to use a discovery server, I am only trying to configure a simple opc-ua server and connect to it from a client.

Is there a way to configure the OpcUa Server node?

mikakaraila, 2020-11-11:
If you run server inside node-red then you should use methods like addFolder, addVariable, setFolder.

k-fyraj, 2020-11-11:
So, I tried to add a folder or a variable and everytime I get an "Error: Error" message. 
I have attached a log file, maybe it can help. The weird thing is that the server starts with endpoint url: `opc.tcp://189f35f0e199:53880` and not with localhost. Is there a way to configure the server?
[log.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/5523225/log.txt)

k-fyraj, 2020-11-11:
I was able to configure establish a server connection from the OPC UA client, however I was not able to do it from UaExpert - unfortunately, I still get the same error.
Furthermore, I was able to add a variable and a folder to the server, but with `ns=1` using `ns` greater than one will result in an error.

Anyway, I believe I have configured something wrong, that is why I cannot connect from UaExpert to my opc-ua server.
One final question is it possible to configure the server?

mikakaraila, 2020-11-11:
There has been discussion on node-red issues and the solution is to use some flag with docker and then alternateHostname.
Server own namespace is ns=1 and it don´t register any other indexes. So it is limited to ns=1.

mikakaraila, 2020-11-13:
Please check this one:
https://github.com/node-opcua/node-opcua/issues/838

erossignon, 2020-12-11:
node-opcua@2.28.3 patch has been released with a  fix for this.

mikakaraila, 2020-12-22:
New version released 0.2.90 that uses node-opcua@2.28.3

closed 2020-12-22 by mikakaraila