mikakaraila/node-red-contrib-opcua #225: Behaviour of subscription interval not as expected

mhechthz, 2020-12-01:
Hi Mika,

we are just wondering, if the behaviour of the opcua-client-node is the "right" way or not.

It is now perfectly working to inject { ..., interval:1000,...} to a client-node with interval=5000. What we get is every 5 seconds 10 values of the sensor. 

If we inject {..., interval:5000,...} and set the opcua-node to interval=1000, then we get a new value every 1 second.

Is this the correct behaviour of opcua?

mikakaraila, 2020-12-01:
Parameter interval is for server publish interval == server will send data to client.
Monitoring contains sampling interval and I expect you have it 1000ms. Server will read raw value every second.
And server will queue all values and then publish them to client if queue size is long enough and it will not discard old values.

interval 5s and sampling 1s => every 5s with queue 5 values, if sampling 500ms => 10 values every 5s.

mhechthz, 2020-12-02:
Hmmm ... I'm not sure if I understand this. I explicitely write down, what I expect and what happens with `subscription `of one variable:

Variable change frequency at server 200 ms; Injection subscription sampling interval 1000; Opcua-client-node interval 5000 
        --> expected: get every 5 seconds 5 values; observed: get every 5 seconds 5 values --> OK

Variable change frequency at server 200 ms; Injection subscription sampling interval 5000; Opcua-client-node interval 1000 
        --> expected: get every 5 seconds 1 value; observed: get every second 1 value --> NOK

Variable change frequency at server 5000 ms; Injection subscription sampling interval 5000; Opcua-client-node interval 1000 
        --> expected: get every 5 seconds 1 value; observed: get every 5 seconds 1 value --> OK

Variable change frequency at server 5000 ms; Injection subscription sampling interval 1000; Opcua-client-node interval 5000 
        --> expected: get every 5 seconds 1 value; observed: get every 5 seconds 1 value --> OK

So only one situation is not as expected. Is this ok or not ok?

mikakaraila, 2020-12-02:
NOK case:
interval is 1s then server will publish new value every second, sampling will update new value every 5th time.
But timestamp will change... and so even value itself changes every 5th second it will publish it every second.

mhechthz, 2020-12-02:
Ok, so it's a feature, i.e. the intended behaviour of opcua. 

Thank you.

closed 2020-12-02 by mikakaraila