mikakaraila/node-red-contrib-opcua #227: Trouble connecting with Factory IO

halcyon0606, 2020-12-10:
As the title says, there is a problem connecting from Factory IO's OPC client to the server built on node red. The error message that pops up is as shown:

[image]

However, an OPC UA client such as UaExpert is able to connect to the server without an issue. That being said, when the properties of the server is inspected, it is shown that the endpoint URL is not the same as the one that was used to connect, for example, it is given as ocp.tcp://localhost:53880 instead of ocp.tcp://localhost:53880/UA/SimpleNodeRedServer, but this is just an observation and I am not certain whether that is an issue at all.

mikakaraila, 2020-12-10:
This is caused actually by node-opcua. Check your node-opcua version!
Etienne added some security checks into the code due CTT.

halcyon0606, 2020-12-10:
After seeing your reply, I have checked my node-opcua version and I have currently updated it to the latest version which 2.28.2, however, it does not seem like the issue is solved as I still get the same error message in Factory IO.

mikakaraila, 2020-12-10:
Have you checked with node-red -v flow.json

What is output on console?
ocp.tcp://localhost:53880
ocp.tcp://localhost:53880/UA/SimpleNodeRedServer,
opc.tcp://192.1.2.3:53880

The last one with the correct IP address of course.

halcyon0606, 2020-12-11:
It says here on the console that the server endpoint url is opc.tcp://MSI:53880. But as far as I know that is not the endpoint specified as the default endpoint would have have ocp.tcp://MSI:53880/UA/SimpleNodeRedServer.

mikakaraila, 2020-12-11:
Endpoint match is done inside node-opcua, I will wait what Etienne will comment to this one...

mikakaraila, 2020-12-11:
There is fix now at node-opcua for this

erossignon, 2020-12-11:
node-opcua@2.28.3  patch has been released.

halcyon0606, 2020-12-12:
Hello, the issue has been fixed on Factory IO's side, thank you for all the support!

closed 2020-12-12 by halcyon0606