mikakaraila/node-red-contrib-opcua #229: Subscribe action vs monitor action

halcyon0606, 2020-12-15:
Hello,

I'm writing this not as an issue per se, but I do have a question regarding the differences between monitoring and subscribing. So correct me if I'm wrong, but to my understanding, if the client was using a subscription node, they would be subscribed to the item but would not be informed of the changes. On the other hand, if it was in monitoring mode, the client would first check whether a subscription has been made and if not, makes the subscription and then periodically checks for whether that particular item has a changed value of more than the deadband. Thus, would that not make the subscription node obsolete? Or is there a use for the subscription node that I am missing out on? Would love to find out more about this.

mikakaraila, 2020-12-16:
Check this first:

https://reference.opcfoundation.org/v104/Core/docs/Part4/5.12.1/

halcyon0606, 2020-12-17:
I have read through that, and I think I get the gist of it. But I am still unclear on how the subscribe action should be used in the node client's case. Could you provide an example of a client subscribing to a particular item and what output should be expected? From what I have gathered, when I use the monitor action, I would pass a nodeId into the client, and a subscription to that particular item is made and whenever there is a change in data the server would send a response back to the client with the updated value.

mikakaraila, 2020-12-17:
We are normally just subscribing all neededvariables without using monitoring parameters absolute/percent change.
So as generally all items are just subscribed and then server will publish those as value changes. Simple and enough.

I expect if there is some optimization needs to minimize notifications/amount of data server sends then we should use monitoring parameters.

halcyon0606, 2020-12-21:
Ahhh I see, their difference seems clearer to me now. I previously had issues receiving notifications with the client subscription, and monitor seemed to work fine so I became confused as to why subscribe was needed. Thank you for the elucidation on this!

closed 2020-12-21 by halcyon0606