mikakaraila/node-red-contrib-opcua #230: Node Red stops after losing connection to OPC UA Server

PorfirioMtz, 2020-12-15:
Hi...

I have a flow in node red that keeps stopping. I always have to open the "node.js command prompt" and run the "node-red" command because it stops at least once every 1 or 2 days.

This is what the command prompt showed me yesterday that I had the problem. I have opc ua nodes working in the flow getting some vaules from an OPC UA Server. The one that is showing is one that lost connection for several minutes and then Node Red stopped. I don't know if that is what is causing the problem because I tried to replicate it (losing connection to the OPC UA Server) to see if that is what stopped it to no results, node red kept on running. This is the log file information that I saved:

14 Dec 11:40:13 - [error] [OpcUa-Client:CT23] Client node error on: CT23
14 Dec 11:40:13 - [error] [OpcUa-Client:CT23] Client node error on: CT23
14 Dec 11:40:13 - [error] [OpcUa-Client:CT23] Client node error on: CT23
14 Dec 11:40:13 - [error] [OpcUa-Client:CT23] Client node error on: CT23
14 Dec 11:40:13 - [error] [OpcUa-Client:CT23] Client node error on: CT23
14 Dec 11:40:13 - [error] [OpcUa-Client:CT23] Client node error on: CT23
17:40:35.943Z :client_base_impl :468 _internal_create_secure_channel err = ClientSecureChannelLayer => Socket is closed !
14 Dec 11:40:36 - [red] Uncaught Exception:
14 Dec 11:40:36 - TypeError [ERR_INVALID_ARG_TYPE] [ERR_INVALID_ARG_TYPE]: The "data" argument must be of type string or an instance of Buffer, TypedArray, or DataView. Received null
at Hash.update (internal/crypto/hash.js:82:11)
at Object.makeSHA1Thumbprint (C:\Users\trumpf\node_modules\node-opcua-crypto\source\crypto_utils.ts:158:38)
at _verify_serverCertificate (C:\Users\trumpf\node_modules\node-opcua-client\source\private\client_base_impl.ts:212:24)
at C:\Users\trumpf\node_modules\node-opcua-client\source\private\client_base_impl.ts:1027:13
at C:\Users\trumpf\node_modules\node-opcua-client\source\private\client_base_impl.ts:188:9
at C:\Users\trumpf\node_modules\async\dist\async.js:2955:19
at wrapper (C:\Users\trumpf\node_modules\async\dist\async.js:268:20)
at replenish (C:\Users\trumpf\node_modules\async\dist\async.js:435:29)
at iterateeCallback (C:\Users\trumpf\node_modules\async\dist\async.js:424:21)
at C:\Users\trumpf\node_modules\async\dist\async.js:321:20

After that, Node-RED stopped and I had to run it again...

C:\Users\trumpf>node-red
14 Dec 13:09:34 - [info]
Welcome to Node-RED

14 Dec 13:09:34 - [info] Node-RED version: v1.2.2
14 Dec 13:09:34 - [info] Node.js version: v12.19.0
14 Dec 13:09:34 - [info] Windows_NT 10.0.17134 x64 LE
14 Dec 13:09:38 - [info] Loading palette nodes

Can anybody help me with this?

mikakaraila, 2020-12-16:
Can you first check that you have latest packages:
   "node-opcua": "^2.28.3",
    "node-opcua-client": "^2.28.2",
    "node-opcua-crypto": "^1.4.8",
    "node-opcua-pki": "^2.3.0",

mikakaraila, 2020-12-22:
Please test with new 0.2.90 version. I added error checks and updated error handler. At least client didn´t crash when I closed Prosys UA Simulation server and after some time started server client connected again and started to read values.

PorfirioMtz, 2020-12-22:
I have been monitoring the flow for the past days. The thing is that sometimes it crashes and sometimes it doesn´t. Where can I check the version of the nodes that I'm using?

I recently added a catch node to get the errors that the system sends everytime the connection is lost and it seems to be helping, given that before, everytime the connection to the server is lost the node.js command promp starts to generate lots and lots of messages until (sometimes) it stops node red. But I'm not sure if that has anything to do with the problem or if it even affects the application at all, but at the time it seems to be helping because now it doesn´t crash as often. 

I hope I can come back with some new relevant feedback if possible in the future, but for now that's all I got.

mikakaraila, 2020-12-26:
I made v0.2.91 but you can go to .node-red/node_modules/node-red-contrib-opcua and check packages.json

closed 2020-12-26 by mikakaraila