mikakaraila/node-red-contrib-opcua #231: Question regarding server features and possible future implementations

halcyon0606, 2020-12-21:
Hi, so I have been using this node package and I have some questions regarding the server node, mainly:

1) In the case that the box labelled "auto accept unknown certificates" is unticked, what happens when it does receive one? Is there a prompt given to accept it or does it get rejected instantly? In my case, I have not run into this case yet so I am just wondering.
2) In the future, is there going to be any future developments to include user authentication and authorization features for the server? And also additional options to control server limits?

Thank you!

mikakaraila, 2020-12-21:
1. it will accept automatically new certificate and place it into the trusted folder. 
    If false then those certificates are in the rejected folder.
2. I can add those later, after holidays. Can you list what limits you want to control?

halcyon0606, 2020-12-22:
Ahhh okay, that makes more sense to me now thank you. As for the limits, I think having one similar to the compact server one helps, so that would be as shown in the picture below:
[image]
Besides that, I would also like to ask whether the server node currently has a feature in which it would report to other clients whether a particular client has disconnected as it is in my interest to have such a function for a project I am currently involved in.

mikakaraila, 2020-12-22:
Thanks, I will add these during holidays and other features...

halcyon0606, 2020-12-23:
Thank you. Besides that, I would also like to ask whether the server node currently has a feature in which it would report to other clients whether a particular client has disconnected? If so, could you explain to me how I could make it work as it is in my interest to have such a function for a project I am currently involved in.

mikakaraila, 2020-12-23:
OK, I will add new topic that server node will send like:

msg.topic = "Client" and msg.payload "Connected: " + client name

msg.topic = "Client" and msg.payload "Disconnected: " + client name

halcyon0606, 2020-12-23:
Alright, thank you so much and happy holidays!

closed 2020-12-23 by halcyon0606

mikakaraila, 2020-12-25:
Testing client connected/disconnect:

[image]

mikakaraila, 2020-12-25:
UI for server settings (needs still some testing):

[image]

mikakaraila, 2020-12-25:
Seems to work:

[image]

[image]

mikakaraila, 2020-12-25:
Updated node help:

[image]

mikakaraila, 2020-12-26:
Implemented & published just v0.2.91

halcyon0606, 2020-12-28:
Thank you so much for your hard work! Just before I end this thread, I just wanna ask whether PubSub would be a feature for the server in the future and if so, when would be the estimated time for it to be released? Just asking as my supervisor is interested to know.

mikakaraila, 2020-12-28:
Etienne is waiting for sponsors:

https://github.com/node-opcua/node-opcua/issues/809

halcyon0606, 2020-12-29:
Ahh I see, alright, thank you so much!