mikakaraila/node-red-contrib-opcua #232: Updating package issues

halcyon0606, 2020-12-22:
Hi mika,

So previously I brought up an issue regarding the Factory IO client connection and that turned out to be an error in node-opcua's side which was fixed accordingly in its latest update. However, I tried all kinds of method to update the node red package and also the nodejs package but the error with the endpoints not matching still persisted. At the end, I had to uninstall node red and node.js completely and then reinstalling both of them for the error to finally be resolved.  I currently have a raspberry pi running this and I would like not to run through the same diagnosis. Any thoughts on how this could be fixed?

mikakaraila, 2020-12-22:
Strange that should not need nodejs update if it was v8 or newer. Most probably some old packages were still installed. Next time uninstall node-opcua, install new version and command "npm dedupe" can help also to optimize packages.

mikakaraila, 2020-12-22:
package.json uses now latest v2.28.3 of node-opcua

closed 2020-12-22 by mikakaraila